Koenkk/zigbee2mqtt #26564: MQTT autodiscovery topic (homeassistant/device_automation) missing after migrating to 2.1.1-1

cvorn, 2025-03-02:
### What happened?

I'm currently in the process of moving from another home automation solution to Home Assistant. I had an existing standalone installation of Zigbee2Mqtt 1.4, and moved it to the Home Assistant add-on 2.1.1-1.
As per [the documentation](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html), I should be able to respond to button presses using MQTT device triggers after having pressed the respective button at least once. However, these triggers don't show up in Home Assistant. As per the [Home Assistant documentation](https://www.home-assistant.io/integrations/device_trigger.mqtt/), there should be a discovery topic called "device_automation", but there isn't when I check with MQTT Explorer.

[image]

Enabling or disabling the Home Assistant legacy options don't change this behaviour.

### What did you expect to happen?

I expected the "device_automation" topic to exist, like it did in my old install:

[image]

### How to reproduce it (minimal and precise)

Copy the config from Z2M 1.4 into a new install of Z2M 2.0, use MQTT explorer to view topics.

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

20221226

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

### Setup

Add-on on Home Assistant OS in a Proxmox virtual machine

cvorn, 2025-03-07:
I've updated to 2.1.3 and I'm still missing the device_automation topic. I've tried restarting, waiting a few days, adding a completely new device (Tuya remote), completely disabling the Home Assistant integration, deleting the entire homeassistant topic from MQTT, but the device_automation topic just won't show up.
If I enable the experimental event entities, the "event" topic shows up instantly.

Is there anything else I can do or try? Thanks!

maniekes, 2025-03-20:
I got the same with Z2M running in docker after upgrading to 2.1.3(from 1.42.0). 

found a small workaround: when you trigger action on device it will start appearing as a topic(and also in OH discovery). 
but i'm not sure if i want to click every possible button/doubleclick on every remote button that i have in my home just to configure it :D

i see this behaviour on all remote devices: https://www.zigbee2mqtt.io/devices/TS0044_1.html#tuya-ts0044_1 https://www.zigbee2mqtt.io/devices/TS0044.html#tuya-ts0044 and https://www.zigbee2mqtt.io/devices/SH-SC07.html#tuya-sh-sc07
on below sshot you can see the same device type, but device_automation depends on what i have clicked:

[image]

closed 2025-05-28 by github-actions[bot]