Koenkk/zigbee2mqtt #26566: zFlora_S (EfektaLab) The device is displayed as unsupported, even though it is actually supported.

almirus, 2025-03-02:
### What happened?

[image]

[image]

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

20240710

### Adapter

egony cc2652p

### Setup

addon HA on PC

almirus, 2025-03-06:
but the image from z2mqtt https://www.zigbee2mqtt.io/images/devices/zFlora_S.png

jamesonuk, 2025-03-06:
> but the image from z2mqtt https://www.zigbee2mqtt.io/images/devices/zFlora_S.png

The images don't mean it is supported, just that someone has added the images
https://github.com/Koenkk/zigbee2mqtt.io/commit/d90635e5fdd598b6f08c65113552dda16cf3abdb

Looks like they added some new devices https://github.com/Koenkk/zigbee-herdsman-converters/pull/8887 which are in 2.1.3 but this device is not one of them.

You must have added this device manually at some point using an external converter as above

closed 2025-05-06 by almirus