Koenkk/zigbee2mqtt #26567: Tuya ZWT198/ZWT100-BH preset are inverted

gn33, 2025-03-02:
### What happened?

Tuya ZWT198/ZWT100-BH preset are inverted again with 2.12 and 2.1.3 it was ok with 2.1.1

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

7.4.4

### Adapter

EFR32MG21

### Setup

NUC linux

pvdw0310, 2025-03-02:
I have the same problem, the auto and manual setting is switched AGAIN. 
If I set in a automation to manual, the thermostat goes in auto or set to auto the thermostat goes to manual.

Please, have a look that I can set the temperature, in a automation, of the thermostat, this is not working .

Koenkk, 2025-03-02:
Could you provide the data/database.db entry of your device?

gn33, 2025-03-03:
Is that did you want :

{"id":26,"type":"EndDevice","ieeeAddr":"0xa4c138150a6061bb","nwkAddr":64638,"manufId":4417,"manufName":"_TZE204_xnbkhhdr","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65487":14400,"65503":"��W/g","65506":56,"65508":0,"65534":0,"manufacturerName":"_TZE204_xnbkhhdr","zclVersion":3,"appVersion":74,"modelId":"TS0601","powerSource":3}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":217189513},"lastSeen":1740988114901}

Date : 02/03/2025 22:26:15 Europe/Paris
Copie à : gn33 ***@***.***>;
   Author ***@***.***>

pvdw0310, 2025-03-03:
This Is my info:

{"id":48,"type":"EndDevice","ieeeAddr":"0xa4c138d8d0d851f6","nwkAddr":55170,"manufId":4417,"manufName":"_TZE204_xnbkhhdr","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65487":14400,"65503":"eYX/\u0019eYX/\u0007\u0000\u0000\u0000\u0000\u0011","65506":56,"65508":1,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE204_xnbkhhdr","powerSource":3,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":217189513},"lastSeen":1741000577861}

Or a second thermostat:

{"id":42,"type":"EndDevice","ieeeAddr":"0xa4c1389d6cbf9488","nwkAddr":12759,"manufId":4417,"manufName":"_TZE204_xnbkhhdr","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65487":14400,"65503":"^YX/\u0019^YX/\u0007\u0000\u0000\u0000\u0000\u0011","65506":56,"65508":1,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE204_xnbkhhdr","powerSource":3,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":217189513},"lastSeen":1741000577974}

Nopraz, 2025-03-09:
Hello, it seems like my PR broke the modes. You can revert it if you want. Sorry about that

Nopraz, 2025-03-09:
I created the revert here : https://github.com/Koenkk/zigbee-herdsman-converters/pull/8952

Koenkk, 2025-03-09:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-03-09 by Koenkk