Koenkk/zigbee2mqtt #26568: All Hue Lights unavailable in HA after 2.1.2-1

DanDon, 2025-03-02:
### What happened?

Updated to 2.1.2-1 and now all zigbee devices that are not Hue lights work okay but Hue lights show as unavailable in HA. Even though they are showing in the Zigbee2MQTT webui completely fine (and working from there)

### What did you expect to happen?

Presumed this update wouldn't break the Hue lights

### Zigbee2MQTT version

2.1.2-1

### Adapter firmware version

20210708

### Adapter

zStack3x0

### Setup

Add-On on HA OS

chris-1243, 2025-03-02:
Check #26551

It is fixed in 2.1.3

DanDon, 2025-03-02:
Ah yes. For some reason my HA instance didn't seem to think there was any update after 2.1.2-1

Installed the 2.1.3 and all back working now. Thanks.

closed 2025-03-02 by DanDon