Koenkk/zigbee2mqtt #26570: Zigbee2mqtt on home assistant fails to set state of switches frequently

fergalish, 2025-03-02:
### What happened?

Recenty, I since upgrading beyond 2.0.x, I find that the z2m add-on intermittently crashes in an unusual manner. HA continues to receive updates about the states of sensors (e.g. temperature, humidity etc) but can no longer set the state of switches (e.g. switch.turn_on service etc).  This is happening now every few hours and is causing problems (specifically for climate control; automation can't switch devices on or off).  Changing the state of a switch via the z2m device page also fails at this point.

The issue is temporarily fixed by restarting the z2m add-on from the HA add-ons page. Restarting z2m from its own web-page fails to successfully restart- HA does not reconnect to the devices.  The watchdog is enabled but does not restart z2m.

I've tried multiple fixes - reboots, eliminating nice-but-not-necessary add-ons & integrations to reduce resource usage, placed the Zigbee dongle on a powered USB2 hub, throttled a particularly chatty zigbee device (Owon PC321), and so on.  No success.

I recently upgraded to Z2M 2.1.3.

N.B. I tried the ember driver when I upgraded some months ago from ConbeeII to  Dongle-E, but had problems and so remained with EZSP driver.  I don't think this should be relevant as everything was working fine until I upgraded beyond z2m 2.0.x in early February.  Trying ember driver now fails with:

```
[2025-03-02 12:54:49] info: 	zh:ember: ======== Ember Adapter Starting ========
...
[2025-03-02 12:54:50] error: 	z2m: Error while starting zigbee-herdsman
[2025-03-02 12:54:50] error: 	z2m: Failed to start zigbee-herdsman
[2025-03-02 12:54:50] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-03-02 12:54:50] error: 	z2m: Exiting...
[2025-03-02 12:54:51] error: 	z2m: Error: Adapter EZSP protocol version (8) is not supported by Host [13-16].
```

One possible hint of the problem is the size of the "node index.js" process in the z2m container, occupying 10.4G of virtual memory! However swap usage remains at zero and the system can remain functional for several hours even with the 10.4G VSIZE before eventually crashing.

```
Tasks: 416 total,   1 running, 414 sleeping,   0 stopped,   1 zombie
%Cpu(s):  1.9 us,  1.2 sy,  0.0 ni, 95.4 id,  1.5 wa,  0.0 hi,  0.0 si,  0.0 st 
MiB Mem :   7945.2 total,   1233.7 free,   1831.4 used,   4906.8 buff/cache     
MiB Swap:   2048.0 total,   2048.0 free,      0.0 used.   6113.7 avail Mem 

    PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                                                                                                      
 818369 root      20   0   10.4g 122932  51456 S   1.0   1.5   0:07.61 node index.js                                                                                                                
   1589 root      20   0 3724420 151976  55168 S   0.7   1.9   6:26.06 /usr/bin/dockerd -H fd:// --containerd=/run/containerd/containerd.sock                                                       
   2179 fergal    20   0  501416  12928  10496 S   0.7   0.2   0:19.98 /usr/libexec/gvfs-udisks2-volume-monitor                                                                                     
      1 root      20   0  316724  11968   8424 S   0.3   0.1   0:09.88 /sbin/init splash                                                                                                            
    282 root      20   0  108472  57236  55180 S   0.3   0.7   3:10.85 /lib/systemd/systemd-journald                                                                                                
   1337 fergal    20   0   20072  10624   8576 S   0.3   0.1   0:08.39 /lib/systemd/systemd --user                                                                                                  
   4544 root      20   0  321996 109936  27776 S   0.3   1.4   0:34.72 python3 -m supervisor                                                                                                        
   6381 root      20   0   20032   6016   4096 S   0.3   0.1   0:56.29 bash /usr/bin/bashio /etc/s6-overlay/s6-rc.d/mariadb-post/run                                                                
 262167 root      20   0 1320632 656272 107776 S   0.3   8.1   4:39.78 python3 -m homeassistant --config /config                                                                                    
 675333 63112     20   0 3235176  98860  95532 S   0.3   1.2   0:06.05 /lib/systemd/systemd-journal-gatewayd                                                                                        
 822392 root      20   0   11840   4992   2816 R   0.3   0.1   0:00.07 top                                                                                                                          
```                                                                                                                         

### What did you expect to happen?

An automation to, eg, switch a radiator valve on or off would successfully switch the valve.
In the attached log I was luckily testing just at the right moment. You can see:

`[2025-03-02 08:58:37] debug:    z2m:mqtt: Received MQTT message on 'zigbee2mqtt/Breakfast Radiator Valves/set' with data 'ON'`

This was successful.  Moments later:

`[2025-03-02 08:58:49] debug:    z2m:mqtt: Received MQTT message on 'zigbee2mqtt/Playroom Radiator Valves/set' with data 'OFF'`

was unsuccessful. All zigbee switches were now unresponsive and I restarted the z2m add-on.

### How to reproduce it (minimal and precise)

Sorry, no idea. The problem is intermittent.

### Zigbee2MQTT version

2.1.1-1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20240220203028-if00

### Setup

RPi5 debian 12 with HA Supervised installation, Sonoff 3.0+ Dongle-E

### Debug log

[log.log](https://github.com/user-attachments/files/19043115/log.log)

Koenkk, 2025-03-02:
You are using the old `ezsp` driver, it's recommended to use the new `ember` drivers ([info](https://github.com/Koenkk/zigbee2mqtt/discussions/21462)), that may fix your instability

fergalish, 2025-03-02:
Thanks @Koenkk .  I tried the new ember driver but it fails to start:

```
[2025-03-02 12:54:49] info: 	zh:ember: ======== Ember Adapter Starting ========
...
[2025-03-02 12:54:50] error: 	z2m: Error while starting zigbee-herdsman
[2025-03-02 12:54:50] error: 	z2m: Failed to start zigbee-herdsman
[2025-03-02 12:54:50] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-03-02 12:54:50] error: 	z2m: Exiting...
[2025-03-02 12:54:51] error: 	z2m: Error: Adapter EZSP protocol version (8) is not supported by Host [13-16].
```

If upgrading firmware on the Dongle-E is the only way to proceed, I can try that (never did that before, am nervous about it). BUT I reiterate that the system was stable until a few weeks ago with existing firmware & ezsp driver.  Something in a recent software upgrade is causing the problem.

I think [CC1352P2_CC2652P_launchpad_coordinator_20240710](https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20240710.zip) is the correct firmware now?

chris-1243, 2025-03-02:
For your dongle-E, you may want to check this link for a suitable firmware

https://github.com/darkxst/silabs-firmware-builder/releases

This one to flash via a web browser

https://darkxst.github.io/silabs-firmware-builder/

@Nerivec is also providing tools and firmware in a bit more experimental way as described [here](https://www.zigbee2mqtt.io/guide/adapters/emberznet.html)

- First flash a v12 firmware (7.3.x) and start Z2M without any modification in `configuration.yaml`.
- Wait for Z2M to run and stop it.
- Modify your `adapter` section in your configuration and set `ember`.
- Flash a v13 (7.4.5) or v14 (8.0.2) firmware.
- Start Z2M and you should be fine with the `ember` driver.

fergalish, 2025-03-03:
Thanks @chris-1243 .  I have flashed the 7.3.3 firmware successfully. All devices are still present (phew!) and working (double-phew!) and the Z2M 'About' page shows:

```
Coordinator type    EZSP v12
Coordinator revision    7.3.3.0 build 239
```

I stopped it and changed the add-on config to use the ember driver and this fails with the same problem as before [full log attached](https://github.com/user-attachments/files/19051666/log.log):

```
[2025-03-03 10:02:13] info: 	zh:ember: ======== Ember Adapter Starting ========
[2025-03-03 10:02:13] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2025-03-03 10:02:13] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-03-03 10:02:13] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2025-03-03 10:02:13] debug: 	zh:ember:uart:ash: Opening serial port with {"path":"/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20240220203028-if00
","baudRate":115200,"rtscts":false,"autoOpen":false,"parity":"none","stopBits":1,"xon":true,"xoff":true}
[2025-03-03 10:02:13] info: 	zh:ember:uart:ash: Serial port opened
[2025-03-03 10:02:13] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-03-03 10:02:13] error: 	zh:ember:uart:ash: Received frame with CRC error
[2025-03-03 10:02:13] debug: 	zh:ember:uart:ash: ---> [FRAME type=RST]
<snip>
[2025-03-03 10:02:14] debug: 	zh:ember:ezsp: <=== [FRAME: ID=0:"VERSION" Seq=0 Len=7]
[2025-03-03 10:02:14] error: 	z2m: Error while starting zigbee-herdsman
[2025-03-03 10:02:14] error: 	z2m: Failed to start zigbee-herdsman
[2025-03-03 10:02:14] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-03-03 10:02:14] error: 	z2m: Exiting...
[2025-03-03 10:02:14] error: 	z2m: Error: Adapter EZSP protocol version (12) is not supported by Host [13-16].
```

The suggested link seems not relevant.

fergalish, 2025-03-04:
> By the way, you did not completely follow what I wrote.😉
Oops!

Firmware 7.4.5 now installed and ember running. Apparently all is fine.
BTW after the first firmware upgrade, a few smart sockets (silvercrest brand) stopped responding. They still gave updates about energy consumption etc, but wouldn't respond to state switch commands. Re-paired and they more-or-less came back.

Will monitor the situation and report back.
Thanks once again.

fergalish, 2025-03-04:
Ok, so z2m is still crashing now several times in the last few hours  but it's exiting gracefully and so the watchdog restarts it. I  can't see anything common to the crashes (such as updates from a specific device) except "ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE", but I read elsewhere this isn't an issue.

The log attached was from a z2m process that dies after appx 40 minutes.

Here is an excerpt showing "ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT".

Any suggestions on what to look for in the logs to pinpoint the source of the crashes?

```
<snip>
[2025-03-04 21:09:54] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4](ackRx=4 frmTx=5)
[2025-03-04 21:09:54] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4 frmNum=2](frmRx=2) Added to rxQueue
[2025-03-04 21:09:54] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=3](ackRx=4)
[2025-03-04 21:09:54] debug:    zh:ember:uart:ash: <--- [FRAME type=ERROR]
[2025-03-04 21:09:54] error:    zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2025-03-04 21:09:54] error:    zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
[2025-03-04 21:09:54] error:    zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2025-03-04 21:09:54] error:    zh:ember: Adapter fatal error: HOST_FATAL_ERROR
[2025-03-04 21:09:54] debug:    zh:controller: Adapter disconnected
[2025-03-04 21:09:54] info:     zh:ember:uart:ash: ASH COUNTERS since last clear:
[2025-03-04 21:09:54] info:     zh:ember:uart:ash:   Total frames: RX=3472, TX=4064
<snip>
[2025-03-04 21:09:54] info:     zh:ember:uart:ash:   ACK timeouts    : RX=0
[2025-03-04 21:09:54] error:    zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
<snip>
[2025-03-04 21:09:54] info:     zh:ember:uart:ash: Port closed.
[2025-03-04 21:09:54] info:     zh:ember:uart:ash: ======== ASH stopped ========
[2025-03-04 21:09:54] info:     zh:ember:ezsp: ======== EZSP stopped ========
[2025-03-04 21:09:54] info:     zh:ember: ======== Ember Adapter Stopped ========

[excerpt.log.gz](https://github.com/user-attachments/files/19078227/excerpt.log.gz)

[2025-03-04 21:09:54] error:    z2m: Adapter disconnected, stopping
[2025-03-04 21:09:54] debug:    z2m: Saving state to file /config/zigbee2mqtt/state.json
[2025-03-04 21:09:54] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2025-03-04 21:09:54] info:     z2m: Disconnecting from MQTT server
[2025-03-04 21:09:54] info:     z2m: Stopping zigbee-herdsman...
[2025-03-04 21:09:54] debug:    zh:controller:database: Writing database to '/config/zigbee2mqtt/database.db'
[2025-03-04 21:09:57] info:     z2m: Stopped zigbee-herdsman
[2025-03-04 21:09:57] info:     z2m: Stopped Zigbee2MQTT
```

chris-1243, 2025-03-04:
It might be related to #26147 and I would definitely have a look to this closed issue #23761

fergalish, 2025-03-05:
I may have found something. Somewhere between 8-12 seconds before each crash, z2m is saving the state file:
```
[2025-03-05 07:16:34] debug:    z2m: Saving state to file /config/zigbee2mqtt/state.json
...
[2025-03-05 07:16:44] error:    zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2025-03-05 07:16:44] error:    zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
[2025-03-05 07:16:44] error:    zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2025-03-05 07:16:44] error:    zh:ember: Adapter fatal error: HOST_FATAL_ERROR
[2025-03-05 07:16:44] debug:    zh:controller: Adapter disconnected
```

I realise z2m saves this state file every 5 minutes, but could be some disk i/o is blocking a timely return? In all cases, this log line is the last before, 8-12 seconds later, I see the timestamp at which the crashing ERROR occurs.

fergalish, 2025-03-05:
Another piece in the puzzle. I was getting the `ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE` errors, a couple of hundred in 20 minutes for some devices. So I figured I'd re-pair the most troublesome. Some repaired without a hitch, while two devices in particular kept crashing z2m:

```
[2025-03-05 15:53:02] debug:    zh:controller: Device leave '0x3c6a2cfffed15e31'
[2025-03-05 15:53:02] debug:    zh:controller: Removing device from database '0x3c6a2cfffed15e31'
[2025-03-05 15:53:02] debug:    zh:controller:database: Writing database to '/config/zigbee2mqtt/database.db'
[2025-03-05 15:53:16] warning:  z2m: Device 'HX DHW Cold' left the network
...
[2025-03-05 15:53:16] error:    zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2025-03-05 15:53:16] error:    zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
[2025-03-05 15:53:16] error:    zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2025-03-05 15:53:16] error:    zh:ember: Adapter fatal error: HOST_FATAL_ERROR
```

So I don't think the device itself caused the crash, but instead triggered a database.db write-out, which caused the crash for reasons yet unknown.

fergalish, 2025-03-05:
Aaargh, I have an embarrassing update. One line will tell you all - i/o errors on the SD card.
`[25194.618432] I/O error, dev mmcblk0, sector 3567616 op 0x0:(READ) flags 0x4000 phys_seg 32 prio class 2`
I assume at this point this is the exclusive cause of the issues, if anything emerges to contradict that, I'll let you know.
Thanks @Koenkk @chris-1243 for the help & guidance.

fergalish, 2025-03-09:
Final comment here. I've migrated (with not insignificant challenges) to a new SD card. Now almost 24h later and there have been no ACK TIMEOUT errors. Seems completely conclusive - i/o errors were causing the ACK timeouts.

As supplementary information, it seems the problematic files were in the MariaDB files; is it possible that the MariaDB add-on does a lot of disk i/o and thereby caused much higher wear on the blocks where its files are stored? I'd've thought that a modern SD card would have built-in wear-levelling, but that's just an assumption.

Cheers to all.

closed 2025-03-09 by fergalish