Koenkk/zigbee2mqtt #26572: [New device support]: Zemismart Fan + Light

UgurSezgin, 2025-03-02:
### Link

https://tr.aliexpress.com/item/1005007757066540.html?spm=a2g0o.order_list.order_list_main.10.455a3d12fcey9q&gatewayAdapt=glo2tur

### Database entry

{"id":21,"type":"Router","ieeeAddr":"0xa4c138af9ce30259","nwkAddr":54788,"manufId":4417,"manufName":"_TZE204_2jnoy8dj","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"LIW/ikIW/ivIW/iyIW/i","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":74,"modelId":"TS0601","manufacturerName":"_TZE204_2jnoy8dj","powerSource":1,"zclVersion":3,"hwVersion":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1740926166210}

### Zigbee2MQTT version

2.1.3

### Comments

Hello,
I bought the device in the link I shared. Actually, I was looking for a Zigbee supported Fan and Light Module for a long time and I saw this device. So, I bought it hoping that Zigbee2Mqtt would support it. I bought one for testing purposes and if I don't have any problems I will buy another one. But my fears came true and I saw that it was not supported. I read a lot of articles explaining how to add a new device but they are very difficult for me. Could you please provide support for this device?

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_2jnoy8dj',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

The device only connects to the zigbee network. But it does not have any features.

merida27, 2025-03-05:
Same situation here. I managed to communicate from HA to the device but no way to receive actions when device is clicked. Thanks!

UgurSezgin, 2025-03-09:
There is a device supported as **TS0601**. Can the infrastructure of this device be used to support the "**Zemismart** Fan + Light" device? I think it can be for turning the light on and off and the fan speed. Also, can it be possible to adjust the RGB light to be displayed when it is on and the RGB light to be displayed when it is off in the slider speed setting from another RGB device?

[https://www.zigbee2mqtt.io/devices/TS0601_fan_and_light_switch.html](url)

closed 2025-05-17 by github-actions[bot]