Koenkk/zigbee2mqtt #26574: Wireless switch with 4 buttons (TS0044) by Tuya not show Action Entity

hongthang1110, 2025-03-02:
### What happened?

After I added the Wireless switch with 4 buttons (TS0044) by Tuya device to Zigbee2MQTT, all entities (Actions) were displayed on the Zigbee2MQTT interface (attached image). But when entering the Device MQTT interface, the Action Entry is missing (picture attached).

[image]
[image]

### What did you expect to happen?

Expect the device in the MQTT list to show all entities such as the Zigbee2MQTT interface (additional action entities)

### How to reproduce it (minimal and precise)

Add a new device Wireless switch with 4 buttons (TS0044) by Tuya and check on 2 interfaces Zigbee2MQTT and MQTT in Devices Hass

### Zigbee2MQTT version

2.1.3-1

### Adapter firmware version

20220928

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

Add-on Home assistant on Intel NUC

ghoz, 2025-03-02:
Hi,
This is a breaking change due to the removal of the legacy actions sensors from defaults in z2m 2.0.0 : https://github.com/Koenkk/zigbee2mqtt/discussions/24198
A search in github issues would have returned a lot of answers...

> All action sensors are now disabled by default (sensor.*_action entities). It's recommended to use the MQTT device trigger instead. In case you really need the action sensors, add the following to your [zigbee2mqtt] configuration.yaml.
> ```
> homeassistant:
>   legacy_action_sensor: true
> ```

closed 2025-03-02 by Koenkk

hongthang1110, 2025-03-04:
Thanks all!