Koenkk/zigbee2mqtt #26576: No devices showing in Web UI after Z2M update

basvdploeg, 2025-03-02:
### What happened?

After updating from 2.1.0-1 to 2.1.3-1, all devices disappeared from the Web UI. Also no logs are available from the web interface.

Note: The devices are still working fine and are still controllable via Home Assistant. The add-on log shows no errors and displays all device updates like normal.

When I revert back to a 2.1.0-1 backup, all devices show up immediately again.

### What did you expect to happen?

Show all my devices in the Web UI.

### How to reproduce it (minimal and precise)

Update Z2M from 2.1.0-1 to 2.1.3-1. Enter the Web UI. 

### Zigbee2MQTT version

2.1.3-1

### Adapter firmware version

20221226

### Adapter

Silicon Labs Sonoff Zigbee 3.0 USB Dongle Plus (zStack3x0)

### Setup

Z2M Add-on, HA Core 2025.2.5 - Supervisor 2025.02.4 - Operating System 14.2, Mac mini M4

Hotabka, 2025-03-02:
Same, but: only 3 devices in web ui...of 30...but all still working...downgraded to 2.1.1 ...not solved problem...still 3 devices...

basvdploeg, 2025-03-02:
@Hotabka I doubt that’s same issue.

My Web UI isn’t showing devices or logs _at all_.

Hotabka, 2025-03-02:
Just restarted HA, all devices back in WebUI

basvdploeg, 2025-03-02:
Yeah, that’s not working here. So clearly a different problem.

Nerivec, 2025-03-02:
Did you try uninstalling/reinstalling the HA add-on?
Also make sure you are not seeing a weird browser caching issue with frontend (clear cache/cookies).

basvdploeg, 2025-03-03:
@Nerivec I just reinstalled the add-on. The results are the same. As soon as version 2.1.3-1 is installed, the devices In the UI are gone.

When I revert back to 2.1.0-1, they show up immediately again.

Nerivec, 2025-03-03:
And other pages of frontend have data? Like the top of `About`?

basvdploeg, 2025-03-03:
@Nerivec The top of `About` is showing up fine. I've made some comparisons screenshots between the two versions:

✅ 2.1.0-1 - **Devices**:
[image]

⛔️ 2.1.3-1 - **Devices**:
[image]

✅ 2.1.0-1 - **Logs**:
[image]

⛔️ 2.1.3-1 -  **Logs**:
[image]

✅ 2.1.0-1 - **About**:
[image]

⛔️ 2.1.3-1 - **About**:
[image]

✅ 2.1.0-1 - **Raw**:
[image]

⛔️ 2.1.3-1 - **Raw**:
[image]

basvdploeg, 2025-03-03:
@Koenkk This is the log after the update to `2.1.3-1`:

[log.log](https://github.com/user-attachments/files/19058593/log.log)

(After taking the log I reverted back to my working `2.1.0-1` backup again.)

Koenkk, 2025-03-04:
I don't see anything wrong yet, z2m seems to start normally. Could you also provide the debug log when starting 2.1.0-1?

basvdploeg, 2025-03-04:
@Koenkk Here is the log after reverting back to `2.1.0-1`:

[log (1).log](https://github.com/user-attachments/files/19076620/log.1.log)

Not sure if you can do something with this information but... I took a look at the network requests in my browser. When using `2.1.0-1` it starts a WebSocket via `api` and that shows me all the streaming device data like this:

[image]

On version `2.1.3-1` it starts the WebSocket, but it closes after receiving some payloads, after which it makes a new `api` request. It never shows the streaming device data:

[image]

Koenkk, 2025-03-06:
I see you are using Safari, could you try with a different browser like Chrome or Edge?

basvdploeg, 2025-03-06:
@Nerivec Yes, I've tried all the usual reboots / restarts. Both HA and the actual virtual machine.

Every time when I revert back to the old `2.1.0-1` version, *poof* it instantly works again.

latek0, 2025-03-07:
Same problem

Koenkk, 2025-03-08:
Looking at the delta I cannot explain why this happens: https://github.com/Koenkk/zigbee2mqtt/compare/2.1.0..2.1.3

- `zigbee2mqtt-frontend` version has not been changed
- `ws` library has been updated from 8.1.0 to 8.1.1 but this only has a change in the tests (https://github.com/websockets/ws/releases/tag/8.18.1)
- Any change you could update the intermediate version? E.g. to z2m 2.1.1 or 2.1.2?

basvdploeg, 2025-03-08:
@Koenkk I don't really know how to do that. I thought that it was only possible to update to the latest version.

Nerivec, 2025-03-08:
Can you check the HA logs, see if you get something around the time of the error in your browser?
`Settings > System > Logs`

basvdploeg, 2025-03-08:
@Nerivec Nothing strange happening there. It shows the same stuff as in the logs that I've shared before. 

I've done a bit more digging...

When I fully remove the Z2M add-on (and it's data) and reinstall it without putting back my `database.db`, it boots up normally.

As soon as I use my old `database.db` (which works fine in `2.1.0-1`) the problem is back again. 

Can there be something wrong with my database? Would it make sense to share it with you, @Koenkk or @Nerivec?

Koenkk, 2025-03-10:
I cannot reproduce the issue with your `database.db` (even with Safari) but I suspect there is a magic char in your database.

Could you try removing the `65503` attributes for line 127 - 133? e.g. 

```json
{"id":131,"type":"Router","ieeeAddr":"0xa4c138b78562c24d","nwkAddr":56465,"manufId":4417,"manufName":"_TZ3000_j1v25l17","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57345,57344,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"q�T/\u0016","65506":53,"65508":0,"65534":0,"modelId":"TS011F","appVersion":68,"manufacturerName":"_TZ3000_j1v25l17","powerSource":1,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":0,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"powerOnBehavior":0,"switchType":0}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":235,"rmsCurrent":0,"activePower":0}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":14}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dc0987","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dc0987","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1741451178822}

```

becomes

```json
{"id":131,"type":"Router","ieeeAddr":"0xa4c138b78562c24d","nwkAddr":56465,"manufId":4417,"manufName":"_TZ3000_j1v25l17","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57345,57344,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65506":53,"65508":0,"65534":0,"modelId":"TS011F","appVersion":68,"manufacturerName":"_TZ3000_j1v25l17","powerSource":1,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":0,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"powerOnBehavior":0,"switchType":0}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":235,"rmsCurrent":0,"activePower":0}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":14}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dc0987","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dc0987","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1741451178822}
```

basvdploeg, 2025-03-10:
@Koenkk Unfortunately that doesn't make a difference.

Again, I've done some more troubleshooting... 

It turns out that it's not about lines `127` to `133` specifically. If I remove **7** of the **Lidl HG08673** (`_TZ3000_j1v25l17`) plugs from the database, the UI shows correctly again. It doesn't matter **which** of the plugs, but **has** to be - at least - seven of them. 

When I add one of the plugs back again, the interface stops showing the devices again after a refresh.

I can add other types of devices (for example an IKEA button) without breaking the interface.

Currently I use 48 pieces of the Lidl HG08673 plugs. Do you think that there is some kind of limit introduced somewhere between `2.1.0-1` and  `2.1.3-1`.

Koenkk, 2025-03-11:
@basvdploeg are you sure it doesn't happen when removing any of the other devices? (even if you remove like 30)

basvdploeg, 2025-03-11:
@Koenkk Just double checked it, but no. If I remove other devices, but keep the Lidl plugs, the interface keeps being empty.

Koenkk, 2025-03-12:
Could you:
- Remove devices until it starts working
- Add 1 device, verify that it stops working now
- Start removing stuff from that device entry, starts with `endpoints`
- See at what point it starts working again.

Koenkk, 2025-03-13:
What if you change it to `"modelId":"TS012F"`? (it will be detected as unsupported, but does the frontend load?)

basvdploeg, 2025-03-13:
@Koenkk Yes! When I change it to `TS012F` the UI works again. 

Like you said, it's detected as unsupported, but the data seems to be coming in fine. 

[image]
[image]

When I add the `TS012F` modification to multiple of the plugs, they all show up in the UI. And they keep actually showing their data too (although not in this screenshot, because I just restarted):

[image]

Koenkk, 2025-03-15:
Could you change it back now (such that it is detected as supported) and try with the following [external converter](https://gist.github.com/Koenkk/8216fd3ae316f857da71e5ab7d3035d7):

- save this as file next to `configuration.yaml` in the `external_converters` directory (create if it doesn't exist yet) as `ext_converter.js` (so `external_converters/ext_converter.js`)
- start z2m, 
- in the frontend, check if the description of this device now ends with `CUSTOM` (this indicates the external converter has been loaded correctly)
- check if issue is fixed, if not change `extend: [tuya.modernExtend.tuyaOnOff()],` to `extend: [],`

Note that not all features are supported, add them back one by one by updating `tuya.modernExtend.tuyaOnOff()` with the commented one.

basvdploeg, 2025-03-15:
The external converter loads correctly.

I can enable all functions without breaking the UI:
`extend: [tuya.modernExtend.tuyaOnOff({electricalMeasurements: true, powerOutageMemory: true, indicatorMode: true, childLock: true})],`

Am I supposed to enable the other commented lines too? Because `onEvent` and `whiteLabel` can be uncommented.

It goes wrong when I uncomment the `options: [exposes.options.measurement_poll_interval().withDescription("Only the energy value is polled for this device.")]` line.

Koenkk, 2025-03-16:
> It goes wrong when I uncomment the options: [exposes.options.measurement_poll_interval().withDescription("Only the energy value is polled for this device.")] line.

Ah! Maybe this one is already fixed in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) , could you try that?

basvdploeg, 2025-03-16:
Awesome! It is indeed fixed in when I use the Edge add-on (and without the external converter). 

Any idea when this will propagate to the stable version?

_(Thanks for all the help!)_

Koenkk, 2025-03-17:
Great, new release = every 1st of the month

closed 2025-03-17 by Koenkk

BenDavidson90, 2025-04-04:
Same problem here, since last update the devices list is empty but devices are working.
Using it via docker. Switched to dev version 2.2.0-dev (commit #37927e698abcb242965423e9ee37623255feadab) via the repo `koenkk/zigbee2mqtt:latest-dev`, same problem...

Here are the logs:
[log.log](https://github.com/user-attachments/files/19609147/log.log)