Koenkk/zigbee2mqtt #26578: Backlight cannot be controlled

devkaiwang, 2025-03-02:
### What happened?

I bought a three gang switch with with neutral wire. It was recognized as ``
I wrote an external converter as following but **I cannot turn the LED backlight off from**. I wonder if there is something I missed or is it an issue of zigbee2mqtt? Thank you very much! 

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {
            modelID: "TS0726",
            manufacturerName: "_TZ3002_rbnycsav",
        }
    ],
    model: 'TS0726',
    vendor: 'Tuya',
    description: 'Tuya Three Switch',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [
        tz.on_off,
        tuya.tz.datapoints
    ],
    configure: tuya.configureMagicPacket,
    exposes: [
        // Switches
        tuya.exposes.switch().withEndpoint('switch_1'),
        tuya.exposes.switch().withEndpoint('switch_2'),
        tuya.exposes.switch().withEndpoint('switch_3'),

        // Backlight options
        e.binary('backlight', ea.ALL, 'ON', 'OFF').withDescription('Backlight Switch'),
        //tuya.exposes.indicatorModeNoneRelayPos(),
        e.enum("light_mode", ea.ALL, ["none", "relay", "pos"]).withDescription("Mode of backlight"),
    ],
    endpoint: (device) => {
        return {
            'switch_1': 1,
            'switch_2': 2,
            'switch_3': 3
        };
    },
    meta: {
        multiEndpoint: true,
        //disableDefaultResponse: true,
        timeout: 5000,
        tuyaDatapoints: [
            // Switches
            [24, 'switch_1', tuya.valueConverter.onOff],
            [25, 'switch_2', tuya.valueConverter.onOff],
            [26, 'switch_3', tuya.valueConverter.onOff],

            // Backlight
            [36, 'backlight', tuya.valueConverter.onOff],
            [37, 'light_mode', tuya.valueConverterBasic.lookup({'none': 0, 'relay': 1, 'pos': 2})],
        ],
    },
};

module.exports = definition;
```

Device properties from Tuya Developer:
```json
[... 268 lines cut ...]
debug 2025-03-03 01:26:14zh:zstack:unpi:parser: --- parseNext []
debug 2025-03-03 01:26:14zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,145,87]
debug 2025-03-03 01:26:14zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,145,87]
debug 2025-03-03 01:26:14zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,145] - 87
debug 2025-03-03 01:26:14zh:zstack:znp: <-- AREQ: AF - dataConfirm - {"status":0,"endpoint":1,"transid":145}
debug 2025-03-03 01:26:14zh:zstack:unpi:parser: --- parseNext []
info 2025-03-03 01:26:14z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138884402f357', payload '{"backlight":"OFF","light_mode":"pos","linkquality":105,"power_on_behavior_l1":"previous","power_on_behavior_l2":"on","power_on_behavior_l3":"previous","state_l1":"ON","state_l2":"OFF","state_l3":"OFF","state_switch_1":"OFF","state_switch_2":"OFF","state_switch_3":"OFF"}'
debug 2025-03-03 01:26:15z2m:mqtt: Received MQTT message on 'zigbee2mqtt/0xa4c138884402f357/set' with data '{"light_mode":"none"}'
debug 2025-03-03 01:26:15z2m: Publishing 'set' 'light_mode' to '0xa4c138884402f357'
debug 2025-03-03 01:26:15zh:controller:endpoint: ZCL command 0xa4c138884402f357/1 manuSpecificTuya.dataRequest({"seq":1,"dpValues":[{"dp":37,"datatype":2,"data":[0,0,0,0]}]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false})
debug 2025-03-03 01:26:15zh:zstack: sendZclFrameToEndpointInternal 0xa4c138884402f357:59266/1 (0,0,1)
debug 2025-03-03 01:26:15zh:zstack:znp: --> SREQ: AF - dataRequest - {"dstaddr":59266,"destendpoint":1,"srcendpoint":1,"clusterid":61184,"transid":146,"options":0,"radius":30,"len":13,"data":{"type":"Buffer","data":[17,67,0,1,0,37,2,0,4,0,0,0,0]}}
debug 2025-03-03 01:26:15zh:zstack:unpi:writer: --> frame [254,23,36,1,130,231,1,1,0,239,146,0,30,13,17,67,0,1,0,37,2,0,4,0,0,0,0,73]
debug 2025-03-03 01:26:15zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
debug 2025-03-03 01:26:15zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
debug 2025-03-03 01:26:15zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
debug 2025-03-03 01:26:15zh:zstack:znp: <-- SRSP: AF - dataRequest - {"status":0}
debug 2025-03-03 01:26:15zh:zstack:unpi:parser: --- parseNext []
debug 2025-03-03 01:26:15zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,146,84]
debug 2025-03-03 01:26:15zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,146,84]
debug 2025-03-03 01:26:15zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,146] - 84
debug 2025-03-03 01:26:15zh:zstack:znp: <-- AREQ: AF - dataConfirm - {"status":0,"endpoint":1,"transid":146}
debug 2025-03-03 01:26:15zh:zstack:unpi:parser: --- parseNext []
info 2025-03-03 01:26:15z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138884402f357', payload '{"backlight":"OFF","light_mode":"none","linkquality":105,"power_on_behavior_l1":"previous","power_on_behavior_l2":"on","power_on_behavior_l3":"previous","state_l1":"ON","state_l2":"OFF","state_l3":"OFF","state_switch_1":"OFF","state_switch_2":"OFF","state_switch_3":"OFF"}'
```

devkaiwang, 2025-03-07:
I have checked the data points on different sections on Tuya IoT Platform and also debugged the backlight property using my Tuya Gateway, the data points are all correct. 

As you can see from my converter code, `36` and `37` are used for `backlight_switch` and `light_mode`. And I have tried different ways of writing the converter, none of them is a success. 

@sjorge @Koenkk  could you please tell me what is the next step I can try? Is there anything I might have missed to support the Tuya backlight from Z2M in this specific case? Thanks!

[image]

sjorge, 2025-03-07:
Not saying this is the correct thing to do, but It's what I would do if I had the device, their gateway and a spare zigbee stick for sniffing.

I'd sniff the entire conversation from pairing the device to the gateway, then changing the configuration.
Then do the same with zigbee2mqtt and compare the results. It might be that some sort of initialization step during pairing is missed or that the payload somehow ends up encoded slightly differently.

The main issue is going to be (aside from needing a lot of hardware) is if you can figure out the decryption key for the gateway.

devkaiwang, 2025-03-07:
Hi @sjorge, thanks so much for your help! The learning curve seems a bit steep for a non-IoT **new-(zig)bee** like me :) But I'll give it a try and hope to find some resources to guide me along the way.

sjorge, 2025-03-07:
There is a guide on how to sniff traffic here https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

Koenkk, 2025-03-08:
This device does not seem to use the Tuya datapoints (even though you can see them in the Tuya portal). Could you try with this ext converter; https://gist.github.com/Koenkk/795117ed4573335fab3ce6494dd48442

devkaiwang, 2025-03-08:
> This device does not seem to use the Tuya datapoints (even though you can see them in the Tuya portal). Could you try with this ext converter; https://gist.github.com/Koenkk/795117ed4573335fab3ce6494dd48442

Hi @Koenkk , thank you for the help! 

**I can turn off the backlight now!** I think it is a big step forward to me! 

However, except for the backlight on/off function, other controls do not work. Additionally, the Light Mode option is missing. Without its configuration, the backlight turns on when the corresponding gang is on.

[image]

devkaiwang, 2025-03-09:
This code is working for three switches, backlight on/off, seems also correct for switch/scene mode toggle. (at least functionally normal and no error/warning in the log)

But we still need `["none", "relay", "pos"]` for `light_mode` and it is not supported by `tuya.modernExtend.tuyaOnOff`.

What should I do now? @Koenkk 

Thanks!

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint("TS0726", ["_TZ3002_rbnycsav"]),
    model: "TS0726_3_gang",
    vendor: "Tuya",
    description: "Test:3 gang switch with neutral wire",
    extend: [
        tuya.modernExtend.tuyaOnOff({
            switchMode: true,
            powerOnBehavior2: true,
            backlightModeOffOn: true,
            endpoints: ["l1", "l2", "l3"],
        }),
    ],
    endpoint: (device) => {
        return {l1: 1, l2: 2, l3: 3};
    },
    meta: {multiEndpoint: true},
    configure: async (device, coordinatorEndpoint) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint);
        for (const ep of [1, 2, 3]) {
            await reporting.bind(device.getEndpoint(ep), coordinatorEndpoint, ["genOnOff"]);
        }
    },
};

module.exports = definition;
```

Koenkk, 2025-03-09:
@devkaiwang the modernExtend can be used together with the old exposes/toZigbee/fromZigbee.

devkaiwang, 2025-03-11:
Same code also works for another switch called T10. I have a 2 gang sample:

@Koenkk seems this model is widely used as solutions for switches with backlights. Would you please consider to support it officially?

[image]

(T10)

[image]

(Common Tuya smart switches with backlights)

```js

/// Works for 2-gang T10 smart switch with backlight
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const utils = require('zigbee-herdsman-converters/lib/utils');

const e = exposes.presets;
const ea = exposes.access;

const fz_converter = {
    cluster: "genOnOff",
    type: ["attributeReport", "readResponse"],
    convert: (model, msg, publish, options, meta) => {
        const result = {};

        if (msg.data.tuyaBacklightSwitch !== undefined) {
            const backlightMode = utils.getFromLookup(msg.data.tuyaBacklightSwitch, { 0: "off", 1: "on" }, null);
            if (backlightMode !== null) {
                result.backlight_mode = backlightMode;
            }
        }

        if (msg.data.tuyaBacklightMode !== undefined) {
            const indicatorMode = utils.getFromLookup(msg.data.tuyaBacklightMode, { 0: "none", 1: "relay", 2: "pos" }, null);
            if (indicatorMode !== null) {
                result.indicator_mode = indicatorMode;
            }
        }

        return result;
    },
};

const tz_converter = {
    key: ["backlight_mode", "indicator_mode"],
    convertSet: async (entity, key, value, meta) => {
        const lookup = key === "backlight_mode" ? { off: 0, on: 1 } : { none: 0, relay: 1, pos: 2 };
        const attribute = key === "backlight_mode" ? "tuyaBacklightSwitch" : "tuyaBacklightMode";
        const result = utils.getFromLookup(value, lookup);

        await entity.write("genOnOff", { [attribute]: result });
        return { state: { [key]: value } };
    },
    convertGet: async (entity, key, meta) => {
        const attribute = key === "backlight_mode" ? "tuyaBacklightSwitch" : "tuyaBacklightMode";
        await entity.read("genOnOff", [attribute]);
    },
};

// Zigbee device definition
const definition = {
    fingerprint: tuya.fingerprint("TS0726", ["_TZ3000_ezqbvrqz"]),
    model: "TS0726_3_gang",
    vendor: "Tuya",
    description: "2 gang switch with neutral wire",
    extend: [
        tuya.modernExtend.tuyaOnOff({
            switchMode: true,
            powerOnBehavior2: true,
            backlightModeOffOn: true, // Backlight mode auto-detected
            endpoints: ["l1", "l2"],
        }),
    ],
    fromZigbee: [fz_converter],
    toZigbee: [tz_converter],
    exposes: [
        exposes.enum("indicator_mode", ea.ALL, ["none", "relay", "pos"])
            .withDescription("Indicator mode"),
    ],
    endpoint: (device) => ({ l1: 1, l2: 2 }),
    meta: { 
        multiEndpoint: true 
    },
    configure: async (device, coordinatorEndpoint) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint);
        for (const ep of [1, 2]) {
            await reporting.bind(device.getEndpoint(ep), coordinatorEndpoint, ["genOnOff"]);
        }
    },
};

module.exports = definition;

```

Koenkk, 2025-03-12:
Great, Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

devkaiwang, 2025-03-13:
Hi @Koenkk , I'm glad and will create a PR later. Please prepare to answer stupid questions during the review process from who is really new to the project!

devkaiwang, 2025-04-10:
Here is my PR - https://github.com/Koenkk/zigbee-herdsman-converters/pull/9158

closed 2025-04-15 by devkaiwang