Koenkk/zigbee2mqtt #26580: Notification spam

corporategoth, 2025-03-02:
### What happened?

I'm getting a metric butt load of notifications spamming my UI, which makes the right side of the UI unusable.

The spam messages are the notification equivalent of the log messages in the debug log.
These are likely either a problem with the ember driver (as they seem to be new, or the frequency is), or the firmware that @Nerivec made for the Dongle-E (I've sent him a message on discord).
Trying to grep the *.yaml, *.db and *.json files Z2M uses don't find the IDs it mentiojns.

However, it is impossible to suppress these messages.  So any button on the right side of the UI cannot be clicked (worse if using a mobile browser).  So, for example, I could not click an OTA update button, or the edit/interview/delete/etc buttons, etc.

Clicking away notifications doesn't help either, they appear too often and keep filling the screen for me to be able to click them away fast enough for me to use the UI effectively.

### What did you expect to happen?

Either 1) these notifications don't happen (if it's an ember bug), or 2) have the ability to suppress or enable notifications, or types of notifications so that I can make the UI usable again.

### Zigbee2MQTT version

2.12

### Adapter firmware version

8.2 GA

### Adapter

Sonof Dongle-E

### Setup

Stand-alone docker container

### Debug log

[2025-03-02 15:06:59] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "55690".
[2025-03-02 15:06:59] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "55696".
[2025-03-02 15:06:59] warning:  zh:ember:ezsp: An ID conflict was detected for network address '55696'. Corresponding devices kicked from the network.
[2025-03-02 15:06:59] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "55696".
[2025-03-02 15:06:59] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "55696".
[2025-03-02 15:06:59] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "55696".
[2025-03-02 15:06:59] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "55690".
[2025-03-02 15:06:59] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "55696".
[2025-03-02 15:06:59] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "55696".
[2025-03-02 15:06:59] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "7123".
[2025-03-02 15:06:59] warning:  zh:ember:ezsp: An ID conflict was detected for network address '7123'. Corresponding devices kicked from the network.
[2025-03-02 15:06:59] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "24299".
[2025-03-02 15:06:59] warning:  zh:ember:ezsp: An ID conflict was detected for network address '24299'. Corresponding devices kicked from the network.
[2025-03-02 15:06:59] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "55187".
[2025-03-02 15:06:59] warning:  zh:ember:ezsp: An ID conflict was detected for network address '55187'. Corresponding devices kicked from the network.
[2025-03-02 15:06:59] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "55187".
[2025-03-02 15:06:59] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "20445".
[2025-03-02 15:06:59] warning:  zh:ember:ezsp: An ID conflict was detected for network address '20445'. Corresponding devices kicked from the network.
[2025-03-02 15:06:59] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "20445".
[2025-03-02 15:07:00] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "55696".
[2025-03-02 15:07:00] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "20445".
[2025-03-02 15:07:00] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "24299".
[2025-03-02 15:07:00] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "55187".
[2025-03-02 15:07:00] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "55187".

corporategoth, 2025-03-02:
FYI - after conversations with @Nerivec the originator of these messages seems to be a misbehaving (maybe) device on my network.

However, the point is, a bunch I should still be able to no disable notifications without reducing logging, or be able to ensure that just because I'm getting a bunch of notifications, the UI is not unusable for anything on the right of the UI.

Nerivec, 2025-03-03:
Let's continue in https://github.com/nurikk/zigbee2mqtt-frontend/issues/2381

closed 2025-03-03 by Nerivec