Koenkk/zigbee2mqtt #26584: Lidl HG08376 - Only one button works

oxodao, 2025-03-02:
### What happened?

Using the Lidl starter kit's remote (The thing with the gateway + 3 bulb + the remote).

The remote pairs correctly and the "I" button works properly (Albeit a bit weird because it sends "switch_scene" and seems to have an internal "action_scene" counter but meh, thats probably the firmware).

The brightness up / down and the "O" button do not triggers any event in mqtt. Looking in the "exposes" tab of Z2M I only get the message about the switch_scene, 

### What did you expect to happen?

It's physically identical to https://www.zigbee2mqtt.io/devices/FB21-001.html so I guess I should have the other event too that are listed on the device's page.

### How to reproduce it (minimal and precise)

Pair this remote, listen for event on mqtt, press any buttons other than "I" => Nothing

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

EZSPv7 / 6.5.0.0 build 188

### Adapter

The lidl gateway jailbroken with the paulbanks.org tutorial (EZSP though since it can't be updated afaik)

### Setup

z2m / mosquitto / HA in docker in a x86_64 VM on xcp-ng

closed 2025-05-12 by github-actions[bot]

milan0207, 2026-01-08:
I have this exact issue, did you manage to fix it?

andrei-lazarov, 2026-01-08:
@milan0207 The logs you provided are **info**, not **debug**.

The 'log level config' setting enables generation of debug logs, but only info logs are sent to the UI (to avoid MQTT traffic).

You have to view debug logs on the host machine: HA > Add-ons > Z2M > Log, or `docker container logs`.
Disable debug logs when you are done.

milan0207, 2026-01-08:
No problem, thank you for trying to help me.

Connected to the lights directly, and with making an automation on the 'ON' action trigger I achieved the desired function, changing the color temperature from the remote.