Koenkk/zigbee2mqtt #26592: Lack of user support and communication on issues

devkaiwang, 2025-03-03:
### What happened?

Hi,

I would like to raise a concern regarding the lack of responses to user issues, including bug reports, questions, and requests for help. Many issues remain unresolved and are automatically closed after 180 days without any feedback. I personally met switch backlight control issue but seems many of similar issues were not solved and just ignored.
e.g.
My issue: 
https://github.com/Koenkk/zigbee2mqtt/issues/26578

Similar issues:
https://github.com/Koenkk/zigbee2mqtt/issues/23717
https://github.com/Koenkk/zigbee2mqtt/issues/21421
https://github.com/Koenkk/zigbee2mqtt/issues/23859
https://github.com/Koenkk/zigbee2mqtt/issues/24708
https://github.com/Koenkk/zigbee2mqtt/issues/25766

While I understand this is a non-profit project and everyone is busy, it would be greatly appreciated if maintainers could provide explanations for unsupported features, offer guidance, or even share code-level insights. A more user-friendly approach would help address user confusion and improve the project. I would love to continue investigation, debugging and contribute to this project, but I need responses from experienced developers.

Thank you to all maintainers, especially @Koenkk , for your hard work. It is one of the coolest project on this planet. I hope this project continues to grow and improve.

Thanks,
Kai

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

    20240710

### Adapter

zStack3x0

### Setup

RPI-5

MSL-DA, 2025-03-03:
You could start by deleting this issue and creating it under discussions. Just so relevant issues don't get drowned in posts like this.

I understand your frustration. But I think there is good feedback on critical issues and support for new devices.

devkaiwang, 2025-03-04:
Hi @MSL-DA  

Would you please check the issues in my previous comment? 

Obviously, those issues were never got replied and they were just ignored. Issues like this are sadly closed automatically after **180 days** (6 months) without any comment!

sjorge, 2025-03-04:
Aside from time, a large part of the challenge is that @Koenkk and other contributors don’t own all these devices.

Many people contribute patches to add support for a device and then disappear. Without the device in hand, it’s often very difficult to provide meaningful help, which can understandably be frustrating.

Troubleshooting is further complicated because not all devices follow the Zigbee Cluster Library (ZCL). On top of that, manufacturers often don’t support third-party gateways/bridges and rarely provide documentation. This makes adding support—or even debugging—extremely difficult, especially for budget brands like Aqara and Tuya.

Guessing what’s wrong with a device we don’t have access to is nearly impossible. Some kind of indicator on the supported devices page—showing how well a device follows ZCL and whether technical documentation is available—would be very useful. (Personally, I’m willing to pay more for vendors that provide proper documentation, and I avoid others known to cause issues.)

A recent example is the IKEA PARASOLL issue: the main thread is 500+ comments long with no solution, and more threads are scattered around because users don’t always check for existing ones. IKEA is difficult to reach, and when they do respond, they mostly just suggest using their own gateway. At best, they "forward the inquiry" to the right team, but there's rarely any follow-up.

(I won’t name names, but at least one company has even threatened legal action when I asked for documentation after reverse-engineering support for a device. That said, some companies do provide excellent technical documentation, stick to standards, and actively help developers—some even provide hardware for testing.)

In the case of IKEA PARASOLL, we do have logs, packet captures, and some theories, but ultimately, we have no control over the device firmware.

In the end, someone who owns the device will likely need to step up, debug, and work on a fix.

As for Tuya, my experience has been that they don’t follow ZCL properly, which can make support difficult. That said, there are some extra resources that might help with debugging:
- https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html
- https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html
- https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

My best guess is that there are multiple variants of the device, and some may not have the same data points for backlight control. The Finding Tuya Data Points guide might be useful in confirming that.