Koenkk/zigbee2mqtt #26594: [New device support]: IKEA  JETSTROM 6060 JP L2207

yunkeru-Z, 2025-03-03:
### Link

https://www.ikea.com/jp/ja/p/jetstroem-led-ceiling-light-panel-smart-dimmable-white-spectrum-80536050/

### Database entry

{"id":12,"type":"Router","ieeeAddr":"0xd44867fffe047b45","nwkAddr":51871,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"JETSTROM 6060 JP","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096,64599,64636],"outClusterLis t":[25],"clusters":{"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":200,"colorTempPhysicalMax":454}},"genBasic":{"attributes":{"modelId":"JETSTROM 6060 JP","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":8,"appVersion":2,"stackVersion":106,"hwVersion":1,"dateCode":"2 0240202","swBuildId":"2.4.8"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":106,"hwVersion":1,"dateCode":"20240202","swBuildId":"2.4.8" ,"zclVersion":8,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1741012443024}

### Zigbee2MQTT version

 2.1.3 commit: ba337bd

### Comments

It seems that the JP version of JETSTROM 6060 JP is not supported.
Pairing is possible but it says "Not supported."
I don't have much skill, so I would appreciate any advice you could give me.

### External definition

```shell
const m = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['JETSTROM 6060 JP'],
    model: 'JETSTROM 6060 JP',
    vendor: 'IKEA of Sweden',
    description: 'Automatically generated definition',
    extend: [m.light({"colorTemp":{"range":[200,454]}})],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

On/off and color temperature change are functional.

Koenkk, 2025-03-03:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-03-03 by Koenkk

yunkeru-Z, 2025-03-04:
@Koenkk Thank you.

The zigbee-herdsman-converters version in the dev branch is still 23.3.0 and has not been updated to 23.3.1 yet. Should I wait?

Koenkk, 2025-03-04:
Yes, until that pr is merged.