Koenkk/zigbee2mqtt #26597: Tuya TS0601_pir - Haozee PIR sensor - zh:controller:device: Device '0xa4c13812e9be8448' is only compliant to revision '21' of the ZigBee specification (current revision: 23).

JordiSG, 2025-03-03:
### What happened?

First of all, sorry if this is dumb but I'm fairly new to Zigbee2mqtt (and related) and couldn't find any solution while searching in the issues here.
Recently bought https://es.aliexpress.com/item/1005007059504735.html?spm=a2g0o.order_list.order_list_main.17.365a194drcMPwC&gatewayAdapt=glo2esp , once powered on, this links to my zigbee2mqtt instance, but I get the error from the title.

zh:controller:device: Device '0xa4c13812e9be8448' is only compliant to revision '21' of the ZigBee specification (current revision: 23).

Additionally, all exposes are "N/A" and the availability is "disabled"

[image]

[image]

So not sure if someone could help here.

Thank you

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

20240316

### Adapter

SMLIGHT Zigbee 3.0 SLZB-07

### Setup

add-on on home assistant

### Debug log

info 2025-03-03 19:43:05zh:controller: Interview for '0xa4c13812e9be8448' started
info 2025-03-03 19:43:05z2m: Device '0xa4c13812e9be8448' joined
info 2025-03-03 19:43:05z2m: Starting interview of '0xa4c13812e9be8448'
info 2025-03-03 19:43:05z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c13812e9be8448","ieee_address":"0xa4c13812e9be8448"},"type":"device_joined"}'
info 2025-03-03 19:43:05z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c13812e9be8448","ieee_address":"0xa4c13812e9be8448","status":"started"},"type":"device_interview"}'
info 2025-03-03 19:43:05z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c13812e9be8448","ieee_address":"0xa4c13812e9be8448"},"type":"device_announce"}'
info 2025-03-03 19:43:06zh:controller: Succesfully interviewed '0xa4c13812e9be8448'
info 2025-03-03 19:43:06z2m: Successfully interviewed '0xa4c13812e9be8448', device has successfully been paired
info 2025-03-03 19:43:06z2m: Device '0xa4c13812e9be8448' is supported, identified as: Tuya Haozee PIR sensor (TS0601_pir)
info 2025-03-03 19:43:06z2m: Configuring '0xa4c13812e9be8448'
info 2025-03-03 19:43:06z2m:mqtt: MQTT publish: topic 'homeassistant/binary_sensor/0xa4c13812e9be8448/occupancy/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"hw_version":0,"identifiers":["zigbee2mqtt_0xa4c13812e9be8448"],"manufacturer":"Tuya","model":"Haozee PIR sensor","model_id":"TS0601_pir","name":"0xa4c13812e9be8448","sw_version":"3.1.1","via_device":"zigbee2mqtt_bridge_0x00124b0028f55f30"},"device_class":"occupancy","object_id":"0xa4c13812e9be8448_occupancy","origin":{"name":"Zigbee2MQTT","sw":"2.1.3","url":"https://www.zigbee2mqtt.io"},"payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0xa4c13812e9be8448","unique_id":"0xa4c13812e9be8448_occupancy_zigbee2mqtt","value_template":"{{ value_json.occupancy }}"}'
info 2025-03-03 19:43:06z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Haozee PIR sensor","exposes":[{"access":1,"description":"Indicates whether the device detected occupancy","label":"Occupancy","name":"occupancy","property":"occupancy","type":"binary","value_off":false,"value_on":true},{"access":1,"description":"Measured illuminance","label":"Illuminance","name":"illuminance","property":"illuminance","type":"numeric","unit":"lx"},{"access":1,"category":"diagnostic","description":"Remaining battery in %, can take up to 24 hours before reported","label":"Battery","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS0601_pir","options":[{"access":2,"description":"Calibrates the illuminance value (percentual offset), takes into effect on next report of device.","label":"Illuminance calibration","name":"illuminance_calibration","property":"illuminance_calibration","type":"numeric"},{"access":2,"description":"Calibrates the illuminance value (percentual offset), takes into effect on next report of device.","label":"Illuminance calibration","name":"illuminance_calibration","property":"illuminance_calibration","type":"numeric"}],"supports_ota":false,"vendor":"Tuya"},"friendly_name":"0xa4c13812e9be8448","ieee_address":"0xa4c13812e9be8448","status":"successful","supported":true},"type":"device_interview"}'
info 2025-03-03 19:43:06z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0xa4c13812e9be8448/illuminance/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"hw_version":0,"identifiers":["zigbee2mqtt_0xa4c13812e9be8448"],"manufacturer":"Tuya","model":"Haozee PIR sensor","model_id":"TS0601_pir","name":"0xa4c13812e9be8448","sw_version":"3.1.1","via_device":"zigbee2mqtt_bridge_0x00124b0028f55f30"},"device_class":"illuminance","enabled_by_default":true,"object_id":"0xa4c13812e9be8448_illuminance","origin":{"name":"Zigbee2MQTT","sw":"2.1.3","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c13812e9be8448","unique_id":"0xa4c13812e9be8448_illuminance_zigbee2mqtt","unit_of_measurement":"lx","value_template":"{{ value_json.illuminance }}"}'
info 2025-03-03 19:43:06z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0xa4c13812e9be8448/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"hw_version":0,"identifiers":["zigbee2mqtt_0xa4c13812e9be8448"],"manufacturer":"Tuya","model":"Haozee PIR sensor","model_id":"TS0601_pir","name":"0xa4c13812e9be8448","sw_version":"3.1.1","via_device":"zigbee2mqtt_bridge_0x00124b0028f55f30"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","object_id":"0xa4c13812e9be8448_battery","origin":{"name":"Zigbee2MQTT","sw":"2.1.3","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c13812e9be8448","unique_id":"0xa4c13812e9be8448_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
info 2025-03-03 19:43:06z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0xa4c13812e9be8448/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"hw_version":0,"identifiers":["zigbee2mqtt_0xa4c13812e9be8448"],"manufacturer":"Tuya","model":"Haozee PIR sensor","model_id":"TS0601_pir","name":"0xa4c13812e9be8448","sw_version":"3.1.1","via_device":"zigbee2mqtt_bridge_0x00124b0028f55f30"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"Linkquality","object_id":"0xa4c13812e9be8448_linkquality","origin":{"name":"Zigbee2MQTT","sw":"2.1.3","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c13812e9be8448","unique_id":"0xa4c13812e9be8448_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
info 2025-03-03 19:43:07z2m: Successfully configured '0xa4c13812e9be8448'
info 2025-03-03 19:45:22z2m: Interviewing '0xa4c13812e9be8448'
info 2025-03-03 19:45:25zh:controller:device: Device '0xa4c13812e9be8448' is only compliant to revision '21' of the ZigBee specification (current revision: 23).
info 2025-03-03 19:45:33z2m: Successfully interviewed '0xa4c13812e9be8448'
info 2025-03-03 19:45:33z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/interview', payload '{"data":{"id":"0xa4c13812e9be8448"},"status":"ok","transaction":"10ccy-4"}'

Nerivec, 2025-03-05:
The compliance note is only informative, it just means the device is not using a recent ZigBee version.
As long as it doesn't say `pre-21`, you should be fine.

As for the missing data, have you tried changing the batteries, then re-pairing the device completely?

JordiSG, 2025-03-08:
Hello @Nerivec , thanks for your suggestions, but unfortunately didn't help.
I'm powering the device via USB and tried to force remove + re-pairing couple times without any luck on getting more data.
The only data I manage to get is the link quality, rest, stays N/A :S

I found similar threads like - https://github.com/zigpy/zha-device-handlers/issues/3061 but not sure whether I would still need to do this integration (I've tried without much success) with my current z2m version. 

Any idea? Suggestion?

Thanks again

closed 2025-05-16 by github-actions[bot]