Koenkk/zigbee2mqtt #26604: Aqara WXKG12LM - dont paring

digi74, 2025-03-04:
### What happened?

I have 2 new Aqara Buttons here, but neither of them can be paired.
Of course I have tried to keep them awake, but without success.
Current Zigbee2MQTT version (Home Assistant).

2 other Aqara buttons have been paired for a long time, I assume this was the previous version. The new ones have a metal cover under the battery cover that holds the battery in place.

### Zigbee2MQTT version

2.1.3-1

### Adapter firmware version

20241127

### Adapter

SLZB-06M

### Setup

Home Assistant OS in Proxmox KVM

Nerivec, 2025-03-05:
Can you provide `debug` logs of the pairing?

digi74, 2025-03-06:
I have tried to connect an Aqara T1 door contact that is lying around, but this does not work either.

Nothing appears in the log.
Neither with Aqara door contact, nor Aqara Button.

However, the Aqara Buttons that have already been integrated work perfectly.

I have just tested whether the pairing does not work at all with a Philips Hue Dimmer Switch, which works right away, flawlessly.

It looks like I can't provide a helpful debug log of the pairing process for the Aqara devices.

Nerivec, 2025-03-06:
Aqara is know for causing pairing troubles (they don't follow proper ZigBee spec, they just don't want you using their devices without also buying their hub...). Allowing them to join requires workarounds on the coordinator, but they also are picky about which routers they allow. And they don't have a proper meshing system, they won't switch to another router even if it becomes necessary...

Can you try whatever procedure that device has to fully reset it to factory default, then try pairing it again. You may want to permit join on specific devices one at a time, see if you find one that works.

digi74, 2025-03-07:
Thanks for your answer and help!

I have now gone very close to the Zigbee antenna and the pairing now works (with reset).
Funnily enough, I tried exactly the same thing 2 days earlier (without success). Well, my problem has been solved.

closed 2025-03-07 by Nerivec