Koenkk/zigbee2mqtt #26606: Aqara T1 door & window contact sensor N/A

theSC, 2025-03-04:
### What happened?

When adding three brand new Aqara T1 door & window contact sensor I only get NA for Contact, Battery and Voltage.  This makes the devices unusable.

### What did you expect to happen?

I expected it to tell me if the contact is open or closed

### Zigbee2MQTT version

2.1.3-1

### Adapter firmware version

20210708

### Adapter

Sonoff Zigbee 3.0 USB dongle

### Setup

Home Assistant OS in a VM

theSC, 2025-03-04:
I have re-added them so many times I have lost count. Never any improvement

theSC, 2025-03-04:
Log from adding one of the sensors:

info 2025-03-04 13:53:26z2m:mqtt: MQTT publish: topic 'homeassistant/binary_sensor/0x54ef441000db6a08/contact/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"hw_version":0,"identifiers":["zigbee2mqtt_0x54ef441000db6a08"],"manufacturer":"Aqara","model":"Door and window sensor E1","model_id":"MCCGQ14LM","name":"0x54ef441000db6a08","via_device":"zigbee2mqtt_bridge_0x00124b0024c9b3fb"},"device_class":"door","object_id":"0x54ef441000db6a08_contact","origin":{"name":"Zigbee2MQTT","sw":"2.1.3","url":"https://www.zigbee2mqtt.io"},"payload_off":true,"payload_on":false,"state_topic":"zigbee2mqtt/0x54ef441000db6a08","unique_id":"0x54ef441000db6a08_contact_zigbee2mqtt","value_template":"{{ value_json.contact }}"}'
info 2025-03-04 13:53:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Door and window sensor E1","exposes":[{"access":1,"description":"Indicates if the contact is closed (= true) or open (= false)","label":"Contact","name":"contact","property":"contact","type":"binary","value_off":true,"value_on":false},{"access":1,"category":"diagnostic","description":"Remaining battery in %, can take up to 24 hours before reported","label":"Battery","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"category":"diagnostic","description":"Indicates if the battery of this device is almost empty","label":"Battery low","name":"battery_low","property":"battery_low","type":"binary","value_off":false,"value_on":true},{"access":1,"category":"diagnostic","description":"Voltage of the battery in millivolts","label":"Voltage","name":"voltage","property":"voltage","type":"numeric","unit":"mV"},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"MCCGQ14LM","options":[],"supports_ota":false,"vendor":"Aqara"},"friendly_name":"0x54ef441000db6a08","ieee_address":"0x54ef441000db6a08","status":"successful","supported":true},"type":"device_interview"}'
info 2025-03-04 13:53:26z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x54ef441000db6a08/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"hw_version":0,"identifiers":["zigbee2mqtt_0x54ef441000db6a08"],"manufacturer":"Aqara","model":"Door and window sensor E1","model_id":"MCCGQ14LM","name":"0x54ef441000db6a08","via_device":"zigbee2mqtt_bridge_0x00124b0024c9b3fb"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","object_id":"0x54ef441000db6a08_battery","origin":{"name":"Zigbee2MQTT","sw":"2.1.3","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/0x54ef441000db6a08","unique_id":"0x54ef441000db6a08_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
info 2025-03-04 13:53:26z2m:mqtt: MQTT publish: topic 'homeassistant/binary_sensor/0x54ef441000db6a08/battery_low/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"hw_version":0,"identifiers":["zigbee2mqtt_0x54ef441000db6a08"],"manufacturer":"Aqara","model":"Door and window sensor E1","model_id":"MCCGQ14LM","name":"0x54ef441000db6a08","via_device":"zigbee2mqtt_bridge_0x00124b0024c9b3fb"},"device_class":"battery","entity_category":"diagnostic","object_id":"0x54ef441000db6a08_battery_low","origin":{"name":"Zigbee2MQTT","sw":"2.1.3","url":"https://www.zigbee2mqtt.io"},"payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0x54ef441000db6a08","unique_id":"0x54ef441000db6a08_battery_low_zigbee2mqtt","value_template":"{{ value_json.battery_low }}"}'
info 2025-03-04 13:53:27z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x54ef441000db6a08/voltage/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"hw_version":0,"identifiers":["zigbee2mqtt_0x54ef441000db6a08"],"manufacturer":"Aqara","model":"Door and window sensor E1","model_id":"MCCGQ14LM","name":"0x54ef441000db6a08","via_device":"zigbee2mqtt_bridge_0x00124b0024c9b3fb"},"device_class":"voltage","enabled_by_default":false,"entity_category":"diagnostic","object_id":"0x54ef441000db6a08_voltage","origin":{"name":"Zigbee2MQTT","sw":"2.1.3","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/0x54ef441000db6a08","unique_id":"0x54ef441000db6a08_voltage_zigbee2mqtt","unit_of_measurement":"mV","value_template":"{{ value_json.voltage }}"}'
info 2025-03-04 13:53:27z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x54ef441000db6a08/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"hw_version":0,"identifiers":["zigbee2mqtt_0x54ef441000db6a08"],"manufacturer":"Aqara","model":"Door and window sensor E1","model_id":"MCCGQ14LM","name":"0x54ef441000db6a08","via_device":"zigbee2mqtt_bridge_0x00124b0024c9b3fb"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"Linkquality","object_id":"0x54ef441000db6a08_linkquality","origin":{"name":"Zigbee2MQTT","sw":"2.1.3","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/0x54ef441000db6a08","unique_id":"0x54ef441000db6a08_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'

theSC, 2025-03-04:
[image]

captainlettuce, 2025-03-05:
Sounds like [this](https://github.com/Koenkk/zigbee2mqtt/issues/20740), try updating the adapter firmware.

theSC, 2025-03-06:
You are a god send! This fixed it! Thank you so much!

closed 2025-03-06 by theSC