Koenkk/zigbee2mqtt #26615: [New device support]: Human presence sensor not supported

marcodcollines, 2025-03-05:
### Link

https://a.aliexpress.com/_EJQmSli

### Database entry

{"id":37,"type":"Router","ieeeAddr":"0xa4c1389c09ed4afa","nwkAddr":755,"manufId":4417,"manufName":"_TZE284_iadro9bf","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0,60672],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\t\u0000\u0000\u0000e\n\u0000\u0000\u0000\u0012\u001a\u0000\u0000\u0000e\u001b\u0000\u0000\u0000\u0012�;[/f�;[/\u0012","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":78,"modelId":"TS0601","manufacturerName":"_TZE284_iadro9bf","powerSource":1,"zclVersion":3,"hwVersion":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":78,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1741190493887}

### Zigbee2MQTT version

2.1.3

### Comments

This device is not supported in the front end. I don't know what to do. It is the first time a create an issue.

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE284_iadro9bf',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

It doesn't work at all.

Koenkk, 2025-03-06:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-03-06 by Koenkk

XAV-01, 2025-03-08:
Hello

This device https://fr.aliexpress.com/item/1005008084934603.html is recognized like the same as this one https://a.aliexpress.com/_EJQmSli
Manufacture : _TZE284_iadro9bf, id:4417

```json
{"id":11,"type":"Router","ieeeAddr":"0xa4c138f6525dda28","nwkAddr":42728,"manufId":4417,"manufName":"_TZE284_iadro9bf","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0,60672],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"W�]/if�]/i","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":78,"modelId":"TS0601","manufacturerName":"_TZE284_iadro9bf","powerSource":1,"zclVersion":3,"hwVersion":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":78,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1741420851717}
```

When using latest-dev tag z2m recongnized the device but it doesn't detect presence.

Get from device gives
```json
{"detection_delay":0,"fading_time":30.5,"linkquality":216,"maximum_range":5.35,"minimum_range":0,"presence":false,"radar_sensitivity":6}
```

May it be that the device is malfunctioning?
Best regards

AlexHidalgo11, 2025-03-12:
Same issue here, I updated repo to latest-dev, sensor is now correctly identified but no data displayed.

[image]

XAV-01, 2025-03-12:
« Presence » is the only value  set to false or true

- The value false to presence attribute indicates that the device detects presence
- The value true to presence attribute indicates that the device does not detect presence

There is no value for illuminance or target distance

Maybe the definition for this maker should be apdated, if I only knew how ….

AlexHidalgo11, 2025-03-12:
I've got the Payload log when pairing the device. 
Does it helps to define how to adapt it?
Thanks for your help!

XAV-01, 2025-03-12:
From what I see you device is working properly 
Presence false means the device detect a human presence

JaCoLJcL, 2025-03-14:
is there any method to add this device without go to dev/edge ? For example: generated_external_definition ?