Koenkk/zigbee2mqtt #26616: Installation fails with latest pnpm (works only with 10.4.1)

dawiinci, 2025-03-05:
### What happened?

Installation of 2.1.3 fails:

` ERROR  Failed to switch pnpm to v10.4.1. Looks like pnpm CLI is missing at "/Users/server/Library/pnpm/.tools/pnpm/10.4.1/bin" or is incorrect
`
Fixed by installing old version: 
`npm install -g pnpm@10.4.1
`
macOS with M1

This should not happen I guess?

### What did you expect to happen?

It should install with the latest version of pnpm 10.5.2 which was released last week.

### How to reproduce it (minimal and precise)

Try installing with pnpm 10.5.2.

My script do do so:

```
git clone --depth 1 https://github.com/Koenkk/zigbee2mqtt.git
pip install --upgrade pip wheel setuptools
pip install nodeenv
nodeenv -p -n lts
npm install -g pnpm
pnpm setup
pnpm add -g pnpm
pnpm self-update

pnpm install --frozen-lockfile
```

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

0

### Adapter

0

### Setup

macOS M1

Koenkk, 2025-03-06:
The `pnpm` version is now locked in the `package.json`, could you try installing it `corepack install` in the z2m dir?

dawiinci, 2025-03-09:
It works now, but it seems to do nothing as of yet. Probably because the latest version is now required and already installed?

`corepack install
No changes to the environment were made. Everything is already up to date.
 ERR_PNPM_GLOBAL_PNPM_INSTALL  Use the "pnpm self-update" command to install or update pnpm
Nothing to stop. No server is running for the store at /Users/server/Library/pnpm/store/v10
The currently active pnpm v10.6.1 is already "latest" and doesn't need an update
Lockfile is up to date, resolution step is skipped`

This is the change of code I have made:

```
git clone --depth 1 https://github.com/Koenkk/zigbee2mqtt.git
pip install --upgrade pip wheel setuptools
pip install nodeenv
nodeenv -p -n lts
npm install -g pnpm
corepack install
pnpm setup
pnpm add -g pnpm
pnpm self-update

pnpm install --frozen-lockfile
```

Koenkk, 2025-03-09:
The `pnpm self-update` shouldn't be needed

closed 2025-03-25 by dawiinci