Koenkk/zigbee2mqtt #26619: Tuya TS0026 being added as TS0726

jawn22, 2025-03-05:
### What happened?

I've added a Tuya TS0026 (4 gang+2 scene/6 gang) however it's being added as a TS0726 (4 gang). It's only exposing 4 switches rather than 6

### What did you expect to happen?

I expected it to be added as TS0026 (4 gang+2 scene)

### How to reproduce it (minimal and precise)

Add to Z2M

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

 7.4.5

### Adapter

EmberZNet - SLZB-06M

### Setup

Add-on on home assistant OS running on intel NUC

hyteplus, 2025-03-13:
The same issues, "TS0726", ["_TZ3000_qhyadm57"]) ,it works on Zigbee2MQTT version 1.39.0 with ext_converter.js, can't works on Zigbee2MQTT version 2.1.3

[https://github.com/Koenkk/zigbee-herdsman-converters/issues/8967](url)

[https://github.com/Koenkk/zigbee-herdsman-converters/commit/e1dbf5b04f2af347902c6b1a6b5a382063b4fe81](url)

jawn22, 2025-03-17:
Great thanks - so should be available in main release on 1st April?

jawn22, 2025-03-17:
by the way @hyteplus Where did you buy your TS0026's from? I got some from a seller in the UK but they were crazy expensive...

closed 2025-03-17 by jawn22

hyteplus, 2025-03-18:
> by the way [@hyteplus](https://github.com/hyteplus) Where did you buy your TS0026's from? I got some from a seller in the UK but they were crazy expensive...

I bought them in bulk through a friend's company in China. It was a bit pricey, but the product's reliability is solid. The finished switches have a great tactile feel, and they really hold up to the test of time visually too