Koenkk/zigbee2mqtt #26621: Philips Hue Remote - stuck in "update_available" loop

Eniste1, 2025-03-06:
### What happened?

Hi,
since some days I have the following behaviour in my system:

(Hue Dimmer Switch [Philips 324131092621](https://www.zigbee2mqtt.io/devices/324131092621.html#philips-324131092621))
Firmware: 6.1.1.28573

Since a few days I try to update the dimmer switch a few times, the update progress takes about 46 minutes and ends without an error but the the firmware is still the old one.
And the next day I get the update information again.

a few weeks ago I already have the same behaviour and I try to update the same device a few times and suddenly it seems to be o.k. but now same Firmware as before and the Z2MQTT tells me every day, that a new update is available.

No solution for me now.
If you need further information don't hesitate to contact me.

Hopefully this could be solved soon.

Kind regards and thank you for your great work
Ralph

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

20220219

### Adapter

ZigStar Stick v4

### Setup

Add On in HomeAssistant

JasonAvg, 2025-03-06:
I have the same issue

Eniste1, 2025-03-06:
a few members of the HomeAssistant Community reports the same issue:

https://community.home-assistant.io/t/zigbee2mqtt-ota-updates-failing/842906/24

Eniste1, 2025-04-03:
Same issue after updating to:

Zigbee2MQTT Version: 2.2.0
HomeAssistant Core: 2025.4.0
Supervisor: 2025.03.4
Operation System 15.1

malgoe, 2025-05-14:
Same issue here.
Z2M version 2.3.0 on docker.
The affected devices are these: https://www.zigbee2mqtt.io/devices/324131092621.html
My controller is a sonoff ZBDongle-E using Ember.

Z2M claims there is an OTA for these remotes but seems to update to the same version they already have. (Build date: 20160302, firmware version 5.45.1.17846) After updating after a day or so it claims they can be updated again, but I guess that is due to the devices checking in looking for updates.

closed 2025-07-21 by github-actions[bot]