Koenkk/zigbee2mqtt #26622: [Feature request]: expose "disabled" setting to home assistant

spali, 2025-03-06:
### Is your feature request related to a problem? Please describe

Try to create a home assistant automation to identify all offline devices. It's based on last_seen. But I would like to ignore all disabled device that are reserve or just sit in a box. I would prefer not to remove them from z2m for keeping an "inventory" overview of all devices and not have to re-pair them when getting used again. Also would like to just ignore some devices seasonal.

### Describe the solution you'd like

Maintaining only the z2m disabled setting would be perfect, as I wouldn't have to maintain also another list in home assistant for ignoring devices. The setting already exists, but can't be used (at least don't know how) in home assistant.
For me it doesn't matter if exposed as "disabled" entity or just as an attribute in the entities or only in the last_seen entity. As long as it is accessible within templating of home assistant.

### Describe alternatives you've considered

Asked already for a potential workaround in #26449, but without success and also didn't found a way by myself.

### Additional context

I think the only thing would be an external extension, but as this may be simple and standard use-case, I think it found fit better into the existing HomeAssistant extension, as it would add only a small change.

spali, 2025-03-14:
Found at least a workaround for now: https://github.com/Koenkk/zigbee2mqtt/discussions/26449#discussioncomment-12502664, but would prefer to have it supported.

closed 2025-05-21 by github-actions[bot]

mattg2, 2025-05-30:
+1

Would be great to have an indicator in the payload. I vote for [availability = {"state": "disabled"}]