Koenkk/zigbee2mqtt #26623: Müller Licht 404006/404008/404004 no longer report color_temp to HA

waclaw66, 2025-03-06:
### What happened?

https://www.zigbee2mqtt.io/devices/404006_404008_404004.html stopped showing value of `color_temp` attribute, (it's always `null`) after update HA to 2025.3.
However MQTT message contains proper `color_temp` value. Color temperature can be set precisely via HA color temp control, but not via `color_temp` sensor attribute as I did within automations.

Z2M 2.1.3
HA 2025.3.0

light/config

```
availability:
  - topic: zigbee2mqtt/bridge/state
    value_template: '{{ value_json.state }}'
brightness: true
brightness_scale: 254
command_topic: zigbee2mqtt/bulb/set
device:
  hw_version: 1
  identifiers:
    - zigbee2mqtt_0x00158d0002fddfdc
  manufacturer: Müller Licht
  model: Tint LED bulb GU10/E14/E27 350/470/806 lumen, dimmable, opal white
  model_id: 404006/404008/404004
  sw_version: '2.0'
  via_device: zigbee2mqtt_bridge_0x00124b0021b7800c
  name: bulb
effect: true
effect_list:
  - blink
  - breathe
  - okay
  - channel_change
  - finish_effect
  - stop_effect
max_mireds: 370
min_mireds: 153
object_id: bulb
origin:
  sw_version: 2.1.3
  name: Zigbee2MQTT
  support_url: https://www.zigbee2mqtt.io
schema: json
state_topic: zigbee2mqtt/bulb
supported_color_modes:
  - color_temp
unique_id: 0x00158d0002fddfdc_light_zigbee2mqtt
name: null
```

### What did you expect to happen?

Sensor `color_temp` attribute shall have current value of color temperature as for other bulb models.

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

20240710

### Adapter

ZBDongle-P

### Setup

docker

frankebobdk, 2025-03-06:
I believe I’m experiencing the same issue with my Philips Hue White Ambiance and Filament bulbs.

Here’s a link to the White Ambiance bulb in question: [Philips 929001953301](https://www.zigbee2mqtt.io/devices/929001953301.html#philips-929001953301).

To regain control over the colors through **Home Assistant**, I downgraded  to version 2025.2.5.

However, my main issue is related to groups, specifically with the White Ambiance bulbs.

chris-1243, 2025-03-06:
This affects White spectrum light only. All my RGBW lights are reporting `color_temp` correctly. On the other hand, white spectrum does not.

srnoth, 2025-03-07:
Same issue across multiple vendors of CCT zigbee lights. 

> Came here looking for similar with some IKEA bulbs as they are now spamming logs with a few issues.
> 
> > Got `color_temp` argument in `turn_on` service, which is deprecated and will break in Home Assistant 2026.1, please use `color_temp_kelvin` argument
> 
> when changing colour and lots of errors due to powercalc trying to read the `color_temp_kelvin` attribute which is null.
> 
> Underlying cause is https://developers.home-assistant.io/blog/2024/12/14/kelvin-preferred-color-temperature-unit/
> 
> Think this effects all white spectrum lights

Interestingly devs were told this would not be a breaking change, so decided to leave Z2M in mireds.
https://github.com/Koenkk/zigbee2mqtt/discussions/25834

I am assuming the issue is related to this breaking change in 2025.3.0:
https://github.com/home-assistant/core/pull/136996

R3Cardoso, 2025-03-07:
Same thing here, using White Ambiance Philips Hue lights and all are reporting color_temp: Null on HA.
Happen after update HA to 2025.3

dasadi, 2025-03-07:
Same for me with IKEA and Philipps Hue CCT bulbs. 

The problem seems, that the light reports: `supported_color_modes: color_temp` in discovery. Furthermore the json state payload contains the field `color_mode: color_temp` which from HA 2025.3 on is deprecated. 
To me it seems like CCT-only lights should from now on not send any `color_mode` -value and should not advertise any `supported_color_modes` at all, but only set the `color_temp` field in the json.

jamesonuk, 2025-03-07:
The more I look at this the more I think it is a HA bug.   I have raised https://github.com/home-assistant/core/issues/140072

chris-1243, 2025-03-08:
It should be fixed in upcoming release of HA

https://github.com/home-assistant/core/pull/140113

Koenkk, 2025-03-09:
Since this is not a Z2M issue I will close it, thanks all!

closed 2025-03-09 by Koenkk