Koenkk/zigbee2mqtt #26624: Lost network socket connection without recovery by watchdog.

weekends, 2025-03-06:
### What happened?

I'm have 2 installation in different places HA+z2m. Use 2 different adapters, one is ZigStarGW with FW: 20240914 second SLZB-06 with FW: v2.8.2.dev3. Bought of them use zigbee fw: 20240710. After update to 2.1.x version i'm has trouble once in 2-4 days on bought configurations:

2025-03-06 13:12:14] error: 	zh:zstack:znp: Socket error Error: read ECONNRESET
[2025-03-06 13:12:14] error: 	z2m: Adapter disconnected, stopping
[2025-03-06 13:12:14] error: 	z2m: MQTT error: write ECONNRESET

WatchDog can't help, more then 1:30 hour nothing change. Only manual z2m restart in HA restore work.

### What did you expect to happen?

Updating to 2.1.x z2m version. Bought configuration work fine with out any issues near 1 year before update.

### How to reproduce it (minimal and precise)

On my side - simple - restart z2m and wait 2-4 days, till z2m crashed.

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

20240710

### Adapter

ZigStarGW, SLZB-06

### Setup

Add-on on HomeAssistant

### Debug log

I'm add debug log later. Now turn it on and wait for crash.

Nerivec, 2025-03-06:
Since both MQTT and the adapter lose connection, looks like you are having issues with your network.
You should start looking there.
Any configuration changes in your network, updates, etc., anything that may trigger at a specific interval (2-4 days) that could be crowding the network...
Try restarting your modem/router, check firewalls, antivirus, etc.

weekends, 2025-03-07:
There are no any changes in network configuration. On one of the configurations, network completely separated only for automation. All IP's in network static, ping's and web access to adapter work with out any issues. Socket also accessed when z2m wrote, that can't.

weekends, 2025-03-13:
[2025-03-04 22:52:11] .[32minfo.[39m:   z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Shuffle', payload '{"countdown":0,"linkquality":109,"power_on_behavior":"off","state":"OFF","switch
_type":"toggle"}'
[2025-03-05 00:12:59] .[31merror.[39m:  zh:zstack:znp: Socket error Error: read ECONNRESET
[2025-03-05 00:12:59] .[31merror.[39m:  z2m: Adapter disconnected, stopping
WATCHDOG: Waiting 0.03333333333333333min before next start try.
Starting Zigbee2MQTT with watchdog (2000,60000,300000,900000,1800000,3600000).
[2025-03-05 14:22:58] .[31merror.[39m:  zh:zstack:znp: Socket error Error: read ECONNRESET
[2025-03-05 14:22:58] .[31merror.[39m:  z2m: Adapter disconnected, stopping
WATCHDOG: Waiting 0.03333333333333333min before next start try.
Starting Zigbee2MQTT with watchdog (2000,60000,300000,900000,1800000,3600000).
[2025-03-06 10:47:39] .[31merror.[39m:  zh:zstack:znp: Socket error Error: read ECONNRESET
[2025-03-06 10:47:39] .[31merror.[39m:  z2m: Adapter disconnected, stopping
WATCHDOG: Waiting 0.03333333333333333min before next start try.
Starting Zigbee2MQTT with watchdog (2000,60000,300000,900000,1800000,3600000).
[2025-03-07 21:01:03] .[31merror.[39m:  zh:zstack:znp: Socket error Error: read ECONNRESET
[2025-03-07 21:01:03] .[31merror.[39m:  z2m: Adapter disconnected, stopping
WATCHDOG: Waiting 0.03333333333333333min before next start try.
Starting Zigbee2MQTT with watchdog (2000,60000,300000,900000,1800000,3600000).
[2025-03-10 12:01:11] .[31merror.[39m:  zh:zstack:znp: Socket error Error: read ECONNRESET
[2025-03-10 12:01:11] .[31merror.[39m:  z2m: Adapter disconnected, stopping
WATCHDOG: Waiting 0.03333333333333333min before next start try.
Starting Zigbee2MQTT with watchdog (2000,60000,300000,900000,1800000,3600000).
[2025-03-11 13:12:05] .[31merror.[39m:  zh:zstack:znp: Socket error Error: read ECONNRESET
[2025-03-11 13:12:05] .[31merror.[39m:  z2m: Adapter disconnected, stopping
WATCHDOG: Waiting 0.03333333333333333min before next start try.
Starting Zigbee2MQTT with watchdog (2000,60000,300000,900000,1800000,3600000).
[2025-03-13 03:14:23] .[31merror.[39m:  zh:zstack:znp: Socket error Error: read ETIMEDOUT
[2025-03-13 03:14:23] .[31merror.[39m:  z2m: Adapter disconnected, stopping
WATCHDOG: Waiting 0.03333333333333333min before next start try.
Starting Zigbee2MQTT with watchdog (2000,60000,300000,900000,1800000,3600000).
[2025-03-13 03:14:28] .[31merror.[39m:  zh:zstack:znp: Socket error Error: connect EHOSTUNREACH 192.168.8.7:6638
[2025-03-13 03:14:28] .[31merror.[39m:  z2m: Error while starting zigbee-herdsman
[2025-03-13 03:14:28] .[31merror.[39m:  z2m: Failed to start zigbee-herdsman
[2025-03-13 03:14:28] .[31merror.[39m:  z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-03-13 03:14:28] .[31merror.[39m:  z2m: Exiting...
[2025-03-13 03:14:28] .[31merror.[39m:  z2m: Error: Error while opening socket
    at Socket.<anonymous> (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:158:24)
    at Socket.emit (node:events:536:35)
    at emitErrorNT (node:internal/streams/destroy:170:8)
    at emitErrorCloseNT (node:internal/streams/destroy:129:3)
    at processTicksAndRejections (node:internal/process/task_queues:90:21)
WATCHDOG: Waiting 1min before next start try.
Starting Zigbee2MQTT with watchdog (2000,60000,300000,900000,1800000,3600000).
[2025-03-13 03:15:29] .[31merror.[39m:  zh:zstack:znp: Socket error Error: connect EHOSTUNREACH 192.168.8.7:6638
[2025-03-13 03:15:29] .[31merror.[39m:  z2m: Error while starting zigbee-herdsman
[2025-03-13 03:15:29] .[31merror.[39m:  z2m: Failed to start zigbee-herdsman
[2025-03-13 03:15:29] .[31merror.[39m:  z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-03-13 03:15:29] .[31merror.[39m:  z2m: Exiting...
[2025-03-13 03:15:29] .[31merror.[39m:  z2m: Error: Error while opening socket
    at Socket.<anonymous> (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:158:24)
    at Socket.emit (node:events:536:35)
    at emitErrorNT (node:internal/streams/destroy:170:8)
    at emitErrorCloseNT (node:internal/streams/destroy:129:3)
    at processTicksAndRejections (node:internal/process/task_queues:90:21)

Last WATCHDOG record's reply much time

[05:06:47] INFO: .[32mPreparing to start....[0m
[05:06:48] INFO: .[32mSocat not enabled.[0m
[05:06:48] INFO: .[32mEnabled Zigbee2MQTT watchdog with value 'default'.[0m
[05:06:48] INFO: .[32mStarting Zigbee2MQTT....[0m
Starting Zigbee2MQTT with watchdog (2000,60000,300000,900000,1800000,3600000).
[2025-03-13 05:06:54] .[31merror.[39m:  zh:zstack:znp: Socket error Error: connect EHOSTUNREACH 192.168.8.7:6638
[2025-03-13 05:06:54] .[31merror.[39m:  z2m: Error while starting zigbee-herdsman
[2025-03-13 05:06:54] .[31merror.[39m:  z2m: Failed to start zigbee-herdsman
[2025-03-13 05:06:54] .[31merror.[39m:  z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-03-13 05:06:54] .[31merror.[39m:  z2m: Exiting...
[2025-03-13 05:06:54] .[31merror.[39m:  z2m: Error: Error while opening socket
    at Socket.<anonymous> (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:158:24)
    at Socket.emit (node:events:536:35)
[... cut ...]
[2025-03-13 05:07:12] .[31merror.[39m:  z2m: Error: Error while opening socket
    at Socket.<anonymous> (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:158:24)
    at Socket.emit (node:events:536:35)
    at emitErrorNT (node:internal/streams/destroy:170:8)
    at emitErrorCloseNT (node:internal/streams/destroy:129:3)
    at processTicksAndRejections (node:internal/process/task_queues:90:21)

After that service stoped till manually start it again.

Nerivec, 2025-03-13:
Yes, it definitely appears to be a network problem, it can't reach your adapter (as if not even connected)
`EHOSTUNREACH 192.168.8.7:6638`
So, it seems either your adapter is crashing and not restarting properly (firmware issue?), or some misconfiguration in your network (firewall, subnet, etc.) is causing problems.

Nerivec, 2025-03-13:
Can you check the HA logs `Settings>System>Logs` (there's a dropdown in top right to switch entity, check the first 4), see if you find anything throwing errors around the time of the Z2M **first** failure (before the first watchdog trigger)?

weekends, 2025-03-13:
@Nerivec currently no. I'm wait for next crash and then collect that log's.

SuperManth, 2025-04-01:
Hi there, I got a similar issue with this device. I'm using Iobroker with Zigbee-Adapter and SLZB-06. Of cause you are right it starts with a network error. So the connection has been lost, but it can not reconnect. It stocks in an endless loop of retries. If this happens I can connect to the web interface of the device over the network. Then I have to stop and start the Zigbee-Adapter. The connection established in a second successful.

weekends, 2025-05-27:
On my side bug reproduced time-to time, "Allow multi-threaded socket connection" enabled for SLZB-06, watchdog also enabled in z2m. Network work fine, mqtt broker accessed, but connection lost and recovered only by manual restart of z2m:

[2025-05-13 02:17:24] error: 	z2m: MQTT error: read ECONNRESET
[2025-05-13 02:17:29] error: 	zh:zstack:znp: Socket error Error: read ECONNRESET
[2025-05-13 02:17:29] error: 	z2m: Adapter disconnected, stopping
WATCHDOG: Waiting 0.03333333333333333min before next start try.
Starting Zigbee2MQTT with watchdog (2000,60000,300000,900000,1800000,3600000).
[2025-05-14 00:53:27] error: 	zh:zstack:znp: Socket error Error: read ECONNRESET
[2025-05-14 00:53:27] error: 	z2m: Adapter disconnected, stopping
[2025-05-14 00:53:27] error: 	z2m: MQTT error: write ECONNRESET
[01:34:44] INFO: Preparing to start...
[01:34:44] INFO: Socat not enabled
[01:34:44] INFO: Enabled Zigbee2MQTT watchdog with value 'default'
[01:34:44] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT with watchdog (2000,60000,300000,900000,1800000,3600000).
[2025-05-26 21:32:53] error: 	zh:zstack:znp: Socket error Error: read ECONNRESET
[2025-05-26 21:32:53] error: 	z2m: Adapter disconnected, stopping
[2025-05-26 21:32:53] error: 	z2m: MQTT error: write ECONNRESET
[23:32:54] INFO: Preparing to start...
[23:32:54] INFO: Socat not enabled
[23:32:54] INFO: Enabled Zigbee2MQTT watchdog with value 'default'
[23:32:54] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT with watchdog (2000,60000,300000,900000,1800000,3600000).
Starting Zigbee2MQTT with watchdog (2000,60000,300000,900000,1800000,3600000).

closed 2025-08-04 by github-actions[bot]