Koenkk/zigbee2mqtt #26639: [New device support]: Haozee Human Presence Sensor Light Switch (NEO NAS-PS10B2) (_TZE204_1youk3hj)

oapy123, 2025-03-06:
### Link

https://www.amazon.com/dp/B0DNF4KK5G

### Database entry

{"id":27,"type":"Router","ieeeAddr":"0xa4c1380b136e2bee","nwkAddr":17551,"manufId":4417,"manufName":"_TZE204_1youk3hj","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZE204_1youk3hj","zclVersion":3,"appVersion":74,"modelId":"TS0601","powerSource":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1946894323},"lastSeen":1741303363887}

### Zigbee2MQTT version

2.1.3

### Comments

**Pairing Logs**
`info 2025-03-06 18:22:11z2m: Zigbee: allowing new devices to join.
info 2025-03-06 18:22:11z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254},"status":"ok","transaction":"lk1la-18"}'
info 2025-03-06 18:22:42zh:controller: Interview for '0xa4c1380b136e2bee' started
info 2025-03-06 18:22:42z2m: Device '0xa4c1380b136e2bee' joined
info 2025-03-06 18:22:42z2m: Starting interview of '0xa4c1380b136e2bee'
info 2025-03-06 18:22:42z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1380b136e2bee","ieee_address":"0xa4c1380b136e2bee"},"type":"device_joined"}'
info 2025-03-06 18:22:42z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1380b136e2bee","ieee_address":"0xa4c1380b136e2bee","status":"started"},"type":"device_interview"}'
info 2025-03-06 18:22:43zh:controller: Succesfully interviewed '0xa4c1380b136e2bee'
info 2025-03-06 18:22:43z2m: Successfully interviewed '0xa4c1380b136e2bee', device has successfully been paired
info 2025-03-06 18:22:43z2m: Device '0xa4c1380b136e2bee' is supported, identified as: NEO Human presence sensor (NAS-PS10B2)
info 2025-03-06 18:22:43z2m: Configuring '0xa4c1380b136e2bee'
info 2025-03-06 18:22:43z2m:mqtt: MQTT publish: topic 'homeassistant/binary_sensor/0xa4c1380b136e2bee/presence/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"hw_version":1,"identifiers":["zigbee2mqtt_0xa4c1380b136e2bee"],"manufacturer":"NEO","model":"Human presence sensor","model_id":"NAS-PS10B2","name":"0xa4c1380b136e2bee","via_device":"zigbee2mqtt_bridge_0x048727fffe1bfc06"},"device_class":"presence","object_id":"0xa4c1380b136e2bee_presence","origin":{"name":"Zigbee2MQTT","sw":"2.1.3","url":"https://www.zigbee2mqtt.io"},"payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0xa4c1380b136e2bee","unique_id":"0xa4c1380b136e2bee_presence_zigbee2mqtt","value_template":"{{ value_json.presence }}"}'
info 2025-03-06 18:22:43z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Human presence sensor","exposes":[{"access":1,"description":"Indicates whether the device detected presence","label":"Presence","name":"presence","property":"presence","type":"binary","value_off":false,"value_on":true},{"access":1,"description":"Human Motion State","label":"Human motion state","name":"human_motion_state","property":"human_motion_state","type":"enum","values":["none","small","large"]},{"access":1,"description":"Current distance of detected motion","label":"Current distance","name":"dis_current","property":"dis_current","type":"numeric","unit":"cm","value_max":600,"value_min":0,"value_step":1},{"access":3,"description":"Presence Time","label":"Presence time","name":"presence_time","property":"presence_time","type":"numeric","unit":"s","value_max":600,"value_min":3,"value_step":1},{"access":3,"description":"Motion Range Detection","label":"Motion far detection","name":"motion_far_detection","property":"motion_far_detection","type":"numeric","unit":"cm","value_max":600,"value_min":150,"value_step":75},{"access":3,"description":"Motion Detection Sensitivity","label":"Motion sensitivity value","name":"motion_sensitivity_value","property":"motion_sensitivity_value","type":"numeric","value_max":7,"value_min":0,"value_step":1},{"access":3,"description":"Motionless Detection Sensitivity","label":"Motionless sensitivity","name":"motionless_sensitivity","property":"motionless_sensitivity","type":"numeric","value_max":7,"value_min":0,"value_step":1},{"access":3,"description":"Work Mode","label":"Work mode","name":"work_mode","property":"work_mode","type":"enum","values":["manual","auto"]},{"access":3,"description":"Output Switch","label":"Output switch","name":"output_switch","property":"output_switch","type":"binary","value_off":"OFF","value_on":"ON"},{"access":3,"description":"Output Times","label":"Output time","name":"output_time","property":"output_time","type":"numeric","unit":"s","value_max":1800,"value_min":10},{"access":3,"description":"Led Switch","label":"Led switch","name":"led_switch","property":"led_switch","type":"binary","value_off":"OFF","value_on":"ON"},{"access":3,"description":"Lux Value","label":"Lux value","name":"lux_value","property":"lux_value","type":"enum","values":["10 lux","20 lux","50 lux","24h"]},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"NAS-PS10B2","options":[],"supports_ota":false,"vendor":"NEO"},"friendly_name":"0xa4c1380b136e2bee","ieee_address":"0xa4c1380b136e2bee","status":"successful","supported":true},"type":"device_interview"}'
info 2025-03-06 18:22:43z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0xa4c1380b136e2bee/human_motion_state/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"hw_version":1,"identifiers":["zigbee2mqtt_0xa4c1380b136e2bee"],"manufacturer":"NEO","model":"Human presence sensor","model_id":"NAS-PS10B2","name":"0xa4c1380b136e2bee","via_device":"zigbee2mqtt_bridge_0x048727fffe1bfc06"},"name":"Human motion state","object_id":"0xa4c1380b136e2bee_human_motion_state","origin":{"name":"Zigbee2MQTT","sw":"2.1.3","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/0xa4c1380b136e2bee","unique_id":"0xa4c1380b136e2bee_human_motion_state_zigbee2mqtt","value_template":"{{ value_json.human_motion_state }}"}'
info 2025-03-06 18:22:43z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0xa4c1380b136e2bee/dis_current/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"hw_version":1,"identifiers":["zigbee2mqtt_0xa4c1380b136e2bee"],"manufacturer":"NEO","model":"Human presence sensor","model_id":"NAS-PS10B2","name":"0xa4c1380b136e2bee","via_device":"zigbee2mqtt_bridge_0x048727fffe1bfc06"},"enabled_by_default":true,"name":"Current distance","object_id":"0xa4c1380b136e2bee_dis_current","origin":{"name":"Zigbee2MQTT","sw":"2.1.3","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/0xa4c1380b136e2bee","unique_id":"0xa4c1380b136e2bee_dis_current_zigbee2mqtt","unit_of_measurement":"cm","value_template":"{{ value_json.dis_current }}"}'
info 2025-03-06 18:22:43z2m:mqtt: MQTT publish: topic 'homeassistant/number/0xa4c1380b136e2bee/presence_time/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/0xa4c1380b136e2bee/set/presence_time","device":{"hw_version":1,"identifiers":["zigbee2mqtt_0xa4c1380b136e2bee"],"manufacturer":"NEO","model":"Human presence sensor","model_id":"NAS-PS10B2","name":"0xa4c1380b136e2bee","via_device":"zigbee2mqtt_bridge_0x048727fffe1bfc06"},"max":600,"min":3,"name":"Presence time","object_id":"0xa4c1380b136e2bee_presence_time","origin":{"name":"Zigbee2MQTT","sw":"2.1.3","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/0xa4c1380b136e2bee","step":1,"unique_id":"0xa4c1380b136e2bee_presence_time_zigbee2mqtt","unit_of_measurement":"s","value_template":"{{ value_json.presence_time }}"}'
[... cut ...]
`error 2025-03-06 18:23:57z2m: Publish 'set' 'motion_sensitivity_value' to '0xa4c1380b136e2bee' failed: 'Error: No datapoint defined for 'motion_sensitivity_value''`

I read that if you have a Tuya hub you might be able to use it to get the datapoint, but I don't have one so hopefully there is some other way?

**Issue #2**
When I set Work Mode to 'Manual' it still behaves like it's in 'Auto' mode.  Meaning the output turns on when presence is detected and off when presence is no longer detected.  I am unsure if this is just a limitation of the device or an issue with Z2M. The logs show it's in manual mode but the output is still controlled by presence.

Jiri72, 2025-03-07:
**Issue 1**
For 'motion_sensitivity_value' there is mismatch between 
https://github.com/Koenkk/zigbee-herdsman-converters/blob/253eced42835408bc9eda0de1f1a953721243069/src/devices/neo.ts#L498
and https://github.com/Koenkk/zigbee-herdsman-converters/blob/253eced42835408bc9eda0de1f1a953721243069/src/devices/neo.ts#L522
there should be "motion_sensitivity" instead of "motion_sensitivity_value"
so the line should look like
`                .numeric("motion_sensitivity", ea.STATE_SET)`

There is also issue for lux_value
https://github.com/Koenkk/zigbee-herdsman-converters/blob/253eced42835408bc9eda0de1f1a953721243069/src/devices/neo.ts#L528
should be 
`                [102, "lux_value", tuya.valueConverterBasic.lookup({"10 lux": 0, "20 lux": 1, "50 lux": 2, "24h": 3})],`
without underscores

Jiri72, 2025-03-07:
For **Issue 2**
I found this repo https://github.com/wzwenzhi/Wenzhi-ZigBee2mqtt/tree/main
and based on code tried to change
https://github.com/wzwenzhi/Wenzhi-ZigBee2mqtt/tree/main
from `                [101, "work_mode", tuya.valueConverterBasic.lookup({manual: 0, auto: 1})],`
to `                  [101, "work_mode", tuya.valueConverterBasic.lookup({manual: tuya.enum(0), auto: tuya.enum(1)})],`

Now, when I set Work mode to manual then switch is not automatically set to ON if presence is detected.

oapy123, 2025-03-07:
Thanks for the suggestions!!  I have no experience with coding or external converters.  I've spent the last few hours with chatgpt and claude trying to make a working external converter from the neo.ts file you linked, with the changes you suggested, but it keeps failing to load for one reason or another and I'm way out of my element.  If somebody can provide me with the external converter I can try it and see if that fixes my issues.  Appreciate any assistance!

lordniksidor, 2025-03-29:
I have the same issue, all the same simptoms and the cause seems identical based on the logs. I've tried cobbling together the external converter at
/homeassistant/zigbee2mqtt/external_converters/neo_nas_ps10b2.js, and it does actually load into Home Assistant based on the logs:
```csharp
[2025-03-30 00:46:04] info: z2m: Loaded external converter 'neo_nas_ps10b2.js'.
```
However, it does not work. When I change the sensitivity value after rebooting and the external converter has been loaded, I still get:

```csharp
[2025-03-30 01:23:38] error: z2m: Publish 'set' 'motion_sensitivity_value' to 'ToiletSensor' failed: 
'Error: No datapoint defined for 'motion_sensitivity_value''
```

Seems like Zigbee2MQTT is not using the custom converter... and I have no idea why.

```javascript
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const ea = exposes.access;
const e = exposes.presets;

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_1youk3hj',
    description: "Human presence sensor",
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        e.presence(),
        e.enum("human_motion_state", ea.STATE, ["none", "small", "large"])
            .withDescription("Human Motion State"),
        e.numeric("dis_current", ea.STATE)
            .withUnit("cm")
            .withValueMin(0)
            .withValueMax(600)
            .withValueStep(1)
            .withLabel("Current distance")
            .withDescription("Current distance of detected motion"),
        e.numeric("presence_time", ea.STATE_SET)
            .withUnit("s")
            .withValueMin(3)
            .withValueMax(600)
            .withValueStep(1)
            .withDescription("Presence Time"),
        e.numeric("motion_far_detection", ea.STATE_SET)
            .withUnit("cm")
            .withValueMin(150)
            .withValueMax(600)
            .withValueStep(75)
            .withDescription("Motion Range Detection"),
        e.numeric("motion_sensitivity", ea.STATE_SET)
            .withValueMin(0)
            .withValueMax(7)
            .withValueStep(1)
            .withDescription("Motion Detection Sensitivity"),
        e.numeric("motionless_sensitivity", ea.STATE_SET)
            .withValueMin(0)
            .withValueMax(7)
            .withValueStep(1)
            .withDescription("Motionless Detection Sensitivity"),
        e.enum("work_mode", ea.STATE_SET, ["manual", "auto"])
            .withDescription("Work Mode"),
        e.binary("output_switch", ea.STATE_SET, "ON", "OFF")
            .withDescription("Output Switch"),
        e.numeric("output_time", ea.STATE_SET)
            .withUnit("s")
            .withValueMin(10)
            .withValueMax(1800)
            .withDescription("Output Time"),
        e.binary("led_switch", ea.STATE_SET, "ON", "OFF")
            .withDescription("LED Switch"),
        e.enum("lux_value", ea.STATE_SET, ["10 lux", "20 lux", "50 lux", "24h"])
            .withDescription("Lux Value"),
    ],
    meta: {
        tuyaDatapoints: [
            [1, "presence", tuya.valueConverter.trueFalse1],
            [11, "human_motion_state", tuya.valueConverterBasic.lookup({none: 0, small: 1, large: 2})],
            [19, "dis_current", tuya.valueConverter.raw],
            [12, "presence_time", tuya.valueConverter.raw],
            [13, "motion_far_detection", tuya.valueConverter.raw],
            [15, "motion_sensitivity_value", tuya.valueConverter.raw],
            [16, "motionless_sensitivity", tuya.valueConverter.raw],
            [101, "work_mode", tuya.valueConverterBasic.lookup({manual: tuya.enum(0), auto: tuya.enum(1)})],
            [104, "output_switch", tuya.valueConverter.onOff],
            [103, "output_time", tuya.valueConverter.raw],
            [105, "led_switch", tuya.valueConverter.onOff],
            [102, "lux_value", tuya.valueConverterBasic.lookup({"10 lux": 0, "20 lux": 1, "50 lux": 2, "24h": 3})],
        ],
    },
};

module.exports = definition;
```

lordniksidor, 2025-03-30:
Okay, I finally got it working with the external converter—this time by actually reading (well, skimming) [Support new Tuya devices](https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html#support-new-tuya-devices) and simply copying all the definitions.

Here’s the code for my working external converter:

For those who understand how this works, would it make sense to submit a pull request to the ```neo ps10b2``` repository with these changes? Or could that potentially break someone else's existing device configuration?

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_1youk3hj',
        },
    ],
    model: 'TS0601_HPS',
    vendor: '_TZE204_1youk3hj',
    description: 'Human presence sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        e.presence(),
        e.enum("human_motion_state", ea.STATE, ["none", "small", "large"])
            .withDescription("Human Motion State"),
        e.numeric("dis_current", ea.STATE)
            .withUnit("cm")
            .withValueMin(0)
            .withValueMax(600) 
            .withValueStep(1)
            .withLabel("Current distance")
            .withDescription("Current distance of detected motion"),
        e.numeric("presence_time", ea.STATE_SET)
            .withUnit("s")
            .withValueMin(3)
            .withValueMax(600)
            .withValueStep(1)
            .withDescription("Presence Time"),
        e.numeric("motion_far_detection", ea.STATE_SET)
            .withUnit("cm")
            .withValueMin(150)
            .withValueMax(600)
            .withValueStep(75)
            .withDescription("Motion Range Detection"),
        e.numeric("motion_sensitivity", ea.STATE_SET)
            .withValueMin(0)
            .withValueMax(7)
            .withValueStep(1)
            .withDescription("Motion Detection Sensitivity"),
        e.numeric("motionless_sensitivity", ea.STATE_SET)
            .withValueMin(0)
            .withValueMax(7)
            .withValueStep(1)
            .withDescription("Motionless Detection Sensitivity"),
        e.enum("work_mode", ea.STATE_SET, ["manual", "auto"])
            .withDescription("Work Mode"),
        e.binary("output_switch", ea.STATE_SET, "ON", "OFF")
            .withDescription("Output Switch"),
        e.numeric("output_time", ea.STATE_SET)
            .withUnit("s")
            .withValueMin(10)
            .withValueMax(1800)
            .withDescription("Output Time"),
        e.binary("led_switch", ea.STATE_SET, "ON", "OFF")
            .withDescription("LED Switch"),
        e.enum("lux_value", ea.STATE_SET, ["10 lux", "20 lux", "50 lux", "24h"])
            .withDescription("Lux Value"),
    ],
    meta: {
        tuyaDatapoints: [
            [1, "presence", tuya.valueConverter.trueFalse1],
            [11, "human_motion_state", tuya.valueConverterBasic.lookup({none: 0, small: 1, large: 2})],
            [19, "dis_current", tuya.valueConverter.raw],
            [12, "presence_time", tuya.valueConverter.raw],
            [13, "motion_far_detection", tuya.valueConverter.raw],
            [15, "motion_sensitivity", tuya.valueConverter.raw],
            [16, "motionless_sensitivity", tuya.valueConverter.raw],
            [101, "work_mode", tuya.valueConverterBasic.lookup({manual: tuya.enum(0), auto: tuya.enum(1)})],
            [104, "output_switch", tuya.valueConverter.onOff],
            [103, "output_time", tuya.valueConverter.raw],
            [105, "led_switch", tuya.valueConverter.onOff],
            [102, "lux_value", tuya.valueConverterBasic.lookup({"10 lux": 0, "20 lux": 1, "50 lux": 2, "24h": 3})],
        ],
    },
};

module.exports = definition;

```

poeggi, 2025-04-28:
For the records, same fix also works for the lux value setting.

Currently, you cannot set the lux value / it has no effect - but if your change this line:

[102, "lux_value", tuya.valueConverterBasic.lookup({"10 lux": 0, "20 lux": 1, "50 lux": 2, "24h": 3})],

to this:

[102, "lux_value", tuya.valueConverterBasic.lookup({"10 lux": tuya.enum(0), "20 lux": tuya.enum(1), "50 lux": tuya.enum(2), "24h": tuya.enum(3)})],

Then setting the lux value works and has the desired effect.

closed 2025-09-01 by github-actions[bot]