Koenkk/zigbee2mqtt #26640: Duplicate properties

xX-Nexus-Xx, 2025-03-07:
### What happened?

Hi, I have just updated to latest version 2.1.3-dev commit (I'm running on ubuntu-docker)

I have now realized that some devices have duplicate properties .. when changing one element, than all duplicate elements get updated!

Tuya-Motion Sensor ZG-204ZL
[image]

Tuya Soil Sensor [TS0601_soil_3] TS0601_soil_3

[image]

### Zigbee2MQTT version

 2.1.3-dev commit

### Adapter firmware version

20221226

### Adapter

Sonoff USB Zigbee 3,0

### Setup

Ubuntu-Docker

cr0wm4n, 2025-03-08:
i get the same with all my TS011F_plug_3 plugs, all the calibration lines are duplicated around 6 times each

xX-Nexus-Xx, 2025-03-08:
[database.zip](https://github.com/user-attachments/files/19146281/database.zip)

pls find attached my db file

mrmaximas, 2025-03-10:
Week ago I fixed this problem by completely removing the devices and adding them back.

Koenkk, 2025-03-10:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-03-10 by Koenkk