Koenkk/zigbee2mqtt #26642: Support device similar to DAC7361C

coo-private, 2025-03-07:
### Link

https://www.aliexpress.com/item/1005008261491099.html?spm=a2g0o.order_list.order_list_main.5.57f41802U1jn5g

### Database entry

{"id":5,"type":"Router","ieeeAddr":"0xa4c1385ab117299e","nwkAddr":20360,"manufId":4417,"manufName":"_TZE204_5scyfj7e","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�\u000f]/i�\u000f]/i�\u000f]/i�\u000f]/i�\u000f]/i�\u000f]/i�\u000f]/i�\u000f]/i�\u000f]/i�\u000f]/i","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":74,"modelId":"TS0601","manufacturerName":"_TZE204_5scyfj7e","powerSource":1,"zclVersion":3,"hwVersion":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1741338060836}

### Zigbee2MQTT version

2.1.3-1

### Comments

Can you please add:

Model: TS0601
Vendor: _TZE204_5scyfj7e

Tuya Smart Three-Phase Bidirectional Electricity Meter with WiFi & ZigBee for KWh Monitoring, featuring RS485 Interface AC 380V

link: https://www.aliexpress.com/item/1005008261491099.html?spm=a2g0o.order_list.order_list_main.5.1c4979d23HFm8t&gatewayAdapt=glo2nld

[image]

Very similar to:

https://github.com/Koenkk/zigbee2mqtt/issues/18750
and
https://www.zigbee2mqtt.io/devices/EA4161C-BI.html
https://www.zigbee2mqtt.io/devices/TS0601_din_3.html
https://github.com/sprut/Hub/issues/3849

but this meter has current transformers.

I have purchased this unit but have not been able to get data from it.

[image]

[image]

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_x8fp01wi',
        },
    ],
    model: 'TS0601_din_3-Phase',
    vendor: 'TuYa',
    description: 'Earu Zigbee 3-Phase bidirictional Meter',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
            // Here you should put all functionality that your device exposes
            tuya.exposes.voltageWithPhase('L1'), 
            tuya.exposes.voltageWithPhase('L2'), 
            tuya.exposes.voltageWithPhase('L3'),
            tuya.exposes.currentWithPhase('L1'), 
            tuya.exposes.currentWithPhase('L2'), 
            tuya.exposes.currentWithPhase('L3'),
            tuya.exposes.powerWithPhase('L1'), 
            tuya.exposes.powerWithPhase('L2'), 
            tuya.exposes.powerWithPhase('L3'),
            //tuya.exposes.powerFactorWithPhase('L1'),
            //tuya.exposes.powerFactorWithPhase('L2'),
            //tuya.exposes.powerFactorWithPhase('L3'),
            //tuya.exposes.energyFlowWithPhase('L1'), 
            //tuya.exposes.energyFlowWithPhase('L2'),
            //tuya.exposes.energyFlowWithPhase('L3'),
            //tuya.exposes.energyWithPhase('L1'), 
            //tuya.exposes.energyWithPhase('L2'),
            //tuya.exposes.energyWithPhase('L3'),
            //tuya.exposes.energyProducedWithPhase('L1'), 
            //tuya.exposes.energyProducedWithPhase('L2'),
            //tuya.exposes.energyProducedWithPhase('L3'),
            
            // Change the description according to the specifications of the device
            e.energy().withDescription('Total forward active energy'),
            e.produced_energy().withDescription('Total reverse active energy'),
            //e.enum('energy_flow', ea.STATE, ['consuming', 'producing']).withDescription('Direction of energy'),
            //e.numeric('update_frequency', ea.STATE).withUnit('s').withDescription('Update frequency'),
            //e.ac_frequency(),
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
                
                [1, 'energy', tuya.valueConverter.divideBy100],
                [2, 'produced_energy', tuya.valueConverter.divideBy100],
                [6, null, tuya.valueConverter.phaseVariant2WithPhase('L1')],
                [7, null, tuya.valueConverter.phaseVariant2WithPhase('L2')],
                [8, null, tuya.valueConverter.phaseVariant2WithPhase('L3')],
        ],
    },
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

device is not supported

[image]

coo-private, 2025-03-08:
I edited the converter a bit and it worked.
But it still doesn't transfer a lot of data.

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_5scyfj7e',
        },
    ],
    model: 'TS0601_din_3-Phase',
    vendor: 'Tuya',
    description: 'EARU Zigbee 3-Phase bidirictional Meter',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
            // Here you should put all functionality that your device exposes
            tuya.exposes.voltageWithPhase('L1'), 
            tuya.exposes.voltageWithPhase('L2'), 
            tuya.exposes.voltageWithPhase('L3'),
            tuya.exposes.currentWithPhase('L1'), 
            tuya.exposes.currentWithPhase('L2'), 
            tuya.exposes.currentWithPhase('L3'),
            tuya.exposes.powerWithPhase('L1'), 
            tuya.exposes.powerWithPhase('L2'), 
            tuya.exposes.powerWithPhase('L3'),
            tuya.exposes.powerFactorWithPhase('L1'),
            tuya.exposes.powerFactorWithPhase('L2'),
            tuya.exposes.powerFactorWithPhase('L3'),
            //tuya.exposes.energyFlowWithPhase('L1'), 
            //tuya.exposes.energyFlowWithPhase('L2'),
            //tuya.exposes.energyFlowWithPhase('L3'),
            //tuya.exposes.energyWithPhase('L1'), 
            //tuya.exposes.energyWithPhase('L2'),
            //tuya.exposes.energyWithPhase('L3'),
            //tuya.exposes.energyProducedWithPhase('L1'), 
            //tuya.exposes.energyProducedWithPhase('L2'),
            //tuya.exposes.energyProducedWithPhase('L3'),
            
            // Change the description according to the specifications of the device
            e.energy().withDescription('Total forward active energy'),
            e.produced_energy().withDescription('Total reverse active energy'),
            e.enum('energy_flow', ea.STATE, ['consuming', 'producing']).withDescription('Direction of energy'),
            e.numeric('update_frequency', ea.STATE).withUnit('s').withDescription('Update frequency'),
            e.ac_frequency(),
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
                
                [1, 'energy', tuya.valueConverter.divideBy100],
                [2, 'produced_energy', tuya.valueConverter.divideBy100],
                [6, null, tuya.valueConverter.phaseVariant2WithPhase('L1')],
                [7, null, tuya.valueConverter.phaseVariant2WithPhase('L2')],
                [8, null, tuya.valueConverter.phaseVariant2WithPhase('L3')],
        ],
    },
};

module.exports = definition;
``` 

[image]

coo-private, 2025-04-07:
@Koenkk 
I followed the instructions
https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#_8-find-your-data-point 
and added the specified registers

1 - Forward Energy Total
2 - Reverse Energy Total
3 - month_energy
4 - daily_energy
6 - Phase A
7 - Phase B
8 - Phase C
9 - Total Active Power
11 - Frozen set
16 - Switch
18 - Meter ID
21 - Forward Energy T1
22 - Forward Energy T2
23 - Forward Energy T3
24 - Forward Energy T4

[image]

[image]

Koenkk, 2025-04-07:
Nice, now it should be a matter of filling the datapoints inside `tuyaDatapoints`

coo-private, 2025-04-07:
> Nice, now it should be a matter of filling the datapoints inside `tuyaDatapoints`

@Koenkk could you provide a link.

Koenkk, 2025-04-08:
It is present in the external converter you shared in the OP.

coo-private, 2025-04-10:
@Koenkk 

I tried this and added the simplest parameter to expose - switch. but it didn't work (

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_5scyfj7e',
        },
    ],
    model: 'TS0601_din_3-Phase',
    vendor: 'Tuya',
    description: 'EARU Zigbee 3-Phase bidirectional Meter',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
            // Here you should put all functionality that your device exposes
            tuya.exposes.voltageWithPhase('L1'), 
            tuya.exposes.voltageWithPhase('L2'), 
            tuya.exposes.voltageWithPhase('L3'),
            tuya.exposes.currentWithPhase('L1'), 
            tuya.exposes.currentWithPhase('L2'), 
            tuya.exposes.currentWithPhase('L3'),
            tuya.exposes.powerWithPhase('L1'), 
            tuya.exposes.powerWithPhase('L2'), 
            tuya.exposes.powerWithPhase('L3'),
            tuya.exposes.powerFactorWithPhase('L1'),
            tuya.exposes.powerFactorWithPhase('L2'),
            tuya.exposes.powerFactorWithPhase('L3'),
            tuya.exposes.switch(),
            //tuya.exposes.energyFlowWithPhase('L1'), 
            //tuya.exposes.energyFlowWithPhase('L2'),
            //tuya.exposes.energyFlowWithPhase('L3'),
            //tuya.exposes.energyWithPhase('L1'), 
            //tuya.exposes.energyWithPhase('L2'),
            //tuya.exposes.energyWithPhase('L3'),
            //tuya.exposes.energyProducedWithPhase('L1'), 
            //tuya.exposes.energyProducedWithPhase('L2'),
            //tuya.exposes.energyProducedWithPhase('L3'),
            
            // Change the description according to the specifications of the device
            e.energy().withDescription('Total forward active energy'),
            e.produced_energy().withDescription('Total reverse active energy'),
            //e.enum('energy_flow', ea.STATE, ['consuming', 'producing']).withDescription('Direction of energy'),
            //e.numeric('update_frequency', ea.STATE).withUnit('s').withDescription('Update frequency'),
            e.ac_frequency(),
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
                
                [1, 'energy', tuya.valueConverter.divideBy100],
                [2, 'produced_energy', tuya.valueConverter.divideBy100],
                [3, 'month_energy', tuya.valueConverter.divideBy100],
                [4, 'daily_eergy', tuya.valueConverter.divideBy100],
                [6, null, tuya.valueConverter.phaseVariant2WithPhase('L1')],
                [7, null, tuya.valueConverter.phaseVariant2WithPhase('L2')],
                [8, null, tuya.valueConverter.phaseVariant2WithPhase('L3')],
                [9, 'total_active_power', tuya.valueConverter.divideBy100],
                [11, 'frozen_set', tuya.valueConverter.divideBy100],
                [16, 'switch', tuya.valueConverter.onOff],
                [18, 'meter_id', tuya.valueConverter.divideBy100],
                [21, 'forward_energy_t1', tuya.valueConverter.divideBy100],
                [22, 'forward_energy_t2', tuya.valueConverter.divideBy100],
                [23, 'forward_energy_t3', tuya.valueConverter.divideBy100],
                [24, 'forward_energy_t4', tuya.valueConverter.divideBy100],
        ],
    },
};

module.exports = definition;

it looks like this 
[image]

after pressing on/off - error

[image]

coo-private, 2025-04-10:
I also noticed that nobody knows what 
forward_energy_t1
forward_energy_t2
forward_energy_t3
forward_energy_t4

but it is clearly described in the documentation as electricity tariffs.

[image]

[image]

closed 2025-06-17 by github-actions[bot]

sunchar, 2026-01-28:
Hi!

I'm considering buying some of this model.
Any news about supporting it? Or any workaround?

Thank you!