Koenkk/zigbee2mqtt #26645: Ikea LED2107C4 don't report color temperature in Kelvin

tomalta, 2025-03-07:
### What happened?

After updating the integration to 2.1.3 my Ikea LED2107C4 do not report color temperature anymore. It's possible to still control it but Home Assistant has no idea and falls back to midreds to report it.

### What did you expect to happen?

See color temperature in the device attributes and being able to see it change in Home Assistant.

### How to reproduce it (minimal and precise)

Update to the latest Zigbee2MQTT version and Home Assistant to 2025.3

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

7.4.5

### Adapter

ITead/Sonoff Zigbee Dongle-E V2

### Setup

Home Assistant OS

chris-1243, 2025-03-07:
Revert to HA 2025.2.5.
This has been reported in #26623 and all white spectrum lights seem impacted.

tomalta, 2025-03-08:
I wasn't experiencing this issue before the Z2M update so that's weird.

chris-1243, 2025-03-08:
What version of Z2M were you before updating? Were you using HA 2025.3.0 before updating Z2M?

Would you be able te reproduce the error?

tomalta, 2025-03-08:
Hi Chris. I remembered incorrectly. After some rounds of rollbacks I figured out the issue surfaced when I upgraded HA to 2025.3 and Z2M to 2.1.3. I didn't test Z2M 2.1.1 on 2025.3 unfortunately.

Koenkk, 2025-03-08:
Let's continue in https://github.com/Koenkk/zigbee2mqtt/issues/26623, seems to be a HA issue https://github.com/home-assistant/core/issues/140072

closed 2025-03-08 by Koenkk