Koenkk/zigbee2mqtt #26646: Avatto ZWSM16-3-Zigbee does not show energy meters

jlmpenedo, 2025-03-07:
### What happened?

I have some Avatto  ZWSM16-3-Zigbee (https://www.zigbee2mqtt.io/devices/ZWSM16-3-Zigbee.html) that I migrated from ZHA, where appears energy metrics.

### What did you expect to happen?

Devices show energy metrics

### How to reproduce it (minimal and precise)

Add a ZWSM16-3-Zigbee to ZHA e look at energy metrics. Pair the device with Z2M and those metrics are gone.

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

20240710

### Adapter

SLZB-06

### Setup

x86-64 samsung notebook core i5 - HassOS

mrmaximas, 2025-03-15:
Are you sure this device supports power measurement?  ZWSM16-1 and ZWSM16-2 do not.

jlmpenedo, 2025-03-15:
> Are you sure this device supports power measurement? ZWSM16-1 and ZWSM16-2 do not. What about post database.db entry for this device?

Yes. When I add to  ZHA it shows energy infomations. I will send a printscrenn as soon as possible

closed 2025-05-23 by github-actions[bot]