Koenkk/zigbee2mqtt #26650: Power-On Behavior isn't working for Innr RB 148 T

davidrpfarinha, 2025-03-07:
### What happened?

When trying to set the power-on behavior for my device it keeps getting me an error, even considering that under the documentation it says that they do support it.
https://www.zigbee2mqtt.io/devices/RB_148_T.html#power-on-behavior-enum
Am I doing something wrong?

[image]

[image]

### Zigbee2MQTT version

1.41.0 commit: cc31cea

### Adapter firmware version

20240710

### Adapter

SLZB-06

### Setup

Plain as a container within Synology NAS

Nerivec, 2025-03-12:
I see the firmware build date is pretty old. It looks like you have a version that does not support that attribute. Doesn't seem like that device has any OTA unfortunately...
Would need to compare to other owners of the same device, see if they have a different firmware version & behavior.

davidrpfarinha, 2025-03-13:
> I see the firmware build date is pretty old. It looks like you have a version that does not support that attribute. Doesn't seem like that device has any OTA unfortunately... Would need to compare to other owners of the same device, see if they have a different firmware version & behavior.

Thanks for the info. That might be it. Would be great to see if other owners have the same issue.
But I might just end up replacing them.

closed 2025-05-21 by github-actions[bot]