Koenkk/zigbee2mqtt #26651: Value of home assistant sensor is "" when it should be 0

henrikseger, 2025-03-07:
### What happened?

Hello,

On a nimly lock device the last_unlock_user is a number. But in Home Assistant it now shows as "" when that number is 0. When creating a custom MQTT sensor there is no problem, but the default discovered sensor shows the wrong value ("") instead of 0. If the value is anything else than 0, it works correctly. I've tracked it down to this line of code being the likely problem, as all the MQTT json etc looks good. It's only the representation in the Home Assistant sensor value that the error occurs.

https://github.com/Koenkk/zigbee2mqtt/blob/ba337bd329aeb4ca17735c0cf09b31293c8cff06/lib/extension/homeassistant.ts#L1214

### What did you expect to happen?

The home assistant sensor value should have been 0

### Zigbee2MQTT version

2.1.3-1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

EZSP v8

### Setup

Add-on on Homeassistant OS

closed 2025-05-14 by github-actions[bot]

bjorne, 2025-09-28:
Seeing the same issue. I indepedently tracked down the issue to the `0` value in the MQTT state not being handled well by `default('', True)`.

But maybe this value should instead be treated as a number?

bjorne, 2025-09-28:
I am thinking that this

https://github.com/Koenkk/zigbee-herdsman-converters/blob/83b4c405a177d1182c47d7408bbf508826b08129/src/devices/onesti.ts#L135

(and the same line for `last_lock_user`) could be changed to `e.numeric`?