Koenkk/zigbee2mqtt #26652: [New device support]: Imou door/windows Sensor

gabrielcolceriu, 2025-03-07:
### Link

https://www.imou.com/en/product/detail/imou-home-security-system

### Database entry

{"id":24,"type":"EndDevice","ieeeAddr":"0x8cf681fffef73b6c","nwkAddr":50885,"manufId":21845,"manufName":"MultIR","powerSource":"Battery","modelId":"ZD1-EN","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280,2821],"outClusterList":[3],"clusters":{"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}},"genBasic":{"attributes":{"modelId":"ZD1-EN","manufacturerName":"MultIR","powerSource":3,"zclVersion":3,"appVersion":4,"hwVersion":1,"dateCode":"20220615","swBuildId":"MC100_v1.2.1"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00212effff054cd2","zoneState":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":4,"hwVersion":1,"dateCode":"20220615","swBuildId":"MC100_v1.2.1","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1741373614443}

### Zigbee2MQTT version

2.1.3

### Comments

Need to add this to the database. Thank you!

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const definition = {
    zigbeeModel: ['ZD1-EN'],
    model: 'ZD1-EN',
    vendor: 'IMOU',
    icon: 'mdi:motion-sensor',
    description: 'Door & window sensor',
    fromZigbee: [fz.ias_contact_alarm_1, fz.battery, fz.ias_enroll],
    toZigbee: [],
    exposes: [
        e.contact(),
        e.battery(),
        e.tamper(),
    ],
};
module.exports = definition;
```

### What does/doesn't work with the external definition?

Everything works with the external converter.

Koenkk, 2025-03-08:
Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/imou.ts)?

gabrielcolceriu, 2025-03-09:
Hi there @Koenkk, i will close the issue as i am not a coder and i do not understand python. I only did this with the help of ChatGPT. I will use the external converter. Thank you!

closed 2025-03-09 by gabrielcolceriu

Koenkk, 2025-03-09:
I added it

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `ZD1-EN.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))