Koenkk/zigbee2mqtt #26656: After Home Assistant  Core Upgrade to 2025.3.1 TRV Current Temperature Wrong

AndyBowden, 2025-03-08:
### What happened?

I have twice upgraded to latest HA Core and Current Temperature of all my Sonoff TRV's is shown as 0.0.
Reversion to earlier HA Core has restored correct operation.

### Zigbee2MQTT version

2.1.3.1

### Adapter firmware version

unsure

### Adapter

unsure

### Setup

HA on HA Blue

### Debug log

[

[45df7312_zigbee2mqtt_2025-03-08T12-31-13.333Z.log](https://github.com/user-attachments/files/19143565/45df7312_zigbee2mqtt_2025-03-08T12-31-13.333Z.log)

](url)

lux73, 2025-03-08:
To me this clearly looks like an HA problem - so why post in the z2m forum?

and 2.1.3.**1** also looks like the ZHA Addon not z2m - please ask in the corresponding HA forum

AndyBowden, 2025-03-08:
Agreed, is clearly an HA problem, but I posted it here for info of other users.

closed 2025-05-15 by github-actions[bot]