Koenkk/zigbee2mqtt #26659: [2025-03-08 16:21:53] error: 	z2m: Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'data')} After pair Model Zigbee TS1201 Producent Zigbee _TZ3290_gnl5a6a5xvql7c2a

shirou93, 2025-03-08:
### What happened?

Model Zigbee
TS1201
Producent Zigbee
_TZ3290_gnl5a6a5xvql7c2a

https://www.zigbee2mqtt.io/devices/iH-F8260.html#tuya-ih-f8260

Zigbee2MQTT
[2.1.3](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.1.3) commit

Hello, my version of this device don’t have battery and propably this provide issues.

Thanks in advance

### What did you expect to happen?

Resolve errors from logs

### How to reproduce it (minimal and precise)

Pair and wait

https://www.zigbee2mqtt.io/devices/iH-F8260.html#tuya-ih-f8260

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

7.4.4

### Adapter

Sonoff emberZ

### Setup

Hassos Debian supervised

Nerivec, 2025-03-12:
Can you post the full debug log so we can trace the location of the error?

shirou93, 2025-03-12:
[log.txt](https://github.com/user-attachments/files/19209573/log.txt) This is debug logs from this remote

Nerivec, 2025-03-12:
I don't see any error in this debug log.
We need the stuff before/after that first line you posted:
```
[2025-03-08 15:28:36] error: z2m: Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'data')}
```

shirou93, 2025-03-12:
` Remaining battery in %, may take up to 24 hours to be reported.` I changed to debug mode and will wait. This error appears randomly. I will reupload new logs.

shirou93, 2025-03-14:
[log.log](https://github.com/user-attachments/files/19251561/log.log)

On this debug log error is visible. @Nerivec Could you check it?

Nerivec, 2025-03-14:
@Koenkk looks like `info` should not be considered valid in the return (it's not above)?
https://github.com/Koenkk/zigbee-herdsman-converters/blob/81f92ae94f4a13ea9e16250fbe5cd5daa9e5c56b/src/lib/zosung.ts#L20

Koenkk, 2025-03-16:
Should be fixed now!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-04-08 by Nerivec

shirou93, 2025-04-08:
I have another errors after update

`
[2025-04-07 22:10:11] error: 	z2m: Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'substring')}
[2025-04-07 22:24:25] error: 	z2m: Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'substring')}
[2025-04-07 22:47:07] error: 	z2m: Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'substring')}
[2025-04-07 22:47:08] error: 	z2m: Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'substring')} `