Koenkk/zigbee2mqtt #26660: Tuya curtain motor (_TZE204_xu4a5rhj/M3TYW-2.0-13) percentage inverted

devkaiwang, 2025-03-08:
### What happened?

I recently purchased a Tuya curtain motor, model `_TZE204_xu4a5rhj`/`M3TYW-2.0-13`. The motor is of good quality and quiet. However, there is an annoying issue when setting the motor's percentage through the Home Assistant cover UI or the Tuya app's curtain animation, the percentage is inverted.

I wrote this converter code, which successfully fixed the problem. I’m curious if others have encountered this same issue or if it’s just a firmware bug specific to my device.

[image]

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const e = exposes.presets;
const ea = exposes.access;

const invertedTuyaCoverControl = {
    key: ['state', 'position'],
    convertSet: async (entity, key, value, meta) => {
        if (key === 'position') {
            value = 100 - value; // Invert the position value
        }
        return await legacy.toZigbee.tuya_cover_control.convertSet(entity, key, value, meta);
    },
};

const definition = {
    zigbeeModel: [],
    fingerprint: [
        ...tuya.fingerprint('TS0601', [
            '_TZE204_xu4a5rhj',
        ]),
    ],
    model: 'TS0601_cover_1',
    vendor: 'Tuya',
    description: 'Curtain motor with corrected percentage setting',
    whiteLabel: [],
    fromZigbee: [legacy.fromZigbee.tuya_cover, fz.ignore_basic_report],
    toZigbee: [invertedTuyaCoverControl, legacy.toZigbee.tuya_cover_options],
    exposes: [
        e.cover_position().setAccess('position', ea.STATE_SET),
        e.composite('options', 'options', ea.STATE_SET)
            .withFeature(e.numeric('motor_speed', ea.STATE_SET)
                .withValueMin(0)
                .withValueMax(255)
                .withDescription('Motor speed'))
            .withFeature(e.binary('reverse_direction', ea.STATE_SET, true, false)
                .withDescription('Reverse the motor direction')),
    ],
};

module.exports = definition;
```

### Zigbee2MQTT version

    2.1.3

### Adapter firmware version

20240710

### Adapter

zStack3x0

### Setup

Pi5

devkaiwang, 2025-05-09:
An option named `cover_position_percent_fix` is added.

closed 2025-05-17 by devkaiwang

Pabl0z, 2026-02-01:
I have a device that externally resembles the one presented by you, including the print of the model, but at the level of the stack zigbee it looks different. 
z2m db drop: 

`{"id":9,"type":"Router","ieeeAddr":"0xb43522fffeec797e","nwkAddr":1070,"manufId":4098,"manufName":"_TZE200_heq2ohyy","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_heq2ohyy","powerSource":1,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genGroups":{"attributes":{"nameSupport":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{},"lastSeen":1769977460894}`

And presents itself as unsupported in z2m :/

[image]