Koenkk/zigbee2mqtt #26670: SCNHEIDER CCT5015-0002w

BleuErdre, 2025-03-09:
### What happened?

Hello,

I try to connect CCT5015-0002W with sonoff zdongle-e Homeassistant zigbeee2mqqt. This device is list when i allow join, but each time after it's left network?
Version Zigbee2MQTT 2.1.3) commit: [unknown]
EmberZNet
coordinateur 7.4.1 
Version de zigbee-herdsman-converters 23.2.0
Version de zigbee-herdsman 3.2.7

Could you help me to solve this problem?

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

7.4.1

### Adapter

Sonoff zdongle-e ember

### Setup

PI5 homeassistant

### Debug log

 2025-03-09 16:37:22z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x94ec32fffeadceda/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0x94ec32fffeadceda/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"hw_version":1,"identifiers":["zigbee2mqtt_0x94ec32fffeadceda"],"manufacturer":"Schneider Electric","model":"Roller shutter module","model_id":"CCT5015-0001","name":"0x94ec32fffeadceda","sw_version":"002.012.002 R","via_device":"zigbee2mqtt_bridge_0xe406bffffe35f1b6"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"Linkquality","object_id":"0x94ec32fffeadceda_linkquality","origin":{"name":"Zigbee2MQTT","sw":"2.1.3","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/0x94ec32fffeadceda","unique_id":"0x94ec32fffeadceda_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
info 2025-03-09 16:37:22z2m: Successfully configured '0x94ec32fffeadceda'
info 2025-03-09 16:37:45z2m: Zigbee: disabling joining new devices.
info 2025-03-09 16:37:45zh:ember: [STACK STATUS] Network closed.
info 2025-03-09 16:37:45z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":0},"status":"ok","transaction":"tz0gs-2"}'
warning 2025-03-09 16:37:46z2m: Device '0x94ec32fffeadceda' left the network

Nerivec, 2025-03-12:
Can you post the full `debug` log from the network opening to the device leaving?
Also, try to updating to firmware 8.0.2, see if it helps.

BleuErdre, 2025-03-13:
Hello,

I can't find firmware 8.0.2 for zbdongle-e, the official site sonoff don't propose this firmware.
https://github.com/itead/Sonoff_Zigbee_Dongle_Firmware/tree/master/Dongle-E (link on sonoff web site)

Have you a link to download this firmware?

I have fewer disconnection problems, but when I reboot HomeAssistant, I can't access Schneider equipment. I need to restart the Mosquitto MQTT and Zigbee2MQTT services.
Without restarting services, I get no informations or commands in my dashboard.
I don't know if problem is broker or zigbee2mqtt.

Nerivec, 2025-03-13:
Official sonoff firmware don't have most of the "community fixes", see here (you'll want the `zigbee_ncp` which is the ZigBee coordinator firmware):
https://github.com/darkxst/silabs-firmware-builder/releases
https://github.com/Nerivec/silabs-firmware-builder/releases

closed 2025-05-21 by github-actions[bot]