Koenkk/zigbee2mqtt #26672: Tuya ZY-M100-24GV3 Move sensitivity and Presence sensitivity are set to state 0

HansMann18, 2025-03-09:
### What happened?

Move sensitivity and Presence sensitivity are set to state 0, which should not exist according to the Zigbee2Mqtt device list.

"The minimal value is 1 and the maximum value is 10"

I have 8 of these sensors in use. Under iobroker, the sensors issued a warning message under Home Assistant. If you're not careful, the log fills up, and you can quickly get 80,000 entries a day.
I tried to create HA automations that reset the value, but without success.

### How to reproduce it (minimal and precise)

The condition is assumed to last for an indefinite period of time.

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

20240710

### Adapter

zstack

### Setup

proxmox LXC

### Debug log

[image]
[image]

`2025-03-09 19:20:53.033 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.aws_kueche_presence_sensitivity: 0 (range 1.0 - 10.0)
2025-03-09 19:20:53.034 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.aws_kueche_move_sensitivity: 0 (range 1.0 - 10.0)`

I ask for help

HansMann18, 2025-03-11:
@jesse-kaufman 
Have you created an automation for Home Assistant? What do you use as a trigger? I noticed that the entities in HA change to the Unknown status and continue to display the numerical value. I can use the sliders to adjust the numerical values ​​up and down, after which the entities are recognized again until the error occurs again.

HansMann18, 2025-03-11:
@jesse-kaufman 
Thank you for the information, I will create an automation with the same triggers and hope that it works.

jjfs127, 2025-03-12:
> [@jesse-kaufman](https://github.com/jesse-kaufman) Thank you for the information, I will create an automation with the same triggers and hope that it works.

Were you able to create an automation that works? I wasn't able to get mine to work.

```
- id: invalid_value_error_for_tuya_presence_sensor
  alias: 'Invalid Value Error for Tuya Presence Sensors'
  mode: restart
  trigger:
  - trigger: state
    entity_id: number.0xa4c1384aa4fb445a_presence_sensitivity
    to: 
    - "0"
    - "unknown"
    - "unavailable"
  - trigger: state
    entity_id: number.0xa4c1384aa4fb445a_move_sensitivity
    to: 
    - "0"
    - "unknown"
    - "unavailable"
  condition: []
  actions:
    - action: number.set_value
      data:
        value: "1"
      target:
        entity_id: "{{ trigger.entity_id }}"
```

jjfs127, 2025-03-14:
I notice that the actual HA entities themselves are not randomly changing to zero but only the UI entities in Z2M are showing Zero, so I'm even sure how to create an automation if the entity states are not set to zero.

jesse-kaufman, 2025-03-14:
Huh my HA entities update as expected. Previously they would get set to 0, but then I think there was an HA update that changed the range. Ever since that update, mine show as "unknown" when they get reset.

HansMann18, 2025-03-14:
@jjfs127 
I haven't been able to create a working automation via HA either. I have the same problem: HA doesn't set the values ​​to 0. I now have a working automation running via ioBroker. In ioBroker, the values ​​are set to 0.
Sorry I can't help you.