Koenkk/zigbee2mqtt #26676: Vimar 14595 he associates but does not communicate

delli87, 2025-03-09:
### What happened?

I can pair the device correctly, but as soon as I try to give any command the error appears: z2m: Device 'Luce Sala Da Pranzo' has no endpoint 'undefined'

### How to reproduce it (minimal and precise)

Trying to turn the light on and off or adjusting the brightness

### Zigbee2MQTT version

 2.1.3 commit: unknown

### Adapter firmware version

7.4.4 [GA]

### Adapter

ITEAD Sonoff ZBDongle-E (ZigBee 3.0 USB Dongle Plus) – v2

### Setup

Add-on on Home Assistant Os

### Debug log

[log.log](https://github.com/user-attachments/files/19152072/log.log)

LeoeLeoeL, 2025-03-10:
Same issue for me after updating FW to 1.1.5.
I tried to pair to Deconz and it works.

[image]

[image]

[image]

My adapter is ITEAD Sonoff ZBDongle-P

delli87, 2025-03-10:
> I could use this workaround, but I'd prefer to use z2m. Luckily I have two
> zigbee antennas.
>
> Same for me. I'm using HUE bridge and Z2M. I would avoid to have the 3rd
> network
> :-)

Me neither, period, I hope it gets fixed soon.

Nicola Delliturri

Nerivec, 2025-03-12:
Have you tried this:
- remove the device from the network (do NOT force!)
- stop Z2M
- start Z2M
- pair the device

If it still doesn't work after this, post a screenshot of the Clusters page in frontend for that device.

delli87, 2025-03-12:
> The device is unable to be removed without the FORCE option. I stopped and started Z2M several times but with no improvement.
> 
> [image] The deconz version: [image]

The clusters are the same as yours.
The only difference is that I was able to dissociate.
But after pairing, the issue persists.

delli87, 2025-03-20:
Is there anything new?

LeoeLeoeL, 2025-03-22:
For info:
I binded an Ikea E1743 to this device and I can switch on/off and dim the light using that.

LeoeLeoeL, 2025-03-26:
Is there anything about this issue in dev branch I can test?

LeoeLeoeL, 2025-03-28:
I didn't find thw fiz in the changelog https://gist.github.com/Koenkk/bfd4c3d1725a2cccacc11d6ba51008ba#new-supported-devices
Possible?
Tons of **New Supported Devices** but what about the **New UNSupported Devices**?

lucacillario, 2025-03-29:
Same problem here... any updates?

LeoeLeoeL, 2025-04-04:
I have one more info.
If i switch on/off, using the physical buttons, the state is regularly update; I can see that in Home Assistant

[image]
Infact I can read the state and current level in dev console.
Only commands via Z2M don't work.

LeoeLeoeL, 2025-04-08:
Can you help me to understand?

Here
https://github.com/Koenkk/zigbee-herdsman-converters/commit/3a10293dcd0563345b33849b3a45889312af0c3e
there is what Koenkk added in Vimar.ts
 {
         zigbeeModel: ['DimmerSwitch_v1.0'],
         model: '14595.0',
         vendor: 'Vimar',
         description: 'IoT connected dimmer mechanism 220-240V',
         extend: extend.light_onoff_brightness({noConfigure: true, disablePowerOnBehavior: true}),
         endpoint: (device) => {
             return {default: 11};
         },
         configure: async (device, coordinatorEndpoint, logger) => {
             await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
             const endpoint = device.getEndpoint(11);
             await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
             await reporting.onOff(endpoint);
             await reporting.brightness(endpoint);
         },
     },

This is what I have in Vimar.js
{
        zigbeeModel: ["DimmerSwitch_v1.0"],
        model: "14595.0",
        vendor: "Vimar",
        description: "IoT connected dimmer mechanism 220-240V",
        extend: [m.light({ configureReporting: true, powerOnBehavior: false })],
        endpoint: (device) => {
            return { default: 11 };
        },
    },

Is it normal or something is missing?

LeoeLeoeL, 2025-04-11:
https://github.com/Koenkk/zigbee2mqtt/issues/18323
@giovannismart @Koenkk 
Can you help us to make this device operative again?
Tks

Koenkk, 2025-04-12:
Could you:
- Force remove the device through the z2m frontend
- Restart z2m
- Provide the debug log of: start z2m, pair the device and try to control it (this fails right?)

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

delli87, 2025-04-12:
> Could you:
> 
> * Force remove the device through the z2m frontend
> * Restart z2m
> * Provide the debug log of: start z2m, pair the device and try to control it (this fails right?)
> 
> See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

[log (2).log](https://github.com/user-attachments/files/19720718/log.2.log)

saveriol, 2025-04-13:
Does it work if you put the attached file in [root of Z2M]/external_converters/  ??

(rename it to VIMAR_dimmer.js without the .txt extension)

[VIMAR_dimmer.js.txt](https://github.com/user-attachments/files/19724706/VIMAR_dimmer.js.txt)

LeoeLeoeL, 2025-04-13:
It seems to work. :-)
Brightness doesn't go below 63.
Is it a Vimar lowest brightness available for this dimmer?

delli87, 2025-04-13:
> It seems to work. :-) Brightness doesn't go below 63. Is it a Vimar lowest brightness available for this dimmer?

I'll try it too as soon as I can.
You have to adjust the minimum brightness by performing combinations with the physical buttons. It's all written in the manual.

LeoeLeoeL, 2025-04-13:
I re-paired again, configured TE and then the lowest possible brightness and now is OK.

For info, if you configure the colour of the LED and its behaviour (when OFF and ON) while the device has the BT firmware then the setting remains later when you flash again zigbee firmware. :-)
This thing doesn't work fir switches. :-(

saveriol, 2025-04-13:
If you confirm it works (I don't have a dimmer) I'll write a PR to change the converter, evidently Vimar changed the endpoint of the dimmer with the last firmware, that's why it didn't work anymore

delli87, 2025-04-13:
> Se mi confermi che funziona (non ho un dimmer) ti scrivo una richiesta di modifica del convertitore, evidentemente la Vimar ha cambiato l'endpoint del dimmer con l'ultimo firmware, ecco perché non funzionava più

If I can, I'll try tomorrow too.

tiagoroldao, 2025-04-14:
> Does it work if you force remove, put the attached file in [root of Z2M]/external_converters/ , restart Z2M and join again??
> 
> (rename it to VIMAR_dimmer.js without the .txt extension)
> 
> [VIMAR_dimmer.js.txt](https://github.com/user-attachments/files/19724706/VIMAR_dimmer.js.txt)

@saveriol I can confirm, having the same issue, that following your procedure the dimmer functions normally, communicating both ways without fault.

closed 2025-04-15 by Koenkk

LeoeLeoeL, 2025-04-15:
Tks @saveriol.

A question. When will be the next release of Z2M, do I have to delete the external converter of Vimar before the update?

delli87, 2025-04-15:
@saveriol, I found another problem, when I perform this action:
`action: light.turn_on
data:
  brightness_pct: 100
target:
  entity_id: light.luce_sala_da_pranzo`
If I set the brightness to the current one (in this case 100%) the light remains off.
I am forced to give two separate commands. One for the ignition and one for the regulation.

saveriol, 2025-04-16:
> [@saveriol](https://github.com/saveriol), I found another problem, when I perform this action: `action: light.turn_on data: brightness_pct: 100 target: entity_id: light.luce_sala_da_pranzo` If I set [...]

Was it working in the past? You should open a new issue for this attaching debug logs when you send these commands to the dimmer.

delli87, 2025-04-16:
@saveriol  I don't know. I used Deconz before. As soon as I have time I will open a new report.