Koenkk/zigbee2mqtt #26679: Can't install zigbee2mqtt/zigbee2mqtt-amd64:2.1.3-1: 500

crazyjchpc, 2025-03-10:
### What happened?

Can't install zigbee2mqtt/zigbee2mqtt-amd64:2.1.3-1: 500 Server Error for http+docker://localhost/v1.47/images/create?tag=2.1.3-1&fromImage=zigbee2mqtt%2Fzigbee2mqtt-amd64&platform=linux%2Famd64: Internal Server Error ("Get "https://registry-1.docker.io/v2/": net/http: TLS handshake timeout")

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

20211210

### Adapter

ztm

### Setup

x86

Nerivec, 2025-03-12:
Looks like troubles with the docker server. Nothing Z2M can do about that. It hopefully should pass after a while (if not, you likely have firewall troubles or something along that line).

closed 2025-03-12 by Nerivec

alexsoh, 2025-03-13:
I am running HAOS on Windows and hitting this. Every other AddOn works except Z2M.

Koenkk, 2025-03-13:
Maybe you are hitting rate limits? Try again in a day or two.

alexsoh, 2025-03-13:
Thanks, have been trying for a long time now, since I am not able to update since 2.1.2-1 is out. I am stuck at 2.1.1-1 and is unable to update because of this. I try to skip the update since it's an eye sore/annoyance at this point but HA errors out as even if I disable automatic update. 

`Automatic updates for this item have been enabled; skipping it is, therefore, unavailable. You can either install this update now or wait for Home Assistant to do it automatically.`