Koenkk/zigbee2mqtt #26684: [New device support]: lumi.light.agl003 Aqara Smart Bulb

MiAutomations, 2025-03-10:
### Link

https://www.aqara.com/en/product/led-bulb-t2/

### Database entry

{"id":200,"type":"Router","ieeeAddr":"0x54ef4410010fc4b5","nwkAddr":6142,"manufId":4447,"manufName":"Aqara","powerSource":"DC Source","modelId":"lumi.light.agl003","epList":[1,21],"endpoints":{"1":{"profId":260,"epId":1,"devId":258,"inClusterList":[5,4,3,0,768,8,6,2820,1794,64704],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.light.agl003","manufacturerName":"Aqara","powerSource":4,"zclVersion":3,"appVersion":27,"stackVersion":27,"hwVersion":1,"dateCode":"20250113"}}},"binds":[],"configuredReportings":[],"meta":{}},"21":{"profId":260,"epId":21,"devId":0,"inClusterList":[12],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":27,"stackVersion":27,"hwVersion":1,"dateCode":"20250113","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1739524122424}

### Zigbee2MQTT version

2.1.3

### Comments

Once I'm not able to open a PR please add this new device

Hello can you please add this new device as a supported device to the new Z2M release?

lumi.light.agl003

external_definition

```
const` m = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['lumi.light.agl003'],
    model: 'lumi.light.agl003',
    vendor: 'Aqara',
    description: 'Automatically generated definition',
    extend: [m.deviceEndpoints({"endpoints":{"1":1,"21":21}}), m.light({"colorTemp":{"range":[111,500]},"color":true}), m.electricityMeter()],
    meta: {"multiEndpoint":true},
};

module.exports = definition;

```

Thank you 

### External definition

```shell
const` m = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['lumi.light.agl003'],
    model: 'lumi.light.agl003',
    vendor: 'Aqara',
    description: 'Automatically generated definition',
    extend: [m.deviceEndpoints({"endpoints":{"1":1,"21":21}}), m.light({"colorTemp":{"range":[111,500]},"color":true}), m.electricityMeter()],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

With the external definition the energy doesn't work

bsunderhus, 2025-04-04:
Power-on behavior also seem to not work. I'm getting:

> z2m: Publish 'get' 'power_on_behavior' to 'lumi.light' failed: 'Error: ZCL command 0x54ef4410010fb2c1/1 genOnOff.read(["startUpOnOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

szczot3k, 2025-04-18:
Dupliacte of #26182 - there's a (semi?)working custom converter.

closed 2025-09-01 by github-actions[bot]