Koenkk/zigbee2mqtt #26687: Orvibo MixSwitches with firmware 6.1.7 randomly turning off and on

vilaemail, 2025-03-10:
### What happened?

I have Orvibo MixSwitches and have narrowed down issue to be affecting only 3 gang and 1 gang switches with firmware version 6.1.7 (20230712).
In random intervals random number of switches turn off and on. This can be 30 seconds or 5 minutes after last "event". The "event" itself is any number of switches turning off and on, it can be 1 or even 6-7. When its more of them they are separated by around 1-2 seconds. Between off and on is also 1-2 seconds.

To help narrow down issue I created separate network/adapter just switches I identified as problematic on channel 21 (I later put devices to Orvibo gateway (MixPad S) and they were on this channel and worked without issues). I can confirm that on the main network (containing other Orvibo switches and other devices) there is no issues, but on this "isolation" network there are issues.

Followed each reset there is log line "device announced itself", I am assuming device is crashing for some reason and on reboot it turns switch off and then on, so the goal is likely to avoid device crashing. Given the logs and sniff from official Orvibo gateway which does not experience issue I am hoping to identify what is "different" on the Z2M network and "change Z2M/network behavior" such that crashes do not happen.

### What did you expect to happen?

Switches do not randomly turn off and on.

### How to reproduce it (minimal and precise)

Use Orvibo MixSwitches with firmware version 6.1.7, pair them to Z2M and wait for 5minutes-1hour for random issue start occurring.

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

20240710

### Adapter

UZG-01 (CC2652P7 chip)

### Setup

Plain on Alpine Linux VM

### Debug log

Here are debug logs (at first I am pairing devices (After 15:23 pairing should be finished), but afterwards resets happen. Always after reset I can see "device announces itself" log): [switches_restarting.log](https://github.com/user-attachments/files/19168505/switches_restarting.log)
Here is sniff of Orvibo gateway network (look only after 212s after i rebooted all devices). There is separate file with the key and another one with some notes on which actions I have performed on the devices: 
[orvibo12sw_gw_capture.zip](https://github.com/user-attachments/files/19168593/orvibo12sw_gw_capture.zip)

vilaemail, 2025-03-10:
If anyone experienced with zigbee protocol could take a look and let me know if something obvious is different between Z2M network with the issue and the Orvibo sniffed network without the issue, that would really help me out.

Nerivec, 2025-03-12:
That sniff contains plenty of malformed frames, and also frames that appear to be encrypted with different keys. And I see dozens of different PAN IDs (which I assume is just more malformed frames that Wireshark doesn't know how to parse).
At first glance, I'd say they use off-spec frames to handle some internal behaviors, which would explain why it doesn't work properly with a standard ZigBee network... unfortunately, that would also mean it is never going to work properly with a standard ZigBee network.

Can you provide a sniff with Z2M?

vilaemail, 2025-03-12:
@Nerivec Do you see those issues past 210 seconds of Wireshark capture (packet 360 and onwards)? If yes, can you please send couple of packet numbers so I can recognize these issues myself in the future?

Reason for asking is that I've been trying to do some analysis of my own and to inexperienced me it actually looks "normal" (routing info, link status, report attribute, read attribute, recall scene) based on the "info" column in wireshark. There are some frames with error which I assume could be because my sniffer cannot pick up messages from the whole mesh (coordinator and sniffer are located close, but on the physical edge of the network, so I assume far away devices messages get "routed through the mesh" and the message to first hop is unlikely to be sniffed correctly).

I will do sniff with Z2M and send it soon.

Nerivec, 2025-03-12:
Here's a set of coloring rules for Wireshark, it should help a lot with identifying stuff. `View` > `Coloring Rules` > `Import`
[wireshark-coloring-rules.txt](https://github.com/user-attachments/files/19209308/wireshark-coloring-rules.txt)

Packets that remain encrypted will usually have a description that remains to `command` or `data`, and the network (or aps) layer will be highlighted (the very first packet for example).
Malformed packets mention it in the description. You can use filter `_ws.col.info matches "malformed" or _ws.col.info matches "bad"` to identify most of them.

vilaemail, 2025-03-12:
Here is Z2M sniff and logs (read info file for when some of the resets happen). Here I first pair all 12 devices and then issue starts happening after several minutes: [orvibo12sw_z2m_capture.zip](https://github.com/user-attachments/files/19210327/orvibo12sw_z2m_capture.zip)

On a hunch I did a sniff of pairing device to Orvibo gateway (hoping the resets might have something to do with "not pairing device as it expects leads it to crash periodically"): [orvibo12swPairing_gw_capture.zip](https://github.com/user-attachments/files/19210323/orvibo12swPairing_gw_capture.zip)

Thanks for sharing! I'll look into those as well as the new sniff and try to see if I see something. If you can take a look and let me know if you see something that would help me greatly.

vilaemail, 2025-03-12:
Diff I found is that:
- 4 scenes get set-up for all Orvibo switches by Orvibo gateway. 2 for all ON and 2 for all OFF.
- Reporting is not configured by Orvibo gateway (but is by Z2M).
- Z2M is reading some additional information from the device, and in particular 2 reads fail for every endpoint (hardware version and stack version)

I have changed converter to stop configuring reporting (diff 2 from above), and there have been no issues for ~50 minutes. I'll keep monitoring and send PR if this fixes the issue.

vilaemail, 2025-04-12:
This was fixed with [this PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/8984)

closed 2025-04-12 by vilaemail