Koenkk/zigbee2mqtt #26692: Hundreds of 'Delivery of BROADCAST failed'

Ltek, 2025-03-10:
### What happened?

I'm wondering if these errors are causing many/most of my devices to drop off the zigbee network. Devices used to stay connected without issues but as of a few versions ago, most of my devices do not connect to a router or conductor.

I get tons 

[log.zip](https://github.com/user-attachments/files/19170459/log.zip)

of these...

Delivery of BROADCAST failed for '65532'
--note they are slightly different--
Delivery of BROADCAST failed for '65533'

[image]

### What did you expect to happen?

All nodes to connect to either a router or conductor

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

7.4.5

### Adapter

Dongle-E

### Setup

PC

Nerivec, 2025-03-12:
Something nearby could have introduced interference in your environment. You might want to check on that. You can use `ember-zli` to run [network scans](https://github.com/Nerivec/ember-zli/wiki/Stack#scan-network), to see how busy the air is. And make sure nothing introduced interference at the USB connection.
Also try updating to 8.0.2, see if it helps.

Ltek, 2025-03-12:
> Something nearby could have introduced interference in your environment. You might want to check on that. You can use `ember-zli` to run [network [...]

I did update to 8.0.2 yesterday... same issue. Even after removing the 3 routers that caused serious node issues, I still get these Broadcast errors, just fewer of them.  

2.4ghz band has almost nothing going on in/around my house. 99% of the traffic is 5Ghz. Walking the house I show no signals all above channel 4 (my own 2.4 is set to channel 2, so this makes sense)... my zigbee is on 11 (z2m) and 25 (deconz).

I changed Z2M to 15 to see if that makes any difference so we'll see if I get any more of these errors. After changing the channel and restarting Z2M all the nodes connected very quickly (as usual - except when I had the other 3 routers powered on)

Ltek, 2025-03-13:
@Nerivec I've spent the better part of 2 days trying to get the network stable... and failing. I've changed channels 3 times and also moved around the routers. The upside (I think) is that the routers now have super-solid connections to the Coordinator, the down side is that half the nodes done connect to anything. It makes zero sense.... some of these nodes are a few feet open air away from the routers.

Most of the errors are gone... every 20-30 mins I get...

`Z2M Server: zh:controller:device: Handling of poll check-in from 0x000d6f0011c74ee7 failed (ZCL command 0x000d6f0011c74ee7/1 genPollCtrl.checkinRsp({"startFastPolling":false,"fastPollTimeout":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false,"sendPolicy":"immediate"}) failed ({"target":861,"apsFrame":{"profileId":260,"clusterId":32,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":100},"zclSequence":238,"commandIdentifier":11} timed out after 10000ms))`

then....

`Z2M Server: zh:ember: Delivery of BROADCAST failed for '65533'.`

Without the routers I had a stable network, for about half a day with the first image. But that crashed and burned about half a day later and half the devices disconnected so I tried the routers again with several channel changes.

stable (half a day) looked like this...
[image]

then it crashed... so another day of work/settling it now looks like this...

[image]

Nerivec, 2025-03-14:
Note: End device can sometimes appear not connected in the network map, it doesn't necessarily mean they don't work though.
Note2: Any network address above 65528, is used for broadcasting (sending to an entire subset of devices, like "all routers").

How is your adapter connected to your machine?
- USB extension cord?
- USB hub? (with separate power supply?)
- What's your machine/setup exactly?

Did you try another USB port / extension cord?

If you haven't already, can you try setting `transmit_power: 20` under `advanced` in your configuration.yaml: https://www.zigbee2mqtt.io/guide/configuration/adapter-settings.html#advanced-configuration

What's the model of the four router plugs?

Ltek, 2025-03-14:
> Note: End device can sometimes appear not connected in the network map, it doesn't necessarily mean they don't work though. Note2: Any network address above 65528, is used for broadcasting (sending [...]

I definitely have some devices that are not connected, lights are not triggering with motion sometimes. Not great if the tool we use to help TShoot doesnt show correct info.

routers are Third Reality plugin outlets

USB extension, no hub
Windows PC, USB 2 & 3
Sonoff Dongle-E with 8.x FW as of 2 days ago
USB cable works fine, the dongle is powered and runs. 

I didnt know about the transmit power, I just pushed it to 20 and will see what happens overnight if that helps.

Nerivec, 2025-03-14:
> it doesn't necessarily mean they don't work though

It relies on the network being stable enough to show proper information though (including the quality of the routers present, since that information is retrieved from them), so, it is limited by that fact...
But in some cases, end devices just don't show as connected because of their current (temporary) state (they consume a very small amount of power, but at the cost of not always being considered online... until triggered).

On Windows, in the device manager (COMx device), you may want to check the settings you have. I'm not sure what are the defaults nowadays...
Make sure you have the latest drivers too, just in case (see bottom of page, not sure which version of the Dongle-E you have, but you can install both without problem, one will work):
https://github.com/Nerivec/ember-zli/wiki#serial-drivers

mrmaximas, 2025-03-16:
@Nerivec Is there a similar ember-zli-like sw for TI devices?

Nerivec, 2025-03-16:
I'm not aware of one, but I'm not very familiar with the zstack eco-system.

Nerivec, 2025-03-17:
@Ltek any success on your previous troubles?

Ltek, 2025-03-17:
> [@Ltek](https://github.com/Ltek) any success on your previous troubles?

looks like the network has stabilized. No more broadcast here are still a few sensors that do not stay connected... I know since a few motion sensors trigger lighting, and they dont read when tested. I've also noticed a huge lag with some of the motion sensors... 3 to 10 seconds. The lag wasnt there even with the ton of errors I was getting... but now I have a lag, when I dont have errors.?

Nothing yet today but these the last 2 days. They are 3 different Bosch Motion Sensors

`3/17/2025 2:47:19 AM

Z2M Server: zh:controller:device: Handling of poll check-in from 0x000d6f001086b6a3 failed (ZCL command 0x000d6f001086b6a3/1 genPollCtrl.checkinRsp({"startFastPolling":false,"fastPollTimeout":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false,"sendPolicy":"immediate"}) failed ({"target":16380,"apsFrame":{"profileId":260,"clusterId":32,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":88},"zclSequence":142,"commandIdentifier":11} timed out after 10000ms))

3/16/2025 8:40:47 AM

Z2M Server: zh:controller:device: Handling of poll check-in from 0x000d6f0011c8d507 failed (ZCL command 0x000d6f0011c8d507/1 genPollCtrl.checkinRsp({"startFastPolling":false,"fastPollTimeout":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false,"sendPolicy":"immediate"}) failed ({"target":61003,"apsFrame":{"profileId":260,"clusterId":32,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":208},"zclSequence":200,"commandIdentifier":11} timed out after 10000ms))

3/16/2025 6:10:28 AM

Z2M Server: zh:controller:device: Handling of poll check-in from 0x000d6f0011c74ee7 failed (ZCL command 0x000d6f0011c74ee7/1 genPollCtrl.checkinRsp({"startFastPolling":false,"fastPollTimeout":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false,"sendPolicy":"immediate"}) failed ({"target":861,"apsFrame":{"profileId":260,"clusterId":32,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":238},"zclSequence":137,"commandIdentifier":11} timed out after 10000ms))`

[image]

Nerivec, 2025-03-17:
And you confirmed with `ember-zli` that no ZigBee channels are used near you? (ZigBee, Thread...)

I think you might benefit from putting a dedicated router somewhere in the middle of your network, so it can take over some of the routes, and improve overall LQI. _You can do this with most adapters (ember/zstack) you'd buy for Z2M, but instead you flash it with a router firmware, then you only need a plug somewhere (and something like a leftover phone charger to power it). It pairs like any regular device after that._
https://github.com/Nerivec/silabs-firmware-builder/discussions/16
https://github.com/Koenkk/Z-Stack-firmware/tree/master/router/Z-Stack_3.x.0/bin

Ltek, 2025-03-17:
The coordinator is in the middle of the house, lower level. I have the routers on the 2 levels above (where most of the sensors are). Oddly the sensors that disconnect and/or have the slowest responses are some of the closest to the coordinator or router.

I have no 2.4 interference from neighbors, at all... my own 2.4 network is Channel 11, 20Mhz bandwidth. Zigbee channels are 11 and 25. When I tried other channels, the Z2M network never connected all nodes - left about half of them off. The Dezonz/Conbee3 network never drops connections (Conbee & Sonoff dongles are in the same location)

At this point, I'm 99% its not external Wifi interference... so that leaves either the Sonoff dongle not being reliable or something in software. This is since I've had Deconz/Conbee for years, using the same devices, in the same locations... and never having a problem with lag or disconnecting devices. I've literally had the same devices on the Deconz for 3 years, 2 different dongle, and a dozen or more updates to Deconz... they never drop off - even when I had the Conbee v1

closed 2025-05-25 by github-actions[bot]