Koenkk/zigbee2mqtt #26699: Z2M Fails to create Node Routes with multiple Routers

Ltek, 2025-03-11:
### What happened?

I thought I've make my network stronger by adding plugin routers (outlets). At first I added 2, and things were working pretty well. About a week later I added 2 more... and almost everything stopped working.

I've attached a few log files.

[log.zip](https://github.com/user-attachments/files/19185388/log.zip)

I spent 2 days trying to make Z2M work with additional 2 routers. I even tried adding a 5th, different brand (Osram) that I removed from my Conbee/Deconz network and know was working properly. Then I tried moving the routers around the house (3 levels - so they were already spread apart properly) and none of that helped.

There is definitely a problem with the routing when you have several routers in the network.

When I had all the routers plugged-in, only 5 or 6 nodes (changed after each restart of Z2M) would be connected to a router. Now its working as expected, even only a few seconds after removing 3 routers and restarting Z2M...

[image]

### How to reproduce it (minimal and precise)

add more than 2 routers

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

8.0.2 [GA]

### Adapter

Dongle-E

### Setup

Windows PC

chris-1243, 2025-03-11:
I would invite you to carefully take a look at your network map.

You have three routers with a really poor link quality. The errors thrown by Z2M in this scenario are normal.

Move your new routers closer to each other or add routers between them. A concrete wall might kill a zigbee signal as well.

I have approximately 50 devices in my network. 2/3 are routers and everything is working well.

Ltek, 2025-03-11:
> I would invite you to carefully take a look at your network map.
>
> You have three routers with a really poor link quality. The errors thrown by Z2M in this scenario are normal.
>
> Move your new routers [...]

They have zero link because I removed (unplugged / no power to them) those so that my network would work... with them plugged in, more than half the nodes do not connect to anything. That's the problem I'm posting about.

You just get to see them in the map since Z2M doesnt remove them from the map even when they are powered off.

I'm not sure where the breaking point is, 3 routers, 4, etc... but I can tell you for sure, when I removed those 3 my network fully connects. I do not have any concrete walls inside my house. I have 107 Zwave devices & ~72 Zigbee devices (2 different networks, diff channels 11 & 25)... all working fine for many years. Until I added routers to Z2M, then Z2M poops out while all other others are fine.

Nerivec, 2025-03-12:
Since you are having clear interference troubles in your other issues, I would say you need to fix that first.
If specific devices cause problems, even if in different places, then in all likelihood, they are the problem. Bad routers are numerous unfortunately, and they can single-handedly take down an entire network.

_Also, Z2M has no say in the network's routes, this is handled by the coordinator and the routers._

To confirm what chris said, z2m/ember was tested with networks of over 300 devices, and working fine.

closed 2025-03-12 by Nerivec

Ltek, 2025-03-12:
Dismissing problems as "must be a problem with 'your' network, not Z2M" doesnt help anyone and reduces confidence, **_even if you are correct_**. Based on numerous user posts with issues like Delivery Broadcast Failed, are common. When Z2M doesnt provide any way to figure out what is causing the issues. There should be a way that Z2M could at least help us isolate what device the issue is coming from... without that, its really hard simply accept without any data supporting it, that Z2M is not the problem. Z2M is far from flawless, I have personally reported several verified bugs (device pairing/config issues, etc - some have been fixed, not all) and I've not used Z2M for very long.