Koenkk/zigbee2mqtt #26700: Can’t change adapter from ezsp to ember

ai, 2025-03-11:
### What happened?

I use SLZB-06M ZigBee adapter. Yes, I know that I need to use it with `ember` but the problem is that I can’t change it.

My config:

```yml
frontend:
  enabled: true
  port: 8124
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://localhost:1883
serial:
  port: mdns://slzb-06
  baudrate: 115200
  adapter: ember
  rtscts: false
advanced:
  channel: 25
  log_level: info
homeassistant:
  enabled: true
  experimental_event_entities: false
  legacy_action_sensor: false
version: 4
```

But I still have in log:

```
[2025-03-11 17:57:26] info: 	z2m: Logging to console, file (filename: log.log)
[2025-03-11 17:57:26] info: 	z2m: Starting Zigbee2MQTT version 2.1.3 (commit #ba337bd329aeb4ca17735c0cf09b31293c8cff06)
[2025-03-11 17:57:26] info: 	z2m: Starting zigbee-herdsman (3.2.7)
[2025-03-11 17:57:26] info: 	zh:adapter:discovery: Starting mdns discovery for coordinator: slzb-06
[2025-03-11 17:57:27] info: 	zh:adapter:discovery: Coordinator Address: 192.168.50.27
[2025-03-11 17:57:27] info: 	zh:adapter:discovery: Coordinator Port: 6638
[2025-03-11 17:57:27] info: 	zh:adapter:discovery: Coordinator Radio: ezsp
[2025-03-11 17:57:27] warning: 	zh:ezsp: 'ezsp' driver is deprecated and will only remain to provide support for older firmware (pre 7.4.x). Migration to 'ember' is recommended. If using Zigbee2MQTT see https://github.com/Koenkk/zigbee2mqtt/discussions/21462
[2025-03-11 17:57:30] error: 	z2m: Error while starting zigbee-herdsman
[2025-03-11 17:57:30] error: 	z2m: Failed to start zigbee-herdsman
[2025-03-11 17:57:30] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-03-11 17:57:30] error: 	z2m: Exiting...
[2025-03-11 17:57:30] error: 	z2m: Error: 'ezsp' driver is not compatible with firmware 8.x.x or above (EZSP v14+). Use 'ember' driver instead.
```

### What did you expect to happen?

To run without `ezsp` warning/error.

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

20250220

### Adapter

SLZB-06M

### Setup

Podman image

### Debug log

```
2025-03-11 18:06:26] debug: 	z2m: Removing old log directory '/app/data/log/2025-03-11.17-51-07'
[2025-03-11 18:06:26] info: 	z2m: Logging to console, file (filename: log.log)
[2025-03-11 18:06:26] debug: 	z2m: Loaded state from file /app/data/state.json
[2025-03-11 18:06:26] info: 	z2m: Starting Zigbee2MQTT version 2.1.3 (commit #ba337bd329aeb4ca17735c0cf09b31293c8cff06)
[2025-03-11 18:06:26] debug: 	z2m: sd-notify loaded
[2025-03-11 18:06:26] info: 	z2m: Starting zigbee-herdsman (3.2.7)
[2025-03-11 18:06:26] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":6754,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[25],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/app/data/database.db\",\"databaseBackupPath\":\"/app/data/database.db.backup\",\"backupPath\":\"/app/data/coordinator_backup.json\",\"serialPort\":{\"rtscts\":false,\"path\":\"mdns://slzb-06\",\"adapter\":\"ember\"},\"adapter\":{\"disableLED\":false}}"'
[2025-03-11 18:06:26] info: 	zh:adapter:discovery: Starting mdns discovery for coordinator: slzb-06
[2025-03-11 18:06:26] info: 	zh:adapter:discovery: Coordinator Address: 192.168.50.27
[2025-03-11 18:06:26] info: 	zh:adapter:discovery: Coordinator Port: 6638
[2025-03-11 18:06:26] info: 	zh:adapter:discovery: Coordinator Radio: ezsp
[2025-03-11 18:06:26] debug: 	zh:ezsp: Adapter concurrent: 8
[2025-03-11 18:06:26] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[25]},"serialPort":{"rtscts":false,"path":"tcp://192.168.50.27:6638","adapter":"ezsp"},"adapter":{"disableLED":false},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json"}'
[2025-03-11 18:06:26] warning: 	zh:ezsp: 'ezsp' driver is deprecated and will only remain to provide support for older firmware (pre 7.4.x). Migration to 'ember' is recommended. If using Zigbee2MQTT see https://github.com/Koenkk/zigbee2mqtt/discussions/21462
```

ai, 2025-03-11:
In Z2M UI I have `Coordinator type EZSP v13` too and not something with Ember.

Nerivec, 2025-03-12:
Change your configuration to this for serial>port:
```yaml
serial:
  port: tcp://192.168.50.27:6638
```

The mDNS discovery issue (not using `ember`) will be fixed in next release.

closed 2025-03-12 by Nerivec