Koenkk/zigbee2mqtt #26705: Unable to start Z2M after latest dev update

mrmaximas, 2025-03-11:
### What happened?

Debug log is every 10s like this:
Nothing else.
```
[2025-03-12 00:17:43.255] debug: 	z2m: Removing old log directory '/Users/m/zigbee2mqtt/data/log/2025-03-12.00-15-51'
[2025-03-12 00:17:43.256] info: 	z2m: Logging to console, file (filename: log.log)
[2025-03-12 00:17:43.257] debug: 	z2m: Loaded state from file /Users/m/zigbee2mqtt/data/state.json
```

### Zigbee2MQTT version

latest dev

### Adapter firmware version

latest

### Adapter

2652P7

### Setup

Mac OS latest

### Debug log

[2025-03-12 00:17:43.255] debug: 	z2m: Removing old log directory '/Users/m/zigbee2mqtt/data/log/2025-03-12.00-15-51'
[2025-03-12 00:17:43.256] info: 	z2m: Logging to console, file (filename: log.log)
[2025-03-12 00:17:43.257] debug: 	z2m: Loaded state from file /Users/m/zigbee2mqtt/data/state.json

mrmaximas, 2025-03-11:
Switching to master master didn't help, delete state.json  didn't help

Nerivec, 2025-03-12:
Likely linked to https://github.com/Koenkk/zigbee2mqtt/pull/26704

mrmaximas, 2025-03-12:
@Nerivec thanks, it's helped. FYI a few days ago update.sh doesn't work as it should on mac os. "No update available."
I was forced to run a sequence of commands in the z2m dir
```
sudo git pull --no-rebase
sudo pnpm i --frozen-lockfile
sudo pnpm run build
```
after this commands i see:

```
git branch --show-current
dev

git rev-list HEAD...origin/"$(git branch --show-current)" --count
0
```

i'll check it before next update.

closed 2025-03-12 by mrmaximas

Nerivec, 2025-03-12:
@Koenkk looks like https://github.com/Koenkk/zigbee2mqtt/pull/26631 could also be causing problems?

mrmaximas, 2025-03-12:
@Nerivec I'll wait for the next dev update and try running `git rev-list HEAD...origin/"$(git branch --show-current)” --count`

mrmaximas, 2025-03-12:
@Nerivec trying to update.
```
zigbee2mqtt % git rev-list HEAD...origin/"$(git branch --show-current)" --count
0
zigbee2mqtt % sudo git pull --no-rebase
Password:
remote: Enumerating objects: 18, done.
remote: Counting objects: 100% (18/18), done.
remote: Compressing objects: 100% (15/15), done.
remote: Total 18 (delta 6), reused 3 (delta 3), pack-reused 0 (from 0)
Unpacking objects: 100% (18/18), 57.88 KiB | 1004.00 KiB/s, done.
From https://github.com/Koenkk/zigbee2mqtt
   fcbb355f..5db11505  dev                                                    -> origin/dev
 + e5ba9cf3...c0c5234a release-please--branches--dev--components--zigbee2mqtt -> origin/release-please--branches--dev--components--zigbee2mqtt  (forced update)
Updating fcbb355f..5db11505
Fast-forward
 lib/extension/frontend.ts | 2 +-
 lib/util/utils.ts         | 2 +-
 package.json              | 2 +-
 tsconfig.json             | 3 +--
 4 files changed, 4 insertions(+), 5 deletions(-)
zigbee2mqtt % sudo git pull --no-rebase
Password:
remote: Enumerating objects: 18, done.
remote: Counting objects: 100% (18/18), done.
remote: Compressing objects: 100% (15/15), done.
remote: Total 18 (delta 6), reused 3 (delta 3), pack-reused 0 (from 0)
Unpacking objects: 100% (18/18), 57.88 KiB | 1004.00 KiB/s, done.
From https://github.com/Koenkk/zigbee2mqtt
   fcbb355f..5db11505  dev                                                    -> origin/dev
 + e5ba9cf3...c0c5234a release-please--branches--dev--components--zigbee2mqtt -> origin/release-please--branches--dev--components--zigbee2mqtt  (forced update)
Updating fcbb355f..5db11505
Fast-forward
 lib/extension/frontend.ts | 2 +-
 lib/util/utils.ts         | 2 +-
 package.json              | 2 +-
 tsconfig.json             | 3 +--
 4 files changed, 4 insertions(+), 5 deletions(-)
zigbee2mqtt % sudo pnpm i --frozen-lockfil
Lockfile is up to date, resolution step is skipped
Already up to date

   ╭──────────────────────────────────────────────────────────────────╮
   │                                                                  │
   │                Update available! 10.4.1 → 10.6.2.                │
   │   Changelog: https://github.com/pnpm/pnpm/releases/tag/v10.6.2   │
   │                Run "pnpm add -g pnpm" to update.                 │
   │                                                                  │
   ╰──────────────────────────────────────────────────────────────────╯

Done in 421ms using pnpm v10.4.1
zigbee2mqtt % sudo pnpm run build

> zigbee2mqtt@2.1.3-dev build /Users/max/zigbee2mqtt
> tsc && node index.js writehash

```

mrmaximas, 2025-03-13:
@Nerivec @Koenkk I tried to update through the script a two minutes ago, but it still  is no update.
```
max@Homebridge zigbee2mqtt % sudo launchctl bootout system/zigbee2mqt
max@Homebridge zigbee2mqtt % sudo /Users/max/zigbee2mqtt/update.sh
No update available.
max@Homebridge zigbee2mqtt % git rev-list HEAD...origin/"$(git branch --show-current)" --count
0
max@Homebridge zigbee2mqtt % sudo git pull --no-rebase
remote: Enumerating objects: 8, done.
remote: Counting objects: 100% (8/8), done.
remote: Compressing objects: 100% (5/5), done.
remote: Total 8 (delta 3), reused 3 (delta 3), pack-reused 0 (from 0)
Unpacking objects: 100% (8/8), 48.24 KiB | 882.00 KiB/s, done.
From https://github.com/Koenkk/zigbee2mqtt
   5db11505..8d2ef6df  dev                                                    -> origin/dev
 + c0c5234a...7f2d3c97 release-please--branches--dev--components--zigbee2mqtt -> origin/release-please--branches--dev--components--zigbee2mqtt  (forced update)
Updating 5db11505..8d2ef6df
Fast-forward
 package.json   |  2 +-
 pnpm-lock.yaml | 10 +++++-----
 2 files changed, 6 insertions(+), 6 deletions(-)
max@Homebridge zigbee2mqtt % sudo pnpm i --frozen-lockfile
Lockfile is up to date, resolution step is skipped
Packages: +1
+
Progress: resolved 1, reused 0, downloaded 1, added 1, done

dependencies:
- zigbee-herdsman-converters 23.6.0
+ zigbee-herdsman-converters 23.7.0

Done in 1.3s using pnpm v10.4.1
max@Homebridge zigbee2mqtt % sudo pnpm run build

> zigbee2mqtt@2.1.3-dev build /Users/max/zigbee2mqtt
> tsc && node index.js writehash

max@Homebridge zigbee2mqtt % sudo launchctl bootstrap system /Library/LaunchDaemons/zigbee2mqtt.plist
max@Homebridge zigbee2mqtt %
```

Koenkk, 2025-03-15:
Should be fixed now, but you will need to manually update the update.sh with the one from the [repo](https://github.com/Koenkk/zigbee2mqtt/blob/dev/update.sh) first.

mrmaximas, 2025-03-15:
@Koenkk right now i see:
 ```
 git rev-list HEAD...origin/"$(git branch --show-current)" --count
0
```

DragonWork, 2025-03-15:
> [@Koenkk](https://github.com/Koenkk) right now i see:
> …
> but update.sh works good! thanks!
> …
> ```
> error: Your local changes to the following files would be overwritten by merge:
> 	update.sh
> Please commit your changes or stash them before you merge.
> Aborting

Just as an info: That‘s because you modified `update.sh` yourself, so `git pull` doesn‘t want to overwrite it without you stashing or deleting your change. The same error would occur without the new check too (in fact, changing any repo-owned file will do the same), as I didn‘t touch that simple `git pull --rebase` part. But you‘ve resolved the hindrance by renaming it. 👍🏻 You can remove that `update1.sh` now.

mrmaximas, 2025-03-17:
@DragonWork strange things. No updates via git rev-list HEAD...origin/"$(git branch --show-current)" --count but updated ;)
```
max@Homebridge zigbee2mqtt % git rev-list HEAD...origin/"$(git branch --show-current)" --count
0
max@Homebridge zigbee2mqtt % sudo /Users/max/zigbee2mqtt/update.sh
remote: Enumerating objects: 23, done.
remote: Counting objects: 100% (23/23), done.
remote: Compressing objects: 100% (12/12), done.
remote: Total 23 (delta 14), reused 13 (delta 11), pack-reused 0 (from 0)
Unpacking objects: 100% (23/23), 90.66 KiB | 919.00 KiB/s, done.
From https://github.com/Koenkk/zigbee2mqtt
   92955c02..325199d4  dev                                                    -> origin/dev
   ba337bd3..6d6372cc  master                                                 -> origin/master
 + f4d98767...1d446353 release-please--branches--dev--components--zigbee2mqtt -> origin/release-please--branches--dev--components--zigbee2mqtt  (forced update)
Update available!
Skipped stopping Zigbee2MQTT, no systemctl found
Updating...
Updating 92955c02..325199d4
Fast-forward
 .github/workflows/update-dependencies.yml |   1 +
 package.json                              |   4 +-
 pnpm-lock.yaml                            | 428 +++++++++++++++++++++++++++++++++++++++++++++++++++++++-------------------------------------------------------
 test/extensions/bridge.test.ts            |   1 -
 4 files changed, 217 insertions(+), 217 deletions(-)
Installing dependencies...
Lockfile is up to date, resolution step is skipped
Packages: +34 -5
++++++++++++++++++++++++++++++++++-----
Progress: resolved 34, reused 30, downloaded 4, added 22, done

dependencies:
- zigbee-herdsman-converters 23.7.0
+ zigbee-herdsman-converters 23.8.0

devDependencies:
- typescript-eslint 8.26.0
+ typescript-eslint 8.26.1

Done in 990ms using pnpm v10.4.1
Building...

> zigbee2mqtt@2.1.3-dev build /Users/max/zigbee2mqtt
> tsc && node index.js writehash

Done!
max@Homebridge zigbee2mqtt % sudo /Users/max/zigbee2mqtt/update.sh
No update available. Use '/Users/max/zigbee2mqtt/update.sh force' to skip the check.
max@Homebridge zigbee2mqtt % git rev-list HEAD...origin/"$(git branch --show-current)" --count
0
max@Homebridge zigbee2mqtt %
```

DragonWork, 2025-03-17:
Nothing unusual or mysterious about it as the `update.sh` does a `git fetch` before the `git rev-list` command now. 😉

mrmaximas, 2025-03-17:
@DragonWork thanks!