Koenkk/zigbee2mqtt #26708: Tuya Energy CT Clamp (_TZE200_rks0sgb7) Freezes Every 2-3 Months — Requires Re-pairing

LightMoon, 2025-03-12:
### What happened?

The Zigbee Tuya Energy CT Clamp (_TZE200_rks0sgb7) disconnects randomly every 2 to 3 months. The exposed parameters (e.g., power readings) freeze and stop updating, while the device availability still shows as "Available". No errors appear in the logs, and the only way to fix it is to remove and re-pair the device — which works immediately, but the issue reoccurs after another 2-3 months.

### What did you expect to happen?

The device should maintain a stable connection, continuously update exposed values, and recover from any temporary disconnections without manual intervention or re-pairing.

### How to reproduce it (minimal and precise)

- Add the Zigbee CT Clamp (_TZE200_rks0sgb7) to Zigbee2MQTT.
- Let it run normally, monitoring the exposed parameters (e.g., Power).
- Wait for 2-3 months — eventually, the values freeze, but the device remains marked as "Available".
- Remove the device, re-enable pairing mode, and re-pair it — it works again immediately for another 2-3 months.

### Zigbee2MQTT version

2.1.3 (commit: unknown)

### Adapter firmware version

7.4.4 [GA]

### Adapter

Sonoff ZBDongle-E (EFR32MG21)

### Setup

TBC

Nerivec, 2025-03-12:
I'd say it's likely something about the device itself (firmware issue, seems a restart -re-pair- fixes it), or the network (could be trying to switch to another router and something doesn't work out when it does).
Hard to debug such a problem (long span...), if nothing in the logs points to anything in particular, I'm not sure we can figure out much.
You can try this in the meantime:
- confirm the values are not just frozen in frontend/HA, but also in the logs (don't see reporting from that device anymore)
- identify the router it is currently connected to, try to force it to connect to another (ideally, directly to the coordinator), it might revert, but if the router has a good enough link, it should stay in place
- see if you can narrow down the pattern, as in, exactly every X days, or on a specific day of the month, anything that could point to a trigger of some kind (on the device, the network, your automations, etc.) that could influence this

LightMoon, 2025-03-13:
I enabled debug mode and couldn't find any errors related to the issue that occurred before re-pairing the device.

> * confirm the values are not just frozen in frontend/HA, but also in the logs (don't see reporting from that device anymore)

Good idea, I will do that when it happens again.

> * identify the router it is currently connected to, try to force it to connect to another (ideally, directly to the coordinator), it might revert, but if the router has a good enough link, it should stay in place

How can I identify the current router, and more importantly, how can I force it to connect to another?

> * see if you can narrow down the pattern, as in, exactly every X days, or on a specific day of the month, anything that could point to a trigger of some kind (on the device, the network, your automations, etc.) that could influence this

I have four Tuya energy CT clamps, and each of them exhibits the same random behavior.

Nerivec, 2025-03-13:
> How can I identify the current router

Using the network map. Or when your device joins, in the `debug` logs, you can also see `parentOfNewNodeId=xxx`.

> how can I force it to connect to another

When pairing, enable "permit join" on a specific device instead of "all".

> each of them exhibits the same random behavior

I assume you mean the same model?
I have [these](https://www.zigbee2mqtt.io/devices/PJ-MGW1203.html#tuya-pj-mgw1203), they don't have the problem (but obviously a bit different, so a different firmware inside).

LightMoon, 2025-03-13:
The issue occurred again much sooner than expected after I repaired the device on Thursday.

**Current Status:**

- P**roblematic Device:** CT_GPOs (shows as "available" but not updating)
- **Working Devices:** CT_AC_Lights & CT_Solar_Growatt (updating as expected)
- I haven’t re-repaired CT_GPOs yet in case you’d like to check something beforehand

**Debug Findings:**
I attempted to locate parentOfNewNodeId in the logs but couldn’t find any mention — possibly checked the wrong location.
The logs didn’t provide much insight from what I could interpret.
Would a screenshot of the Zigbee map help diagnose the issue? I’m happy to provide one if needed.

Below are the logs from the Zigbee2MQTT add-on for further review:
```
2025-03-13 23:20:32] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/CT_AC_Lights', payload '{"ac_frequency":50,"current":2.97,"current_a":0,"current_b":3.25,"energy":1555.67,"energy_a":125.76,"energy_b":1429.91,"energy_produced":0,"energy_produced_a":0,"energy_produced_b":0,"last_seen":"2025-03-13T12:20:32.921Z","linkquality":128,"power":699,"power_a":0,"power_b":869,"power_factor_a":0,"power_factor_b":88,"voltage":252.3}'
[2025-03-13 23:20:33] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-03-13 23:20:33] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4](ackRx=4 frmTx=4)
[2025-03-13 23:20:33] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4 frmNum=1](frmRx=1) Added to rxQueue
[2025-03-13 23:20:33] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=2](ackRx=4)
[2025-03-13 23:20:33] debug: 	zh:ember:ezsp: <=== [CBFRAME: ID=89:"INCOMING_ROUTE_RECORD_HANDLER" Seq=59 Len=20]
[2025-03-13 23:20:33] debug: 	zh:ember:ezsp: ezspIncomingRouteRecordHandler(): callback called with: [source=60617], [sourceEui=0xb43522fffee0f718], [lastHopLqi=124], [lastHopRssi=-69], [relayCount=1], [relayList=345]
[2025-03-13 23:20:33] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-03-13 23:20:33] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4](ackRx=4 frmTx=4)
[2025-03-13 23:20:33] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4 frmNum=2](frmRx=2) Added to rxQueue
[2025-03-13 23:20:33] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=3](ackRx=4)
[2025-03-13 23:20:33] debug: 	zh:ember:ezsp: <=== [CBFRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=59 Len=40]
[2025-03-13 23:20:33] debug: 	zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=UNICAST], [apsFrame={"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":256,"groupId":0,"sequence":112}], [packetInfo:{"senderShortId":60617,"senderLongId":"0xFFFFFFFFFFFFFFFF","bindingIndex":255,"addressIndex":255,"lastHopLqi":128,"lastHopRssi":-68,"lastHopTimestamp":0}], [messageContents=08360a01002044e2ff201fe4ff2000]
[2025-03-13 23:20:33] debug: 	zh:controller: Received payload: clusterID=0, address=60617, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=128, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":54,"commandIdentifier":10},"payload":[{"attrId":1,"dataType":32,"attrData":68},{"attrId":65506,"dataType":32,"attrData":31},{"attrId":65508,"dataType":32,"attrData":0}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2025-03-13 23:20:33] debug: 	zh:controller:endpoint: ZCL command 0xb43522fffee0f718/1 genBasic.defaultRsp({"cmdId":10,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"transactionSequenceNumber":54,"writeUndiv":false})
[2025-03-13 23:20:33] debug: 	zh:ember: ~~~> [ZCL to=0xb43522fffee0f718:60617 apsFrame={"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":4352,"groupId":0,"sequence":0} header={"frameControl":{"reservedBits":0,"frameType":0,"direction":1,"disableDefaultResponse":true,"manufacturerSpecific":false},"transactionSequenceNumber":54,"commandIdentifier":11}]
[2025-03-13 23:20:33] debug: 	zh:ember:ezsp: ===> [FRAME: ID=52:"SEND_UNICAST" Seq=60 Len=26]
[2025-03-13 23:20:33] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=4 frmRx=3](ackRx=4)
[2025-03-13 23:20:33] debug: 	z2m: Received Zigbee message from 'CT_GPOs', type 'attributeReport', cluster 'genBasic', data '{"65506":31,"65508":0,"appVersion":68}' from endpoint 1 with groupID 0
[2025-03-13 23:20:33] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/CT_GPOs', payload '{"ac_frequency":50,"current":1.68,"current_a":0.892,"current_b":0.858,"energy":1394.25,"energy_a":692.11,"energy_b":702.14,"energy_produced":0,"energy_produced_a":0,"energy_produced_b":0,"last_seen":"2025-03-13T12:20:33.222Z","linkquality":128,"power":219,"power_a":117,"power_b":101,"power_factor_a":56,"power_factor_b":40,"voltage":252.6}'
[2025-03-13 23:20:33] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-03-13 23:20:33] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5](ackRx=4 frmTx=5)
[2025-03-13 23:20:33] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5 frmNum=3](frmRx=3) Added to rxQueue
[2025-03-13 23:20:33] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=4](ackRx=5)
[2025-03-13 23:20:33] debug: 	zh:ember:ezsp: <=== [FRAME: ID=52:"SEND_UNICAST" Seq=60 Len=7]
[... cut ...] tor_b":0,"voltage":253.4}'

```

Logs filtered based on CT_GPOs

```
info 2025-03-13 23:45:08z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/CT_GPOs', payload '{"ac_frequency":50,"current":1.68,"current_a":0.892,"current_b":0.858,"energy":1394.25,"energy_a":692.11,"energy_b":702.14,"energy_produced":0,"energy_produced_a":0,"energy_produced_b":0,"last_seen":"2025-03-13T12:45:08.104Z","linkquality":120,"power":219,"power_a":117,"power_b":101,"power_factor_a":56,"power_factor_b":40,"voltage":252.6}'
```

Nerivec, 2025-03-13:
I don't see any joining in these logs.
You'd see a line like this:
```
zh:ember:ezsp: ezspTrustCenterJoinHandler(): callback called with: [newNodeId=xxx], [newNodeEui64=xxx], [status=xxx], [policyDecision=xxx], [parentOfNewNodeId=xxx]
```

Can you take screenshots of:
- network map, after isolating (drag them away from others so I can see it better) the device in question, devices with direct lines to it (not dashed lines) and the coordinator
- the `Reporting` page of the device

LightMoon, 2025-03-17:
Strange! It seems the `reporting` page doesn’t show anything useful. I’ve also uploaded the logs in case you’d like to take a look.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

[45df7312_zigbee2mqtt_2025-03-17T11-32-08.491Z.log](https://github.com/user-attachments/files/19285279/45df7312_zigbee2mqtt_2025-03-17T11-32-08.491Z.log)

closed 2025-05-25 by github-actions[bot]

MilesTEG1, 2025-11-12:
Hello 👋🏻 
I experience the same problem with my PC311-Z-TY clamps…
After some times (I don’t see a pattern, it’s random, sometimes it’s after some weeks, sometimes it’s months , or days…) the device doesn’t publish energy …
I need to switch off the circuit breaker to which my two PC311-Z-TY modules (2 clamps each) are connected for a few tens of seconds and then switch it back on.

Is there a long-term solution?
Thanks in advance

stadros83, 2026-01-07:
> Hello 👋🏻 I experience the same problem with my PC311-Z-TY clamps… After some times (I don’t see a pattern, it’s random, sometimes it’s after some weeks, sometimes it’s months , or days…) the device [...]

I have the same problem with two devices! I have to turn them off and then on again for them to work again.