Koenkk/zigbee2mqtt #26712: [New device support]: NOUS B4Z

mdraaijer, 2025-03-12:
### Link

https://www.zigbee2mqtt.io/devices/B4Z.html

### Database entry

{"id":53,"type":"Router","ieeeAddr":"0xa4c138789f82ec09","nwkAddr":23920,"manufId":4417,"manufName":"_TZ3000_yruungrl","powerSource":"Mains (single phase)","modelId":"TS130F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":514,"inClusterList":[4,5,6,258,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS130F","manufacturerName":"_TZ3000_yruungrl","powerSource":1,"zclVersion":3,"appVersion":71,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"tuyaBacklightSwitch":0,"tuyaBacklightMode":2}},"closuresWindowCovering":{"attributes":{"32768":0,"currentPositionLiftPercentage":72,"tuyaMovingState":1,"tuyaCalibration":1,"tuyaMotorReversal":0,"moesCalibrationTime":100}},"manuSpecificTuya_3":{"attributes":{"switchType":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":71,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1741703520319}

### Zigbee2MQTT version

 1.37.1 commit: cda867a

### Comments

This is fantastic hardware, i bought many modules.
The only problem is there are modules which behave differently.
These modules need a software upgrade.

As a developer i would like to contribute in upgrading the firmware.
Or maybe some of you already did this ?
At least in the latest version i do not see this.

Can someone help ?

### External definition

```shell
const {onOff, windowCovering} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS130F'],
    model: 'TS130F',
    vendor: '_TZ3000_yruungrl',
    description: 'Automatically generated definition',
    extend: [onOff({"powerOnBehavior":false}), windowCovering({"controls":["lift","tilt"]})],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

 OTA does not work

Koenkk, 2025-03-13:
You are using an outdated z2m version, this device is supported in recent z2m versions.

closed 2025-03-13 by Koenkk

mdraaijer, 2025-03-15:
I upgraded Zigbee2MQTT version [2.1.3](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.1.3) commit: [ba337bd](https://github.com/Koenkk/zigbee2mqtt/commit/ba337bd)

forced delete the module and rebooted the server
then paired the module again.

model still TS130F and still does not show up in the OTA list, which is the attempt to upgrade the firmware of the B4Z module.

**generated_external_definition:**

const m = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS130F'],
    model: 'TS130F',
    vendor: '_TZ3000_yruungrl',
    description: 'Automatically generated definition',
    extend: [m.onOff({"powerOnBehavior":false}), m.windowCovering({"controls":["lift","tilt"]})],
    meta: {},
};

module.exports = definition;

mdraaijer, 2025-03-15:
also in 

/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman-converters@23.2.0/node_modules/zigbee-herdsman-converters/dist/devices/tuya.js

 {
        zigbeeModel: ["TS130F"], 
        model: "TS130F",
        vendor: "Tuya",
        description: "Curtain/blind switch",
        fromZigbee: [fz.cover_position_tilt, tuya.fz.indicator_mode, fz.tuya_cover_options, tuya.fz.backlight_mode_off_on],
        toZigbee: [
            tz.cover_state,
            tz.cover_position_tilt,
            tz.tuya_cover_calibration,
            tz.tuya_cover_reversal,
            tuya.tz.backlight_indicator_mode_2,
            tuya.tz.backlight_indicator_mode_1,
        ],
        meta: { coverInverted: true },
        whiteLabel: [
            tuya.whitelabel("Danor", "SK-Z802C-US", "Smart curtain/shutter switch", ["_TZ3000_8h7wgocw"]),
            { vendor: "LoraTap", model: "SC400" },
            tuya.whitelabel("LoraTap", "SC500ZB", "Roller Shutter Blind Module", ["_TZ3000_e3vhyirx"]),
            tuya.whitelabel("Nous", "**B4Z**", "Curtain switch", ["_TZ3000_yruungrl"]),
            tuya.whitelabel("Nous", "L12Z", "Smart ZigBee Curtain Module L12Z", ["_TZ3000_jwv3cwak"]),
            tuya.whitelabel("Zemismart", "ZN-LC1E", "Smart curtain/shutter switch", ["_TZ3000_74hsp7qy"]),
        ], 
        exposes: (device) => {
            const exps = [
                e.cover_position(), 
                e.enum("moving", ea.STATE, ["UP", "STOP", "DOWN"]),
                e.binary("motor_reversal", ea.ALL, "ON", "OFF"),
            ];
            if (device?.manufacturerName !== "_TZ3000_cet6ch1r") {
                exps.push(e.binary("calibration", ea.ALL, "ON", "OFF"), e.numeric("calibration_time", ea.STATE).withUnit("s").withDescription("Calibration time"));
            }
            else {
                exps.push(e.binary("calibration_up", ea.ALL, "ON", "OFF"), e.numeric("calibration_time_up", ea.STATE).withUnit("s").withDescription("Calibration time up"), e.binary("calibration_down", ea.ALL, "ON", "OFF"), e.numeric("calibration_time_down", ea.STATE).withUnit("s").withDescription("Calibration time down"));
            }
            if (device?.manufacturerName !== "_TZ3210_xbpt8ewc" && device?.manufacturerName !== "_TZ3000_e3vhyirx") {
                exps.push(tuya.exposes.indicatorMode(), tuya.exposes.backlightModeOffOn());
            }
            return exps;
        },
    }