Koenkk/zigbee2mqtt #26720: [New device support]: _TZ3000_mup3wbfp - 2 gang wall mounted sockets with USB C and USB a charging ports and power monitoring.

ViscountHermit, 2025-03-12:
### Link

https://a.aliexpress.com/_EJc10Ag

### Database entry

{"id":111,"type":"Router","ieeeAddr":"0xa4c138969b0ce622","nwkAddr":44284,"manufId":4417,"manufName":"_TZ3000_mup3wbfp","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_mup3wbfp","powerSource":1,"zclVersion":3,"appVersion":78,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1}},"seMetering":{"attributes":{"divisor":100,"multiplier":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x003c84fffec777e3","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x003c84fffec777e3","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x003c84fffec777e3","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":257}],"meta":{}},"2":{"profId":260,"epId":2,"devId":266,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":78,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1741820010293}

### Zigbee2MQTT version

2.1.3

### Comments

I'm no great tech on Z2m sadly. This is a dual gang socket but it only comes up as having a single socket when I look at the exposes tab.

### External definition

```shell
const m = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS011F'],
    model: 'TS011F',
    vendor: '_TZ3000_mup3wbfp',
    description: 'Automatically generated definition',
    extend: [m.deviceEndpoints({"endpoints":{"1":1,"2":2}}), m.onOff({"powerOnBehavior":false,"endpointNames":["1","2"]}), m.electricityMeter()],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Only shows 1 socket / switch.

dagryl, 2025-03-28:
I can see this socket in the database https://www.zigbee2mqtt.io/devices/TS011F_2_gang_power.html
How to map '_TZ3000_mup3wbfp' to TS011F_2_gang_power?

antfurn, 2025-04-11:
Hi, I have one of these double gang sockets reporting as '_TZ3000_mup3wbfp' too.
Would be great to get z2m update to support it correctly.

shopsD, 2025-06-09:
> I can see this socket in the database https://www.zigbee2mqtt.io/devices/TS011F_2_gang_power.html How to map '_TZ3000_mup3wbfp' to TS011F_2_gang_power?

Running into the same issue with my set of sockets

antfurn, 2025-09-09:
If anyone can add full support for this device, that would be great.

closed 2026-01-18 by github-actions[bot]