Koenkk/zigbee2mqtt #26728: Interview failed with error 'Error: Interview failed because can not get node descriptor

sebenik, 2025-03-13:
### What happened?

If I try to add device [[Vimar 14593](https://www.zigbee2mqtt.io/devices/14593.html#vimar-14593)] to the Zigbee network it fails with an error:

### What did you expect to happen?

That device gets connected. 
I have other devices of this type already connected to the network but those were connected lon ago (before z2m upgrade to 2.0).

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

 8.0.2 [GA]

### Adapter

ember

### Setup

Plain on Docker container

### Debug log

info 2025-03-13 14:50:09zh:controller: Interview for '0xf4ce36c67f2aceb0' started
info 2025-03-13 14:50:09z2m: Device '0xf4ce36c67f2aceb0' joined
info 2025-03-13 14:50:09z2m: Starting interview of '0xf4ce36c67f2aceb0'
info 2025-03-13 14:50:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xf4ce36c67f2aceb0","ieee_address":"0xf4ce36c67f2aceb0"},"type":"device_joined"}'
info 2025-03-13 14:50:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xf4ce36c67f2aceb0","ieee_address":"0xf4ce36c67f2aceb0","status":"started"},"type":"device_interview"}'
info 2025-03-13 14:50:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xf4ce36c67f2aceb0","ieee_address":"0xf4ce36c67f2aceb0"},"type":"device_announce"}'
error 2025-03-13 14:50:38zh:controller: Interview failed for '0xf4ce36c67f2aceb0 with error 'Error: Interview failed because can not get node descriptor ('0xf4ce36c67f2aceb0')'
error 2025-03-13 14:50:38z2m: Failed to interview '0xf4ce36c67f2aceb0', device has not successfully been paired
info 2025-03-13 14:50:38z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xf4ce36c67f2aceb0","ieee_address":"0xf4ce36c67f2aceb0","status":"failed"},"type":"device_interview"}'

Nerivec, 2025-03-13:
Did you try pairing it to a specific router (pick one in the permit join dropdown) instead of "all"?
If you can (not too far), try directly to the coordinator.

Could be related to https://github.com/Koenkk/zigbee2mqtt/issues/26676

sebenik, 2025-03-13:
> Did you try pairing it to a specific router (pick one in the permit join dropdown) instead of "all"?

I tried this now and it paired successfully.
Thanks.

closed 2025-03-13 by sebenik