Koenkk/zigbee2mqtt #26732: Unable to configure 'power_on_behavior' for SPLZB-134 device (UNSUPPORTED ATTRIBUTE).

siandco, 2025-03-13:
### What happened?

Unable to configure 'power_on_behavior' for SPLZB-134 device (UNSUPPORTED ATTRIBUTE).
Prior to 2.x this worked as expected.

[image]

### What did you expect to happen?

To be able to configure 'power_on_behavior'

### How to reproduce it (minimal and precise)

Open device in GUI, click on any option for 'power_on_behavior'

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

20221226

### Adapter

SONOFF Zigbee 3.0 USB Dongle P

### Setup

Dockert container

### Debug log

Log contains on entries relevant to device

[2025-03-13.14-55-08.log](https://github.com/user-attachments/files/19233181/2025-03-13.14-55-08.log)

Koenkk, 2025-03-15:
It seems the device doesn't support it, could you provide the debug log when changing this pre z2m 2.0?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

siandco, 2025-03-15:
Rolling back to 1.39.0 (i had backup) the option to configure 'power_on_behavior' is no longer visible.

[image]

What is weird it that the entry on this website show it as being supported [https://www.zigbee2mqtt.io/devices/SPLZB-134.html](https://www.zigbee2mqtt.io/devices/SPLZB-134.html).

If it was never supported can we prevent Z2M from trying to configure it at startup and removal from GUI please?

ImSkully, 2025-04-15:
Can confirm `power_on_behavior` has no effect.

Stumbled upon this issue whilst trying to troubleshoot an issue with SPLZB-134 failing to pass interview on Z2M after 2.0 *(was previously connected but fell off network after updating)*:
```log
[2025-04-15 14:44:06] error: 	zh:controller: Interview failed for '0x0015bc002f00edc9 with error 'Error: Interview failed because can not get active endpoints ('0x0015bc002f00edc9')'
[2025-04-15 14:44:06] error: 	z2m: Failed to interview '0x0015bc002f00edc9', device has not successfully been paired
```

Managed to eventually get it connected by force removing from Z2M and then holding the power button down for 30 seconds to factory reset.

I don't have any logs to share prior to updating though I'm fairly confident that `power_on_behavior` was working correctly in 1.40.2 where it was first introduced and the docs were updated to include this attribute in [`28a8e3c`](https://github.com/Koenkk/zigbee2mqtt.io/commit/28a8e3c2585e4e3a80722364a6fef32c6d2eaa4e#diff-69939fdae4819724e4e92f7e65298091927fdc73c18508a6d7ffa8c147e121d2R116-R121).

siandco, 2025-06-20:
still present, see post from @ImSkully also confirming the issue.

senna1992, 2025-07-13:
Same issue for splzb-131 here, seems to be related