Koenkk/zigbee2mqtt #26736: BUFFER FULL, does anyone know what's the cause?

Tee86, 2025-03-13:
### What happened?

I have the latest HA, latest z2m, latest emqx, Dongle-P with latest FW.
3 days ago I got 
```failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)```
on all my powered devices (a.k.a. routers) it happened every day since then, yesterday twice.

@Koenkk , do we know what's behind that error. I'm on the latest Z-stack FW for at least 3 months and never had this issue. I did not make any significant changes to my setup recently. Just added one IKEA TRETAKT plug, but there were already 3 before this one, never had issue with TRETAKT's.

I have currently 32 devices connected to coordinator. Is it too much? Is there any way to limit that amount?
I'm gonna try tomorrow that 2022 firmware that was suggested in another post, although, it has a mixed feedback... I'd rather find out what the culprit is...

Funny thing is that somehow this affects only powered devices, end-devices remain online, even tho their routers reports to be offline. First time it happened I noticed it after 15 hours, all my 41 powered devices were offline, and all 47 of my end-devices were reporting still online

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

20240710

### Adapter

Sonoff Dongle-P

### Setup

Docker installation, Pi

Koenkk, 2025-03-15:
Could you try the fw from https://github.com/Koenkk/Z-Stack-firmware/issues/518#issuecomment-2698726919 ?

Tee86, 2025-03-15:
Thank you @Koenkk . I downgraded the firmware before your reply to 20221226, I did that about 24 hours ago. I have had no "BUFFER FULL" so far.
Is there some significant improvement in 99250304 in comparison to 20221226?
I noticed that with 20221226 devices react better than with 2024 version I had originally.
The only downside I'm noticing is lower LQI - which might be caused by the mesh rebuilding, right?

@Koenkk , There is one thing I found out yesterday. I run this server for a client of mine, and he received [this device](https://octopus.energy/blog/octopus-home-mini/) a few days ago from his energy supplier. I confirmed that it communicates via Zigbee with the energy meter. Could that be related? Is it possible that it might disrupt the network and cause error like that?

Koenkk, 2025-03-16:
> I confirmed that it communicates via Zigbee with the energy meter. Could that be related? 

Maybe it could cause interference, but it is hard to judge that.

Could you also give `99250304` a try?

Tee86, 2025-03-19:
@Koenkk ,a few updates:
I ran 20221226 since Friday 11AM till Tuesday 11AM, so full 4 days. I had no "buffer full" errors whatsoever during that time, but as I mentioned, the LQI's were quite low, and I had several devices dropping out temporarily (mains powered), and one device (Sonoff door sensor - battery powered) went offline completely and had to be repaired.
I flashed 99250304 on Tuesday, no errors so far, but it's been just one day. I'll monitor it in upcoming days and will drop an update sometime next week.
Thanks for your help btw

Tee86, 2025-03-23:
@Koenkk , here's my update after 5 days with the 99250304... it doesn't seem to suffer from the “buffer full” error. However, I don't know if it's firmware related, but some battery powered devices like to drop out from the network. Interesting is that temperatures sensors are okay (got about 15 of them), however, door and window sensors are a different song. I have only 8 of them and 4 are already offline, to be specific:
1x Sonoff SNZB-04
3x IKEA E2013
Any idea what might be causing this?

Koenkk, 2025-03-23:
@Tee86 maybe this was caused by the fw update itself, could you try the fw from https://github.com/Koenkk/Z-Stack-firmware/discussions/545 and re-pair those devices?

Tee86, 2025-04-01:
Hey @Koenkk , I updated FW last Monday, so I'm on 20250321 for 8 days. Here's my feedback:
- I've had zero "buffer full" errors so far
- Battery operated devices seem to operate more reliably, I had just one SNZB-04 sensor offline twice after the availability timeout ran out (as it hadn't been used during that period), but as soon as I closed that door it went back to online status, so all good on this front.
- LQI is not the best, some devices report LQI 3 or 5. I just got one Moes thermostat offline (mains powered), which had LQI 3 before it went off grid... but so far when this happened with any device, they always came back. I'll try to manually re-pair some devices this week, hopefuly it helps with LQI. Maybe I'll add one or two router sockets, although I think the device density is okay within the flat. Will keep you posted on this.
- It also seems that reaction time of devices is quite good, def better than with 2024 FW. That goes especially for battery-operated devices like scene buttons / switches

Koenkk, 2025-04-01:
Great! Could you also put that feedback here: https://github.com/Koenkk/Z-Stack-firmware/discussions/545

closed 2025-04-01 by Koenkk